MarlinFirmware/Marlin #7446: Unable to control Marlin 1.1.4 with host programs

richgain, 2017-08-06:
Having installed Marlin 1.1.4 on my E3D BigBox IDEX, I find that I cannot print files or move the X, Y and Z axes using a host program like Octoprint or Simplify 3D.
The information displayed on the LCD updates in real time, so the commands are reaching the printer, but none of the movements happen until I use the LCD rotary controller to send a command. Then any previous commands sent from the host are executed as well.
[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/1203488/Configuration_files.zip)
Video showing the issue can be seen here.
https://youtu.be/8mxL7S9IMKc
Anyone know what is going on here?

-------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------

teemuatlut, 2017-08-07:
This has been fixed by #7386. Please update to the latest [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip) or wait for release v1.1.5.

richgain, 2017-08-07:
Perfect! Many thanks for the speedy response. Printer is happily printing away again now.

closed 2017-08-07 by richgain