MarlinFirmware/Marlin #7449: Function "get_sdcard_commands" will cause wdt reset when print sdcard file

MS1987, 2017-08-07:
------------------------------------------------------------
### Bug Report
- Description: ---
  When I print a gcode file (from sdcard) which has plenty of comment lines continuously, it causes the board reset.

- Analasy: ---
  As the watchdog was enabled, in the function "get_sdcard_commands()" there is code as following:
`  if (!sd_count) continue; // skip empty lines (and comment lines)`
 This "continue" will cause long time in the get command loop when the gcode file has enougth comment  lines to  consume.
- Solve: ---
  Replace the "continue" above with "return"

- reproduce:
  Use the gcode file to test.

[test.zip](https://github.com/MarlinFirmware/Marlin/files/1204049/test.zip)

------------------------------------------------------------

iosonopersia, 2017-08-08:
Hi @MS1987 
> Replace the "continue" above with "return"

I suggest to edit code as follows:

```
  // skip empty lines (and comment lines)
  if (!sd_count) {
      #if ENABLED(USE_WATCHDOG)
          watchdog_reset();
      #endif
      continue;
  }
```

Do you think this would work?

MS1987, 2017-08-08:
Yes, I think so!

2017-08-08 

LMS

MS1987, 2017-08-08:
But it will still block the other operation( like button operation and lcd refresh). So I  think skip out will be better. What do you think?

2017-08-08 

李满森
Menson
广州谦辉信息科技有限公司
广州市荔湾区东沙工业区荷景南路23号C栋407-408房
Phone：020-81653576
Fax： 020-81653576
Mobil：13794310267
www.makerbase.com.cn

发件人："lms228"<lms228@163.com>
发送时间：2017-08-08 17:47
主题：Re: Re: [MarlinFirmware/Marlin] Function "get_sdcard_commands" will cause wdt reset when print sdcard file (#7449)
收件人："MarlinFirmware/Marlin"<reply@reply.github.com>
抄送：

Yes, I think so!

2017-08-08 

LMS

MS1987, 2017-08-08:
But it will still block the other operation( like button operation and lcd refresh). So I  think skip out will be better. What do you think?

MS1987, 2017-08-08:
But it will still block the other operation( like button operation and lcd refresh). So I think skip out will be better. What do you think?

2017-08-08 

Menson

iosonopersia, 2017-08-12:
Hi @MS1987,
I thought about it and I think your approach would be better than mine. We should prepare a PR for that.

That behaviour should be used both in get_sdcar_commands() and in get_serial_commands().

I add that in those functions, the cycle continues also if MAX_CMD_SIZE is reached. If that happens, those functions continue to read chars until they find a \n or a \r char. So if someone sends a very long string without \n and \r, the printer will also get stuck for a while, until those EOL characters are found. I'm wondering if we could exit the cycle in that case too.

thinkyhead, 2018-02-09:
Patch will be merged shortly.

thinkyhead, 2018-02-09:
> if someone sends a very long string without \n and \r, the printer will also get stuck for a while

Less worried about this.

closed 2018-02-09 by thinkyhead