MarlinFirmware/Marlin #7450: [MKS Gen 1.4] Fan0 is not working and even crashing all other PWMs

milkpirate, 2017-08-07:
------------------------------------------------------------
### Bug Report
- Description: I am having a MKS Gen 1.4 board configured with `#define MOTHERBOARD BOARD_MKS_13` (EFBF). `PIDTEMP`, `PIDTEMPBED` and `FAST_PWM_FAN` enabled (defined), `FAN_KICKSTART_TIME` 100 and `//#define SLOW_PWM_HEATERS`. The other PWM related settings are pretty much like default.
- Expected behaviour: Controll hotend, heat bed, fan0 and fan1 seperatly via corresponding gcode with  different PWM values (0-255). I.e. gcodes like: `M104 S190` (Extruder=190°C), `M140 S55` (Bed=55°C), `M106 P1 S127` (Fan1=50%)...
- Actual behaviour: All works fine until I set `M106 P0 Sx`=`M106 Sx` with `x != 0`. So if I turn on the extruder fan (fan0) all other PWMs seem to stop with 0% duty cycle (even fan0 isnt turning on). When I issue `M106 S0` the other PWMs return to normal operation. 
- Steps to reproduce: See above

So when I turn on fan0 everything else turn off.

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1204189/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1204190/Configuration.h.txt)

wywywywy, 2017-08-16:
So have you tried it without `FAST_PWM_FAN`?  And are you on 1.1.4?

milkpirate, 2017-09-26:
Seem like it fixed the problem so far. But I might have a dead transistor, ordered new ones, takes time, will do further testing when they arrive. How does it come that `FAST_PWM_FAN` causes the problem?

3daddict, 2017-12-11:
@milkpirate I also have an MKS Gen 1.4 and I am trying to configure independent dual extruder cooling fans and also 1 part cooling fan all running on 24v.
Can you share where to wire these on the mainboard? I had overlooked this and have now seemed to hit a wall...

closed 2018-02-10 by thinkyhead

3daddict, 2018-02-10:
In my case I ended up using MOSFET Drivers and then setting custom pins in marlin.
http://3daddict.com/marlin-custom-fan-pins/
In case anyone needs info on how to set that up.

thinkyhead, 2018-02-14:
Nice! MOSFETS are definitely one of our favorite components.