MarlinFirmware/Marlin #7452: BLtouch compensates poorly and room not in the center of the plate to print

Tam-Koowan, 2017-08-07:
I have a geeetech Prusa I3 pro B with a BLtouch, with the original version of the Marlin (1.0.2) nothing worked properly
So I'm going to go under "Marlin-bugfix-1.1.x_e16743d-PI3_Pro_B_3DTouch", since when I print my piece is not in the center of the board, but I also my nozzle is crushed on the left and too hasty on the Right of the plate;
I do not know how to solve this problem despite looking for forums.

Thank you for your help.

PS / if necessary I can provide you with photos or videos

[Marlin-bugfix-1.1.x_e16743d-PI3_Pro_B_3DTouch.zip](https://github.com/MarlinFirmware/Marlin/files/1205679/Marlin-bugfix-1.1.x_e16743d-PI3_Pro_B_3DTouch.zip)

Roxy-3D, 2017-08-07:
You can control where you piece is printed in your Slicer also.    If the reason you are trying to print something small in the center of the bed is because you are trying to bring up UBL, being offset a little is OK.

RalphSch, 2017-08-09:
Marlin 1.1.x now uses two separated coordinate systems, one for your machine (the real mechanical distance from the endstop to the end of the axis) and one for your workspace.
You can easily switch to the old behavior with NO_WORKSPACE_OFFSET in Configuration_adv.h. If you like to stay with the new system, then you have to travel to your workspace (bed) 0/0/0 (front, left corner, nozzle just touching the print surface) and hit "Set homeoffset" in the lcd menu.

@thinkyhead This is the worst issue when upgrading from 1.0.x to 1.1.x (beside of the by default disabled safe-z-homing which always leads to crashes when the sensor ist mounted in front or left of the nozzle). Allmost all people are crushing their nozzles into the bed and having big problems to get this fixed. Also the "Err: Too far" error when setting the homeoffset is quite confusing ... you have to search through the code to find this very important statement:
  /**
   * M428: Set home_offset based on the distance between the
   *       current_position and the nearest "reference point."
   *       If an axis is past center its endstop position
   *       is the reference-point. Otherwise it uses 0. This allows
   *       the Z offset to be set near the bed when using a max endstop.
   *
   *       M428 can't be used more than 2cm away from 0 or an endstop.
   *
   *       Use M206 to set these values directly.
   */

fiveangle, 2017-08-09:
@RalphSch - The coordinate systems are aligned unless the user changes them so why do you think home offsets is relevant here ?  

It seems haphazard to throw in using M206 to try and resolve what appears to be more likely a configuration/usage problem.

@Tam-Koowan - what is output of this command:

 `M503 S1`

RalphSch, 2017-08-10:
@fiveangle Because I help fixing exact this problem more than once every day in our Anet/Prusa FB group (very active, about 5k people) since we migrated the Skynet3D changes for Anet back to Marlin. Both coordinate systems are only aligned by default if your endstops are sitting at the 0 position and that's not given with the Anet printers and many others. This happens when you are upgrading, like Tom (his bed is 12mm away from the x min endstop), and you just grab your working settings from your old Configuration.h and put them in the new Configuration.h. Also you just have to hit the "set home offset" entry in the lcd menu by accident on your home-point (which normally isn't Z0) and your z-home-offset is wrong (this entry should be placed in the configuration section of the menu not in the prepare section next to homing and movement). The "hidden" 20mm max for the distance to 0 or the endstop is also an bad idea, e.g. Anet A8 has often more than this distance and because of that you can't just set your correct home offsets because it aborts with"Err: Too far!". 
If you just activate NO_WORKSPACE_OFFSET then your old and good working settings will work with 1.1.x as before .... that's what normal users prefer when upgrading so this would be the better default (and because the actual situation leads to crashes, bended rods and broken printer, like the disabled z_safe_homing default does when your sensor is located in the front or left of the nozzle).

I'm curios to see the M503 output ;)

Roxy-3D, 2017-08-10:
> If you just activate NO_WORKSPACE_OFFSET then your old and good working settings will work with 1.1.x as before .... that's what normal users prefer when upgrading so this would be the better default 

I think I agree this is a better default.   It would be helpful to hear from people that don't agree with this thinking why they see it differently.

Tam-Koowan, 2017-08-11:
12:59:36.557 : N30 M105*20
12:59:36.573 : echo:  G21 ; Units in mm
12:59:36.573 : echo:  M149 C ; Units in Celsius
12:59:36.573 : echo:Filament settings: Disabled
12:59:36.573 : echo:  M200 D3.00
12:59:36.573 : echo:  M200 D0
12:59:36.573 : echo:Steps per unit:
12:59:36.573 : echo:  M92 X80.24 Y80.41 Z2560.00 E97.00
12:59:36.573 : echo:Maximum feedrates (units/s):
12:59:36.573 : echo:  M203 X400.00 Y400.00 Z2.00 E45.00
12:59:36.573 : echo:Maximum Acceleration (units/s2):
12:59:36.573 : echo:  M201 X5000 Y5000 Z50 E5000
12:59:36.573 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
12:59:36.573 : echo:  M204 P2000.00 R1000.00 T1000.00
12:59:36.573 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
12:59:36.573 : echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
12:59:36.573 : echo:Home offset:
12:59:36.573 : echo:  M206 X0.00 Y0.00 Z0.00
12:59:36.573 : echo:Auto Bed Leveling:
12:59:36.573 : echo:  M420 S0 Z0.00
12:59:36.573 : echo:Material heatup parameters:
12:59:36.573 : echo:  M145 S0 H200 B60 F170
12:59:36.573 : M145 S1 H240 B110 F170
12:59:36.573 : echo:PID settings:
12:59:36.573 : echo:  M301 P12.33 I0.51 D74.50
12:59:36.573 : echo:  M304 P234.88 I42.79 D322.28
12:59:36.573 : echo:Z-Probe Offset (mm):
12:59:36.573 : echo:  M851 Z-3.38
12:59:39.599 : N31 M105*21

fiveangle, 2017-08-11:
Now that we've put that to rest…

I think we need to see a pic of the whole bot from front looking down the bed edge with the carriage close to the bed, and the video of the problem you mentioned.

Tam-Koowan, 2017-08-12:
[image]
This is my problem !  :(

fiveangle, 2017-08-12:
@Tam-Koowan - a few things: 

* You need to enable bed levelling every time you want to use it with M420 S1: is that being done ?
* Assuming you're doing that, what is output of `M420 V` to see the stored leveling data ?
* Assuming all that looks good, enable `Z_MIN_PROBE_REPEATABILITY_TEST`, compile, run `M48`, and show us the output

Do you have that picture from straight-on looking down the bed with the carriage close to it so we can see the relation of the cariage with the frame/bed ?

Tam-Koowan, 2017-08-12:
Cette photos à été prise depuis l'avant de l'imprimante. 
Je tester ça en rentant car je suis en week-end chez des amis

Roxy-3D, 2017-08-12:
It looks like you bed has a very significant tilt to it.   It would be wise to correct that as much as possible.

Tam-Koowan, 2017-08-13:
@Roxy-3D :Yes, this is the problem I have, my BLtouch does not compensate as it shoul,
my bed level is good.

Roxy-3D, 2017-08-14:
OK...   But this picture doesn't make it look like the physical bed leveling is good.  Was bed leveling turned on when this picture was taken?   https://github.com/MarlinFirmware/Marlin/issues/7452#issuecomment-321967990

Tam-Koowan, 2017-08-15:
@Roxy-3D : It is not systematically activated in the "Marlin-bugfix-1.1.x_e16743d-PI3_Pro_B_3DTouch"?
If not I do not know if it is!

Roxy-3D, 2017-08-15:
Look at your grid numbers.  You have the right side of the mesh 1mm higher than the left side.

Roxy-3D, 2017-08-15:
- Activate the bed leveling...
- lower the nozzle to 1.5mm or 2mm above the bed
- Make some big moves from left to right or front to back.  You should see the Z-Axis compensate if the system is alive and working.

Tam-Koowan, 2017-08-15:
I have to put the M420 S1 code in my boot code I guess, but I have to put it before the G29?

Roxy-3D, 2017-08-15:
After the G29.

Tam-Koowan, 2017-08-15:
Ok thank you, I test right away

Roxy-3D, 2017-08-15:
@fiveangle    I left a short message for you over at 3DPrintboard.com

Tam-Koowan, 2017-09-03:
Well I returned from vacation, and I wanted to re-download the bugfix marlin ... in my printer and the I have an error code, except that I did not touch the setting in the marlin.
Here's the code:  'INVERT_X_DIR' was not declared in this scope

You would not have a petrol on you and a match? :(

Roxy-3D, 2017-09-03:
It sounds like a ; got deleted somewhere in the configuration.h file...

Tam-Koowan, 2017-09-04:
Ok except that I did not touch the file, it is the same as I provided you in the Zip above ...
So I do not understand

Roxy-3D, 2017-09-04:
You deleted an important line of the configuration.h file:

[image]

Tam-Koowan, 2017-09-04:
Well, I managed to put the marlin back in my printer.
Except that I have nothing to this line before to deliver it

Roxy-3D, 2017-09-04:
Because Y is inverted...  It is more likely X is inverted than not.   Just pick a true or false value for
```cpp
#define INVERT_X_DIR true
```
You will know as soon as you try to move the X axis if it is moving in the right direction.

Tam-Koowan, 2017-09-04:
What method of bed leveling should I use?
for the best result.
My problem of centering of plateau will be solved because I find the solution on a French Forum ;)

Roxy-3D, 2017-09-04:
I wrote the UBL (Unified Bed Leveling) System.   I'm not sure I would be considered an objective source.

If you want my assistance with the bed leveling issues...  You need to be running UBL.

Tam-Koowan, 2017-09-04:
Very well, I would change the settings of my marlin tomorrow for tried.
thank you very much

Roxy-3D, 2017-09-04:
Make sure you can print a small object at the center of the bed before you turn on any bed leveling system.   Once you can do that,  turn these options on.  Probably the existing values are fine for you.
```cpp
#define AUTO_BED_LEVELING_UBL
 //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  #define UBL_MESH_INSET 1          // Mesh inset margin on print area
  #define UBL_MESH_NUM_X_POINTS 10  // Don't use more than 15 points per axis, implementation limited.
  #define UBL_MESH_NUM_Y_POINTS 10
  #define UBL_PROBE_PT_1_X 39       // These set the probe locations for when UBL does a 3-Point leveling
  #define UBL_PROBE_PT_1_Y 180      // of the mesh.
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20
  #define UBL_G26_MESH_EDITING    // Enable G26 mesh editing
```

Tam-Koowan, 2017-09-04:
After testing the compilation it puts this error:
The given UBL_PROBE_PT_1_Y can not be reached by the Z probe.

I have therefore modified:

  #define UBL_MESH_INSET 1 // Mesh inset margin on print area
   #define GRID_MAX_POINTS_X 10 // Do not use more than 15 points per axis, implementation limited.
   #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
   #define UBL_PROBE_PT_1_X 39 // UBL does a 3-Point leveling
   #define UBL_PROBE_PT_1_Y 130 // (Old to 180) of the mesh.
   #define UBL_PROBE_PT_2_X 39
   #define UBL_PROBE_PT_2_Y 20
   #define UBL_PROBE_PT_3_X 180
   #define UBL_PROBE_PT_3_Y 20
   #define UBL_G26_MESH_EDITING // Enable G26 mesh editing

Roxy-3D, 2017-09-04:
Yes.   And those values will probably never get used by you.  They are just there in case you want to do a 3-point tilting of the mesh.

Tam-Koowan, 2017-09-04:
But why did I get this error when I activated this function?

Roxy-3D, 2017-09-04:
Because you have a large offset from the nozzle to where the probe is located.   The nozzle can make it to that location but the probe can not go there.   So that Sanity Check code detected the problem and informed you about it so that it could get fixed.

Tam-Koowan, 2017-09-04:
I no longer have any error when compiling.
But when I run an impression, the sensor does not activate, as if I had the original confirmation?
You have an idea ?
Voici une vidéo de ce que donne l'impression

Roxy-3D, 2017-09-05:
Have you done a G26 Mesh Validation Pattern?   Can you print one and post the picture of it?   That will help us know what is out of adjustment.    It is hard to tell from the video but it looks like the nozzle is too far from the bed.   Also...  Please post a G29 W    It looks like UBL is not activated.   (You do that with G29 A)

Tam-Koowan, 2017-09-10:
After putting back my Offset Z I made several impressions with success despite the lag between the two sides. But after 4 good impression I got an error message.
I think my BLtouch is dead?
So I put back the Z's mini level sensor and I re-done the setting with the sheet, but I can not print correctly

closed 2018-02-06 by thinkyhead