MarlinFirmware/Marlin #7453: Heating Problem

alexliuwet72, 2017-08-07:
### Bug Report
- Description: when both nozzles and the bed heat up, the heaters on d10 and d9 suddenly turn off,and the screen dims for a while. Then, the screen flashes and the firmware restarts. It has nothing to do with the thermal runaway settings
- Expected behaviour: When both nozzles heat up the printer can stay on and do it's job
- Actual behaviour: It seems to shut off when the printer reaches it's temperature
- Steps to reproduce:
  -1. Go to prepare
  - 2.Go to preheat ABS
  -3 Select preheat ABS all
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1206054/Configuration.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

p3p, 2017-08-07:
I'm not saying this is the issue, but the screen dimming sounds like you're drawing too much current from the power supply, maybe triggering a protection mode, browning out the MCU and causing a reset. There is nothing Marlin firmware itself can do to dim an RRD GLCD screen.

alexliuwet72, 2017-08-08:
when the screen dims, heaters d10 and d9 turn off. D8 Stays on. D8's 11a mosfet is not used. I use a biqu hotbed mosfet

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead

alexliuwet72, 2018-06-05:
why close this. it is still happenind even when i used mosfets for everything

thinkyhead, 2018-06-07:
We close issues after a long period of inactivity, especially if they are hardware problems unrelated to the firmware. This keeps issues from piling up forever.

So first, try to get help from the community. If you have been unable to get helpful answers from other users and you still have an issue (and are pretty sure it's due to a bug in the firmware) then you should open a new issue, being sure to follow the updated and improved instructions for posting issues.