MarlinFirmware/Marlin #7456: "ENABLE_LEVELING_FADE_HEIGHT"  adds hight to printed object

moth4017, 2017-08-08:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
Marlin version RC8 1.14
enable "ENABLE_LEVELING_FADE_HEIGHT"
Setting fade height via start up script:-
G1 E-5 F20 ;(to retract at 90mm/s)1 
M502 ; load defaults from eprom
M500  ; save to eprom
M111 0 ;S247  Debug level
G28 ; home All
G29 ; probe bed  to level
M420 Z6 ;  fade height 
M851 Z1.1; 1.2 is closer to the bed than 1.1 
 ;do prime 
G1 X1.0 F2000
G1 Z0.1 F1000
G1 Y60.0 E4.0 F1000.0 ;prime
G1 Y100.0 E8.5 F1000.0; prime
G92 E0

If i print a object say 12mm tall  it will come out at 13.3 , if i comment  out the "M420 Z6 ;" the print comes out at 12mm

 ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

richgain, 2017-08-10:
I have the same issue with the 1.1.x Bugfix build from 8/8/17. I have just printed a whole batch of parts that should have been 15.8 mm tall, but are actually 16.6 mm tall and will not fit into the 16.2 mm slot designed for them.
On inspection,it is clear that the lowest 2 mm of every part has fatter layers than the upper part of the print. 
My startup script includes:
M420 S1
M420 Z2

richgain, 2017-08-10:
Sorry, I should have said "the lowest 10 layers of every part have fatter layers than the rest". 
Since I printed using 0.2 mm layer height, the first 10 layers *should* measure 2.0 mm, but actually measure more than 3 mm.

Roxy-3D, 2017-08-10:
Does your mesh have numbers that are primarily negative by 1.0 mm ?  That would explain it if that is the case.

richgain, 2017-08-10:
Hi Roxy
Yes,  that is almost exactly the case.
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:  G29 S3 X1 Y1 Z-0.82500
Recv: echo:  G29 S3 X2 Y1 Z-0.82500
Recv: echo:  G29 S3 X3 Y1 Z-0.80000
Recv: echo:  G29 S3 X1 Y2 Z-0.80000
Recv: echo:  G29 S3 X2 Y2 Z-0.72500
Recv: echo:  G29 S3 X3 Y2 Z-0.70000
Recv: echo:  G29 S3 X1 Y3 Z-0.87500
Recv: echo:  G29 S3 X2 Y3 Z-0.80000
Recv: echo:  G29 S3 X3 Y3 Z-0.65000

What does that mean? 
Is it a bug or a feature?
Thanks,
Richard

Roxy-3D, 2017-08-10:
The reason I asked is this...   The bed leveling system moves the nozzle up or down to track the bed based on the mesh point values.   If the bulk of your bed has large negative mesh point values, the nozzle is going to be moved down from the first layer height to track the bed and get good adhesion. 

The amount the nozzle moves down 'Fades' as you go up to 10 mm.    So...  The first 10mm of layers are trying to fill in the extra space.   And after 10mm of Z Height, no correction is being applied any more.

You need to get your Z_PROBE_OFFSET_FROM_EXTRUDER set more accurately.   Probably by about .8mm.    And then regenerate your mesh.     At that point, I think the problem will be gone...

moth4017, 2017-08-10:
Hi  Roxy 3D
i have a 6mm thick aluminium bed and this is the flattest  piece i could get in NZ

READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4      5      6      7      8
READ:  0 +0.395 +0.266 +0.132 -0.007 -0.152 -0.298 -0.463 -0.629 -0.807
READ:  1 +0.455 +0.339 +0.212 +0.080 -0.062 -0.207 -0.371 -0.544 -0.737
READ:  2 +0.503 +0.405 +0.292 +0.168 +0.033 -0.123 -0.281 -0.463 -0.659
READ:  3 +0.494 +0.415 +0.315 +0.206 +0.073 -0.078 -0.241 -0.433 -0.649
READ:  4 +0.424 +0.365 +0.276 +0.170 +0.043 -0.107 -0.279 -0.488 -0.730
READ:  5 +0.330 +0.282 +0.200 +0.099 -0.027 -0.166 -0.347 -0.564 -0.834
READ:  6 +0.203 +0.156 +0.077 -0.022 -0.138 -0.279 -0.461 -0.668 -0.926
READ:  7 +0.040 -0.008 -0.083 -0.173 -0.282 -0.419 -0.590 -0.796 -1.049
READ:  8 -0.165 -0.205 -0.279 -0.364 -0.468 -0.595 -0.753 -0.946 -1.180

moth4017, 2017-08-10:
HI Roxy

I have leveled the center of the print bed , see below
 Bilinear Leveling Grid:
READ:       0      1      2      3      4      5      6      7      8
READ:  0 -0.410 -0.357 -0.308 -0.261 -0.226 -0.188 -0.163 -0.139 -0.129
READ:  1 -0.353 -0.286 -0.228 -0.181 -0.137 -0.100 -0.078 -0.061 -0.063
READ:  2 -0.309 -0.223 -0.151 -0.092 -0.047 -0.013 +0.007 +0.016 +0.006
READ:  3 -0.320 -0.218 -0.131 -0.059 -0.006 +0.030 +0.046 +0.044 +0.015
READ:  4 -0.393 -0.266 -0.169 -0.093 -0.035 -0.002 +0.009 -0.009 -0.061
READ:  5 -0.488 -0.353 -0.251 -0.169 -0.105 -0.065 -0.058 -0.084 -0.161
READ:  6 -0.619 -0.478 -0.373 -0.286 -0.219 -0.174 -0.165 -0.185 -0.251
READ:  7 -0.776 -0.640 -0.530 -0.438 -0.363 -0.312 -0.290 -0.305 -0.368
READ:  8 -0.983 -0.840 -0.729 -0.629 -0.546 -0.485 -0.457 -0.460 -0.507
READ: 

and printed a 30 Dia x 21mm High, 
Actual  printed size  29.98 x 22.2

Roxy-3D, 2017-08-10:
Take a look at the first 10mm of the print.  Are the layers a uniform thickness compared to the rest of the print?   If so...  You may need to adjust your Z-Steps per mm number to get things perfect.

In the picture up above, you could actually see a difference between the layers in the first 10mm of the print.

moth4017, 2017-08-11:
My z Height fade is at 6mm and you can just about see a slight difference , but why  would the  print be 1.2 mm Taller when the bed off set is only Min -0.006 and max 0.169 offset. it is very consistent on the  added amount of  1.2mm . when i comment out the fade height in my start up script the print height and actual height are with in 0.2 mm

Roxy-3D, 2017-08-11:
I am not an expert on Bi-Linear leveling....   If this was happening within the UBL Bed Leveling system I could find an answer for you.

moth4017, 2017-08-11:
Hi Roxy.. i tried UBL and got into a real mess..   my print head kept rising to +20 when it got close to the edge of the build plate, im guessing because  its in the area which can not be probed so went back to Bilinear.

Roxy-3D, 2017-08-11:
Did you do a G29 T after the automated mesh generation?    Did you fill in the dot's ?     And are you using the bugfix-1.1.x that is current?

moth4017, 2017-08-11:
HI Roxy
Re G29 T  and current firmware , 

is this for UBL?  if so yes to the current bug fix and no to the " fill or the dots"

Who is the expert on Bi Linear?

Cheers Mark

moth4017, 2017-08-20:
Who is the expert on Bi Linear?

Roxy-3D, 2017-08-20:
> Who is the expert on Bi Linear?

https://github.com/MarlinFirmware/Marlin/blame/bugfix-1.1.x/Marlin/Marlin_main.cpp#L2880-L2884

echo465, 2017-09-11:
Arggggg I think this behavior explains what I'm seeing with UBL+Fading.  I've been pulling my hair out treating it like a motor driver issue.  First several layers are stretched, circles become ovals, etc.

Configuration.h:
`#define Z_PROBE_OFFSET_FROM_EXTRUDER 0 `
Printer console:
```
Send: G29 P5
Recv: # of samples: 49
Recv: Mean Mesh Height: -2.899775
Recv: Standard Deviation: 0.113639
Recv: ok
Send: M503
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Unified Bed Leveling:
Recv: echo:  M420 S1 Z4.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active.
Recv: 
Recv: Active Mesh Slot: 1
Recv: z_offset: 0.000000
Recv: EEPROM can hold 17 meshes.
Recv: 
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.95
```
So.. how to correct this?  I could set Z_PROBE_OFFSET_FROM_EXTRUDER to the mesh mean height, or are there other ways to do it?

Roxy-3D, 2017-09-11:
Post a G29 W so we can see what the UBL system thinks the world looks like.

echo465, 2017-09-11:
G29 W:
```
Send: G29 W
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 4.0000
Recv: zprobe_zoffset: -0.9500000
Recv: UBL_MESH_MIN_X  10=10
Recv: UBL_MESH_MIN_Y  10=10
Recv: UBL_MESH_MAX_X  (300 - (10))=290
Recv: UBL_MESH_MAX_Y  (300 - (10))=290
Recv: GRID_MAX_POINTS_X  7
Recv: GRID_MAX_POINTS_Y  7
Recv: MESH_X_DIST  46.67
Recv: MESH_Y_DIST  46.67
Recv: X-Axis Mesh Points at: 10.000  56.667  103.333  150.000  196.667  243.333  290.000
Recv: Y-Axis Mesh Points at: 10.000  56.667  103.333  150.000  196.667  243.333  290.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: ubl_state_at_invocation :0
Recv: 
Recv: ubl_state_recursion_chk :0
Recv: 
Recv: Meshes go from 0x0270 to 0x0F7F
Recv: sizeof(ubl) :  1
Recv: 
Recv: z_value[][] size: 196
Recv: 
Recv: EEPROM free for UBL: 0x0D0F
Recv: EEPROM can hold 17 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: ok
```
G29 T:
```
Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (10,290)                                        (290,290)
Recv:  -2.719   -2.844   -2.977   -3.044   -3.029   -2.994   -2.882
Recv: 
Recv:  -2.607   -2.773   -2.912   -3.000   -2.987   -2.934   -2.835
Recv: 
Recv:  -2.606   -2.761   -2.904   -2.984   -2.973   -2.920   -2.811
Recv: 
Recv:  -2.660   -2.820   -2.952   -3.016   -2.981   -2.914   -2.794
Recv: 
Recv:  -2.781   -2.916   -3.015  [-3.042]  -2.998   -2.937   -2.797
Recv: 
Recv:  -2.781   -2.916   -3.015   -3.042   -2.998   -2.937   -2.797
Recv: 
Recv:  -2.781   -2.916   -3.015   -3.042   -2.998   -2.937   -2.797
Recv: (10,10)                                            (290,10)
Recv: (0,0)                                             (6,0)
Recv: ok
```

Roxy-3D, 2017-09-11:
Turn off any UBL in Configuration.h.   Rebuild the firmware.   See if you can print a small object at the center of your bed.   Those -3 mm mesh numbers are a VERY big concern.

echo465, 2017-09-11:
Airprint.  Roughly 3.8mm above the bed.

Roxy-3D, 2017-09-11:
> Airprint. Roughly 3.8mm above the bed.

Yes.   Please get your settings correct so you can print a small object without any bed leveling active.  Once that can be done, it won't be difficult to turn on UBL and get it setup.

JerryDurand, 2017-09-11:
Just as a side note, when I'm leveling a table and I don't know how far off it may be, I use a technique I use on milling machines...  Get a reference object, I use a lathe cutting tool (1/4" or 3/8" precisely square bar) and set that between the head and the table at each leveling step.  Now there's no way to crash the head (worse on a milling machine) into the table when moving to a new location.  Once it's all leveled, then just set the overall height lower so you're at the table surface.

echo465, 2017-09-11:
`#define Z_MIN_POS -3`
I had changed it ages ago under the mistaken idea that the Z-offset wouldn't work unless it was above Z_MIN_POS.  I wouldn't have noticed before because whatever bed leveling method would have adjusted for it and fading wasn't a thing.

No leveling:
[image]

after changing Z_MIN_POS back to 0 and re-probing:
```
Send: G29 P5
Recv: # of samples: 49
Recv: Mean Mesh Height: 0.086475
Recv: Standard Deviation: 0.116922
Recv: ok
```
[image]

G29 W:
```
Send: G29 W
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 4.0000
Recv: zprobe_zoffset: -0.9500000
Recv: UBL_MESH_MIN_X  10=10
Recv: UBL_MESH_MIN_Y  10=10
Recv: UBL_MESH_MAX_X  (300 - (10))=290
Recv: UBL_MESH_MAX_Y  (300 - (10))=290
Recv: GRID_MAX_POINTS_X  7
Recv: GRID_MAX_POINTS_Y  7
Recv: MESH_X_DIST  46.67
Recv: MESH_Y_DIST  46.67
Recv: X-Axis Mesh Points at: 10.000  56.667  103.333  150.000  196.667  243.333  290.000
Recv: Y-Axis Mesh Points at: 10.000  56.667  103.333  150.000  196.667  243.333  290.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: ubl_state_at_invocation :0
Recv: 
Recv: ubl_state_recursion_chk :0
Recv: 
Recv: Meshes go from 0x0270 to 0x0F7F
Recv: sizeof(ubl) :  1
Recv: 
Recv: z_value[][] size: 196
Recv: 
Recv: EEPROM free for UBL: 0x0D0F
Recv: EEPROM can hold 17 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: ok
```

G29 T:
```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (10,290)                                        (290,290)
Recv:   0.026   -0.041   -0.097  [-0.102]  -0.024    0.075    0.257
Recv: 
Recv:   0.171    0.079    0.008   -0.022    0.028    0.134    0.306
Recv: 
Recv:   0.228    0.114    0.028    0.003    0.060    0.153    0.320
Recv: 
Recv:   0.173    0.088    0.011   -0.019    0.065    0.176    0.325
Recv: 
Recv:   0.105    0.020   -0.048   -0.031    0.046    0.164    0.318
Recv: 
Recv:   0.105    0.020   -0.048   -0.031    0.046    0.164    0.318
Recv: 
Recv:   0.105    0.020   -0.048   -0.031    0.046    0.164    0.318
Recv: (10,10)                                            (290,10)
Recv: (0,0)                                             (6,0)
Recv: ok
```
Thanks for your help.

Roxy-3D, 2017-09-11:
If you do a G26 and post a picture...   You will be able to fine tune the mesh.  You should be able to get good 100% adhesion across 100% of the bed 100% of the time.

closed 2018-02-09 by thinkyhead

leytanelectronics2, 2019-06-04:
i have been trying to find more information on this fade height thing but can't find any video's on youtube about it or any meaningful information off google, is there a thing here that explains it?

CarlosGS, 2020-03-29:
This feature is very nice since it prevents distortion of the whole parts, adjusting only the first layers and converging to the original design.

However, the fade height is [initialized as 10mm](https://github.com/MarlinFirmware/Marlin/blob/659316f797e086d84ef94c5aa407f618d6e81600/Marlin/src/module/configuration_store.cpp#L624), which should be OK, **BUT it is then [overwritten with 0mm](https://github.com/MarlinFirmware/Marlin/blob/659316f797e086d84ef94c5aa407f618d6e81600/Marlin/src/module/configuration_store.cpp#L2474). By mistake the default setting is disabled.**

In my machine I have set 1.5mm. This way the correction only affects the first layers. Other users of leveling will notice that by default there is Z correction thorough the whole print every time.

Thanks for this software, I hope you are all safe & well!!