MarlinFirmware/Marlin #7458: latest bugfix-1.1.x: Recv: ? in get_z_correction

040medien, 2017-08-08:
When running G29 J or G29 J2 (or J3, J4, etc.), I'm getting these error messages:

Recv: ? in get_z_correction(lx0=10.00, ly0=302.00)
Recv: ? in get_z_correction(lx0=197.00, ly0=302.00)
Recv: ? in get_z_correction(lx0=197.00, ly0=42.00)
Recv: ok
Recv: echo:endstops hit:  Z:0.69

Afterwards when I run G29 T1:

Send: G29 T1
Recv: Invalid Y location specified.

Roxy-3D, 2017-08-08:
The code in ubl_motion.cpp is complaining that you are positioning the nozzle outside of the area that the mesh knows how to correct.   As a result of this...   No correction for that move can be generated.

If you recompile the firmware with an appropriate bed size (and re-generate the mesh) you can make this problem go away.

> Send: G29 T1
> Recv: Invalid Y location specified.

This may be a bug.   A lot of logic was forced into ternary operators and Y isn't initialized all the time (any more as it should be!!!!).   This needs to be changed back to the way it was.   But resetting the machine and doing a G29 T will get you the current map.

040medien, 2017-08-08:
Thanks. The probe points were directly on the outside perimeters of the mesh (in areas I can probe). It might make sense to document that it needs to go have an additional inset from the mesh borders.

oysteinkrog, 2017-08-08:
@Roxy-3D: I agree with you here, it's obvious that you are frustrated and I think you have a right to be.
In this case I think @thinkyhead went too far, there is no good reason to prefer ternary operators for this code.

fiveangle, 2017-08-08:
While the point can be easily made in this example that ternary operators should be used where they improve code readibility and not used where they reduce it, there are surely more constructive ways to convey this than a passive-agressive sarcastic post to a 3rd party who has no power to directly effect change (imho).

040medien, 2017-08-21:
Just wanted to bring this up again. This issue persists, rendering G29 J useless.

Roxy-3D, 2017-08-21:
Can you post your Configuration.h and Configuration_adv.h ?   (add a .txt suffix to the name)

It looks like you have stuff set up such that the nozzle is going off the mesh.

040medien, 2017-08-21:
Sure thing:

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1239163/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1239162/Configuration_adv.h.txt)

Roxy-3D, 2017-08-21:
In Configuration.h you have:
```cpp
// The size of the print bed
#define X_BED_SIZE 207
#define Y_BED_SIZE 312

#define UBL_MESH_INSET 10          // Mesh inset margin on print area
```

In Configuration_adv.h you have:
```cpp
  #define UBL_MESH_MIN_X UBL_MESH_INSET
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y UBL_MESH_INSET
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
```

  #define UBL_MESH_MIN_X 10            = 10
  #define UBL_MESH_MAX_X (207 - 10)    = 197
  #define UBL_MESH_MIN_Y UBL_MESH_INSET= 10
  #define UBL_MESH_MAX_Y (312 - 10)    = 302

Recv: ? in get_z_correction(lx0=10.00, ly0=302.00)
Recv: ? in get_z_correction(lx0=197.00, ly0=302.00)
Recv: ? in get_z_correction(lx0=197.00, ly0=42.00)

The locations it is complaining about are right on the MAX_X_POS and MAX_Y_POS locations.  If you adjust the inset to be 9 instead of 10, you will be able to go to X=197 or Y=302 

Alternatively...  You could make your X_BEDSIZE and Y_BEDSIZE slightly bigger.

040medien, 2017-08-22:
I'm not following

G29 J should use the three points specified here, which are all well within the mesh dimensions:
`  #define UBL_PROBE_PT_1_X 20       // These set the probe locations for when UBL does a 3-Point leveling
  #define UBL_PROBE_PT_1_Y 292      // of the mesh.
  #define UBL_PROBE_PT_2_X 187
  #define UBL_PROBE_PT_2_Y 292
  #define UBL_PROBE_PT_3_X 187
  #define UBL_PROBE_PT_3_Y 52`

closed 2017-11-14 by thinkyhead