MarlinFirmware/Marlin #7460: software endstops ignore Z offset

Chaos99, 2017-08-08:
### software endstops ignore Z offset
- Description: My z-probe triggers a few mm below the nozzle. I did a normal `G28` homing with the probe. then I've measured the distance and configured it with `M206 Z3.35` as a home offset.

- Expected behavior: Coordinate system should be shifted. `G0 Z0` shoud bring the nozzle right down over the bed.
- Actual behavior: `G0 Z0` stops with 3.35mm below the nozzle. Repetier Host shows _0.00_ as current z position. `M114` reports _3.35_ as current z position. Another `G0 Z0` does nothing. Starting a print will try to print the first layer into thin air. 
Disabling the software endstops with `M112 S0` and repeating `G0 Z0` will actually move the nozzle down over the bed. Both Repetier Host and `M114` now show z position as _0.00_. A print starts fine.
Seems like soft endstops work on the original coordinate system without taking the home offset into account.

- Steps to reproduce:
  - (M5002 + M500)
  - G28
  - M206 Z3.35
  - G0 Z0
  - M211 S0
  - G0 Z0  

Is this a feature or a bug? Am I doing it wrong? 
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/1207246/conf.zip)

jon012198, 2017-08-08:
try a -z probe that is how mine is Z-1.20 like that

RalphSch, 2017-08-09:
This "negative shifting" isn't very intuitive ;)

M206 - Set Home Offsets
....
This effectively shifts the coordinate space in the negative direction. See examples below.
....
Example:
Raise Z up a little bit on the first layer: M206 Z-0.2

Chaos99, 2017-08-09:
Not sure that I understand you correctly. You mean a negative value for the home offset?
I tried that. Worked as expected. Moved the nozzle away from the bed.
Or do you mean a different parameter? Maybe Z_PROBE_OFFSET_FROM_EXTRUDER?
I've also set this to a negative value, but not the full -3.35. I'll try to correct this and test if it makes a difference. I though this parameter only is used during bed leveling.

fiveangle, 2017-08-09:
With MBL you need to run levelling procedure in order to be able to print, otherwise your nozzle will never be close to the bed.  Do yourself a favor and disable workspace offsets entirely and don't mess with it.  It isn't intended to be used for what you're trying to do, and just complicates things further.

fiveangle, 2017-08-09:
I notice you don't have any LCD - that makes MBL levelling procedure tedious.  If you decide to load octoprint (on a mac, linux, or raspberry pi host) you can use this custom controls to easily run through the procedure using the octoprint interface: https://github.com/foosel/OctoPrint/issues/1682

Chaos99, 2017-08-10:
The MBL is just activated for experimentation. I intend to have an automated bed leveling and found out to late that both MBL and UBL need manual input (which is cumbersome without a display).
I'll remove it at see if it makes any difference for the problem stated above.

@fiveangle Saying not to use workspace offset begs the question: what to use instead to define the distance between the nozzle and the trigger point of the z probe?

fiveangle, 2017-08-11:
*_PROBE_OFFSET_FROM_EXTRUDER + M851 (once you set up ABL or UBL)

Chaos99, 2017-08-13:
Changed Configuration.h to use bilinear bed leveling, set PROBE_OFFSET_FROM_EXTRUDER to -3.25. Uploaded, reset to defaults with M502+M500+M501.
G28 does homing and correctly touches teh nozzle to the ground.
But doing a G0 Z0 directly after will crash my nozzle down trying to go 3.25mm into the build platform.
(G29 also does a proper probing run. But again, afterwards a G0 Z0 crashes the print head.)
Home offset is 0.0 now , as reported by M503.

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead