MarlinFirmware/Marlin #7465: [FR] how to enable Z goes down a little distance after the printing finish?

hackqiang, 2017-08-09:
### Feature Request
As I remember, in the old version of Marlin(0.9), the Z will go down for a little distance after the printing finish.
But in v1.1, the Z won't move after the printing finish. So the Nozzle and the model are connected.
Is there any configure items for enable this feature?
( my machine is CoreXY)
thanks!

klcjr89, 2017-08-09:
Just change your ending script to move the axes where you want them. Or use the nozzle park feature gcode, G27 after enabling it in the firmware. :)

RalphSch, 2017-08-09:
The behavior after finishing the print job is part of the gcodes which are generated by the slicer.

hackqiang, 2017-08-09:
I see the G27 docs: http://marlinfw.org/docs/gcode/G027.html
but I still confused: in the config:
`P0  (Default) If Z is below park Z raise the nozzle.`
`#define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }`
what does the "20" meaning?
If I add the `G27` in my .gcode file, after finish the printing, Z will go down for 20mm?

FHeilmann, 2017-08-09:
the 20 here means if the nozzle is below 20mm it will be raised to 20mm, otherwise the nozzle will simply move away.

If you don't want to use G27 simply paste:

```
G91
G0 Z+5
G90
```
at the end of your g-code. This will raise the nozzle by 5mm, no matter which Z it is currently at.

fiveangle, 2017-08-09:
No. G27 is used to park the nozzle, not move nozzle above a printed model.  The answers above are correct: set the after-print movement actions in your slicer (e.g. G1 Z[max position size]).  If you G27 immediately after a print with Z park at "20", it will move the nozzle down to 20mm off the bed and potentially crush your already printed piece.

teemuatlut, 2017-08-09:
By default G27 will not crush your print as it will only raise the nozzle. You can force the Z destination with P1 parameter however.

OP can use G27 P2 to raise the nozzle (or in his case lower the bed) and park the nozzle at point XY.

[[doc]](http://marlinfw.org/docs/gcode/G027.html)

fiveangle, 2017-08-09:
"lower the bed" (!)

hackqiang, 2017-08-11:
thanks all, I will enable the G27 in firmware and add below G code in the end of .gcode:
`G27 P2`

closed 2017-08-11 by hackqiang