MarlinFirmware/Marlin #7466: ADVANCED PAUSE  compilation error

michu556, 2017-08-09:
Hello i have compilation error

Arduino:1.6.7 (Windows 7), Płytka:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:39:0,

                 from E:\MEGAsync\Marlin-RC\Marlin-1.1.x.1\Marlin\Marlin.ino:31:

SanityCheck.h:323: error: #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."

     #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."

      ^

exit status 1
#error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."

This is pure marlin 1.1.4 with option enabled

Configuration.h
#define FILAMENT_RUNOUT_SENSOR
Configuration_adv.h
#define ADVANCED_PAUSE_FEATURE

  help

FHeilmann, 2017-08-09:
The error message gives you all the information you need about this error:

`ADVANCED_PAUSE_FEAUTRE` can only be used with an LCD display 

so you need to enable one in the config or disable `ADVANCED_PAUSE_FEATURE`

michu556, 2017-08-09:
LCD is on.  Does it not work with LCD 2004?
#define DISPLAY_CHARSET_HD44780 JAPANESE
#define ULTRA_LCD   // Character based
//#define DOGLCD      // Full graphics display
#define SDSUPPORT

fiveangle, 2017-08-09:
I think we need to take ULTRA_LCD and DOGLCD out of Configuration.h because this keeps coming up.

```
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #if DISABLED(NEWPANEL)
    #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."
```

NEWPANEL is supposed to be set by controllers with a display.  LCD 2004 is reprap smart controller not ULTRA_LCD.

FHeilmann, 2017-08-09:
You also need a controller type, not just an LCD type. As @fiveangle uncomment `#define REPRAP_DISCOUNT_SMART_CONTROLLER` if you have a 2-line character LCD or `#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` if you have a full graphic lcd

Roxy-3D, 2017-08-09:
> uncomment #define REPRAP_DISCOUNT_SMART_CONTROLLER if you have a 2-line character LCD or

I have a 20x4 LCD on my Folger Tech.   This works for that size also.

michu556, 2017-08-09:
Thank you very much for the prompt and reliable help. Everything works ;)
And i for two // i lost 8 hours hehe

Left to do Rotary encoder on displays with dynamic speed ;)

fiveangle, 2017-08-12:
please close this if it worked for you

closed 2017-08-12 by michu556

michu556, 2019-01-10:
> Hi friends
> I have exactly the same problem
> #error "ADVANCED_PAUSE_FEATURE currently requires an LCD controller."
> I recently installed a filament runout sensor and I made the settings that
> were needed
> I use the MKS tft 2.8 display and MKS Gen 1.4
> @fiveangle <https://github.com/fiveangle>
> @Roxy-3D <https://github.com/Roxy-3D>
>
> http://s9.picofile.com/file/8348489734/Configuration.h.html <http://url>
> Configuration.h
> http://s8.picofile.com/file/8348489784/Configuration_adv.h.html
> <http://url> Configuration_adv
>
> Please help me dears
> Thanks...

Hi
this error is caused by not selecting the screen in the marlin
settings. you must delete 2 //  appropriate screen in the settings
Regard
M.

rouhit110, 2019-01-10:
Thanks dear @michu556 but i uncomment and select #define DOGLCD in configuration.h. 
I did not notice closing the thread. Excuse me
I created a new topic to solve this problem