MarlinFirmware/Marlin #7468: LCD displays SD card menu's incorrect

3DFreezeMe, 2017-08-09:
### **LCD displays SD card menu's incorrect**

3D Printer: Creality Ender-2
Configured Board: BOARD_MELZI_CREALITY
Configured Graphical LCD: MINIPANEL
Running on latest Marlin-bugfix-1.1.x firmware
Link to zip: https://github.com/3DFreezeMe/Marlin/archive/bugfix-1.1.x.zip

**OK:**
- LCD display
- controller functionality
- 'SD card ok' when '//#define SHOW_CUSTOM_BOOTSCREEN' is set in Configuration.h

**NOK:**
- 'SD init fail' when '#define SHOW_CUSTOM_BOOTSCREEN' is set in Configuration.h
- parsing of the 'Print from SD' menu; black stripes alternating with text
- parsing of 'Change SD card', ' Init. SD card' and 'Refresh' options do create a dark bar, but they go away when scrolling

**Links to screenshots of the LCD display:**
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/1.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/2.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/3.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/4.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/5.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/6.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/7.jpg

**(possible) related Marlin Github issues:**
5342
5335

**Tried following suggestions found in Github Marlin issues:**

5335

**1.** tried to add the following lines mentioned in this issue to the ultralcd.cpp (on a couple of places)

```
  screen_changed = true;
  lcdDrawUpdate = LCDVIEW_CLEAR_CALL_REDRAW;
```
	
**-->** although I did have some result in an earlier attempt, now on the bugfix-1.1.x firmware I do not get any results.
I read that it only concerns character LCD screens anyway.

fiveangle, 2017-08-09:
https://github.com/MarlinFirmware/Marlin/issues/7467#issuecomment-321286962

3DFreezeMe, 2017-08-09:
Tested with the delays added to my configuration.h, nut no change.
```
// MakerLab Mini Panel with graphic
// controller and SD support - http://reprap.org/wiki/Mini_panel
//
#define MINIPANEL // 3DFreezeMe
#define ST7920_DELAY_1 DELAY_5_NOP // 3DFreezeMe
#define ST7920_DELAY_2 DELAY_5_NOP // 3DFreezeMe
#define ST7920_DELAY_3 DELAY_5_NOP // 3DFreezeMe
```
The SD card is btw not on the LCD panel, it is on the Creality Melzi board itself. 
I have added some photo's of the board and LCD panel. We believe it is a minipanel (or clone), as that LCD panel is the only one in the Configuration.h giving any good result.

https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/8.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/9.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/10.jpg
https://github.com/3DFreezeMe/Marlin/blob/bugfix-1.1.x/11.jpg

3DFreezeMe, 2017-08-27:
Currently all linked files are broken, I am working on that.

thinkyhead, 2017-08-28:
You can also try the option…
```cpp
  #define DOGM_SPI_DELAY_US 6
```
Make it longer or shorter. The original value from Marlin 1.0 was 10.

Svenson74, 2017-09-22:
I have the same issue here and I assume that it has s.th. to do with the SPI communication. Whenever there is sdcard activity happen (while printing from sd card e.g.), these "black bars" appear on the screen.

edulabrennes, 2017-09-28:
Is someone is able to try this firmware on another 12864 LCD ? Perhaps Creality give us a special version of this screen.

thinkyhead, 2017-11-15:
Please re-open if this issue persists.

closed 2017-11-15 by thinkyhead

TheMasterFX, 2017-11-23:
Thats not an general issue with the Minipanel. The issue is that you've used the creality CR-10 config as base. For the Minipanel you have to use DOGLCD_A0 and DOGLCD_CS.
And because the DOGLCD_CS is on an other pin than on SANGUINOLOLU you just have to change the pins_MELZI_CREALITY.h

```
//...
#undef DOGLCD_CS    

#define LCD_SDSS        31 // Smart Controller SD card reader (rather than the Melzi)
#define DOGLCD_CS     28 // st9720 CS
//...
```

tann200, 2017-12-03:
Hi, 
Do I understand correctly, that the the lines must be added to pins_MELZI_CREALITY.h and in Configuration.h  #define MINIPANEL has to be enabled?
Is there anything else that needs to be done?

thinkyhead, 2017-12-05:
@TheMasterFX Would you say this is a good general solution, adding the following to `pins_MELZI_CREALITY.h`?:
```diff
 // Alter timing for graphical display
 #define ST7920_DELAY_1 DELAY_2_NOP
 #define ST7920_DELAY_2 DELAY_2_NOP
 #define ST7920_DELAY_3 DELAY_2_NOP

+#if ENABLED(MINIPANEL)
+  #undef DOGLCD_CS    
+  #define DOGLCD_CS        LCD_PINS_RS
+#endif
+
 /**
```

3DFreezeMe, 2017-12-05:
On the Facebook Group 'Creality Ender-2 3d Printer User Group', user Timmo Hellemann posted the solution he found:

Timmo Hellemann is feeling fantastic.
November 19 at 4:16pm

Finally I get rid of the black bars with custom Marlin 1.1.6
Here is how to do it:

Use CR-10 config and comment out
//#define CR10_STOCKDISPLAY
And uncomment
#define MINIPANEL
Don't forget to change printer size etc.!
Edit pins_MELZI_CREALITY.h
Comment out the LCD Stuff and add DOGLCD:
#undef DOGLCD_CS
#define LCD_SDSS 31 // Smart Controller SD card reader (rather than the Melzi)
#define DOGLCD_CS 28 // DOG CS
//#define DOGLCD_A0 30 // DOG A0 is already defined in pins_SANGUINOLOLU_11.h
//#define LCD_PINS_RS 28 // st9720 CS
//#define LCD_PINS_ENABLE 17 // st9720 DAT
//#define LCD_PINS_D4 30 // st9720 CLK
#define FIL_RUNOUT_PIN -1 // Uses Beeper/LED Pin Pulled to GND

TheMasterFX, 2017-12-09:
> On the Facebook Group 'Creality Ender-2 3d Printer User Group', user Timmo Hellemann posted the solution he found:
Yeah thats me ;-)