MarlinFirmware/Marlin #7469: UBL : Z-position is lost with toolchanges (T0/T1) when UBL is active (release 1.4).

joris57, 2017-08-09:
[topo.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1211940/topo.gcode.txt)
[test2.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1211939/test2.gcode.txt)

- Description: Under certain bed topology, Marlin seems to lose the Z-position after toolchanges (T0/T1) when UBL is active (release 1.4).

The attached G-code, which consists basically only of z moves and toolhead changes, physically does NOT leave z in the end at 10.0 as expected (and as indicated by M114), but at 6.xx (as measured).

When the toolhead changes are removed or when UBL is deactivated, behaviour is as expected.
 
- Expected behaviour: after executing the attached script, the physical z position is expected to be 10.0.

- Actual behaviour: after executing the attached script, M114 reports the z position to be 10.0. In reality, however, z is physically somewhere at 6.xx

- Steps to reproduce:
  - Load UBL data (topo.gcode) 

attached.
  - Execute 'test2.gcode'

Roxy-3D, 2017-08-09:
I'm not sure how to proceed.  I don't have any printers with multiple nozzles.   

I wonder if there are any Marlin users that both know C code and have a dual extruder machine that can help figure this out?

fiveangle, 2017-08-09:
Does same happen if you change this to false ?

#define DISABLE_INACTIVE_E true

fiveangle, 2017-08-09:
also, I know there have been some fixes in this area (like #7315) can you please try latest bugfix1.1.x and confirm you still have the issue ?

joris57, 2017-08-09:
@fiveangle changing DISABLE_INACTIVE_E to false does not solve the problem. I already tried the current bugfix1.1.x earlier today to no avail. 

@Roxy-3D I don't think you actually need the 2nd hotend physically installed to reproduce this problem. The only thing the T1 command does in this case is to 'shift' all subsequent positions by the hot-end offset (in this case -18mm in the x-axis) and route the extrusion instructions to the relevant extruder.

What I noticed is that, although this shift means that the 2nd hotend moves to the SAME location relative to the bed (and hence no further UBL correction should be applied), the Z axis still moves. Other strange thing is that when re-enabling the first hotend, another correction is made to the z height, but IN THE SAME DIRECTION, effectively further increasing the error. Also, the magnitude of the correction seems not to be proportional when compared to the correction applied with a regular move of the same distance.

joris57, 2017-08-10:
I think I boiled this down to it's most basic reproduction:

1) Set the offset of the 2nd hot-end to (0, 0). This means that the two hot-ends 'coincide' and that T0 or T1 commands do NOT result in a movement in the XY plane.

2) Invalidate the UBL grid and fill the grid homogeneously with 0.1. Activate UBL
```
G29 I 999
G29 P3 C0.1 R999
G29 A
```
3) Position the head at Z=3.0
```
G28
G1 Z3.0 F300.0
```
4) Execute a number of toolchanges :
```
T0
T1
T0
T1
T0
T1
T0
T1
T0
T1
T0
T1
...
```

Expected behaviour : no movement whatsoever.

Observed behaviour : nozzle moves up in Z (but no change in M114 reported z position).

Note:
- When changing the grid fill value to -0.1 : nozzle moves down in Z (but no change in M114 z position).
- Steps fade when approaching Z=10 (as reported by M114).

Since, in this example there is no movement in any direction, implicit or explicit, it seems likely that the problem lies in the tool change code.

FHeilmann, 2017-08-10:
Seems like the correction is applied on every toolchange

oysteinkrog, 2017-08-10:
This is interesting, I do have a dual carriage machine that seems to work OK with UBL.
I have tool offsets on my Z though, maybe that helps? (e.g. M118 T1 X250, Y3, Z-0.6)

joris57, 2017-08-10:
@oysteinkrog are you saying that when you do as described above (load the UBL grid with .1 and then execute a number of toolchanges > 20), you don't see an increase in the Z position (at Z sufficiently low)?

thinkyhead, 2017-11-27:
Fixed in the current `bugfix-1.1.x` branch?

joris57, 2017-12-02:
Printer is currently being converted back to single extruder/nozzle, so not able to test. Closing for he time being. Will re-open if/when I'm able to test.

closed 2017-12-02 by joris57