MarlinFirmware/Marlin #7474: Automatic Z-probe offset / Paper test

tafadafi, 2017-08-10:
Hi all,

I have a "vision" in my mind and still struggling how to set it up ... maybe someone has a good idea on that.

I'm using the following setup:

Anet A8 - Firmware: Marlin 1.1.4
Autoleveling with BLtouch Sensor
Octoprint on Raspberry pi 3B

The Z-Probe offset is changing from print to print by max. 0.1 mm and to get accurate results I perform the famous paper test before each print.

The idea is to use the nozzle as a probe :-)

I installed a aluminum stripe on one edge of the bed and measure via a multimeter the continuity while going down (0.02 steps) on the aluminum stripe.

The other end is connected to the extruder (I just used one of the fan screws).

Once I get continuity I can easily calculate the z-probe offset in send it via M851 Z-x.xx command followed by M500.

This works quite well and I have a consistent offset now.

Now it comes to the automatic approach .... the idea is to use the gpio 3.3 V output (if possible) from the Pi and connect it to one point. The second point is connect to a GND pin. I command the Z-axis down until I get continuity, read out the Z value, do the math and send the M851 to correct the offset.

So - my question ... is it possible to use one of the GPIOs for such a setup or do I have to use something else.

Second - is it possible to have a kind of loop. Go down by 0.02 mm, check continuity, NO -> go down by 0.02 mm and check again; YES -> read out Z value, do the math, write M851 blabla

Third - how can I get the actual Z position to do the math automatic on that.

Many thanks for your help.

Cheers!

Tafadafi

FHeilmann, 2017-08-10:
Check out BABYSTEPPING and DOUBLECLICK_FOR_Z_BABYSTEPPING in Configuration_adv.h

What it does is, you start your print (which ideally has a skirt in the beginning) and as soon as the print starts you double click your wheel on the display. This takes you into a menu where you can adjust the Z-height in very small steps until the first layer is perfect (ideally you are still printing your skirt while you are adjusting)

If you also uncomment BABYSTEP_ZPROBE_OFFSET the change you make will also be used to update your Z-Offset, so your next print will already take this correction into account.

oysteinkrog, 2017-08-10:
Just hook up your nozzle as the switch as you describe, do probing and use the probe z offset to add however much offset you want.
Don't use the rpi for anything, use the actual controller board.

benlye, 2017-08-10:
@tafadafi I used to have the same problem with my Anet A8, until I realized that I needed to give the hotend around 5 minutes to fully warm up so that the thermal expansion could settle down.

Try doing your paper test and setting your Z offset after having the hotend and bed at printing temperature for at least 5 minutes, then only start printing after a warmup of the same time - you should hopefully find your z-offset no longer drifts.

closed 2018-02-09 by thinkyhead