MarlinFirmware/Marlin #7475: Marlin 1.1.4 - Fixed Probe - Intermittent Trigger

jbronks, 2017-08-10:
I have an NPN inductive sensor that is about 1 mm above the nozzle. I loaded Marlin 1.1.4 and now the Z sensor triggers intermittently. I shut down everything, disconnected Repetier and turned it back on and put a metal object up to it, light comes on but M119 shows open. Hit the home button and it seems to home correctly but metal object up to sensor and it still shows open.
I didn't have issues with earlier Marlin versions, any ideas? Using Arduino 1.8.3 if that matters.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1214863/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1214865/Configuration_adv.txt)

Roxy-3D, 2017-08-10:
Inductive probes are going to be more sensitive to electrical noise that switch type probes.   It might be beneficial to turn on QUIET_PROBING so the heaters are not switching during the actual probe event.   I kind of doubt this is going to help you.  But it is worth trying because if this is the problem, nothing else is going to help.

jbronks, 2017-08-11:
I was not heating the bed or hotend while doing any of the checks.  Why are they more sensitive now that I have new version of Marlin and Arduino?  I didn't have this issue in the past.  I read something about the endstop pullups, could that be the issue, it seems the language in 1.1.4 is a bit different than old version I was using.

EnumC, 2017-11-08:
I seem to also have this issue. Normally the sensor triggers fine, with the red indicator light lighting up on the NPN fixed sensor. But with the latest Marlin build, the sensor sometimes do not trigger despite the indicator light turning on, leading to the nozzle crashing into the bed surface.

benlye, 2017-11-08:
@EnumC - check if you have `#define ENDSTOP_INTERRUPTS_FEATURE` enabled in configuration.h, and if so, try disabling it.  Also, make sure you're using the latest bugfix-1.1.x build.

I had the same problem on my printer when that feature was enabled.  I suspect there is a bug lurking in there somewhere, but there haven't been enough reports of investigation to verify it.

thinkyhead, 2017-11-14:
All changes between 1.1.3 and 1.1.4…

- [Commits](https://github.com/MarlinFirmware/Marlin/compare/1.1.3...1.1.4)
- [Pull Requests](https://github.com/MarlinFirmware/Marlin/pulls?utf8=%E2%9C%93&q=is%3Apr%20merged%3A2017-06-04..2017-07-04)

boelle, 2019-02-19:
@jbronks @EnumC 

is the problem still there with latest bugfix 2.0?

boelle, 2019-03-16:
I cant reproduce this one either, inductive probe and it never failed when using M119

closed 2019-03-17 by p3p