MarlinFirmware/Marlin #7478: Problem offset anet A8

funaro13, 2017-08-10:
Hi, I am here to ask for help from you, I am having adjustment problem in the displacement of my anet a8 capacitive sensor LJC18A3-HZ / BX connected directly to the board, the problem is the following of the calibration procedure With M851Z0, M500, G28, G1X110 Y110 correct Gap, G92. Z0, when I run the command G30 X110 Y110, the offset values of the Z axis are never equal and are with a very large variation, eg first command G30 offset 1.30 second command G30 1.38 and therefore it is different each time, I already checked the Z axis bars all ok, please help me
Paulo

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

benlye, 2017-08-10:
I suspect you have your probe connected to the 5V endstop pins?  If so, you might want to try one of the various wiring options to power it with 12V.  These probes are quite unreliable when driven from 5V.

To confirm that the probe is the problem enable `Z_MIN_PROBE_REPEATABILITY_TEST` in configuration.h and use `M48` to test your probe.

benlye, 2017-08-10:
Also, capacitive probes can be affected by the current running through the bed so you may like to try enabling `PROBING_HEATERS_OFF` (or just probing with the bed heater disabled).

funaro13, 2017-08-10:
How do I enable this command to delete #define and put enable ??

benlye, 2017-08-10:
Delete the slashes (`//`) at the start of the line, so it begins `#define `.

funaro13, 2017-08-10:
ok

funaro13, 2017-08-11:
How do I use the M48 command in the initial script ??

benlye, 2017-08-11:
You don't put M48 in your print code.  

You need to connect to the printer with some kind of serial host - the Arduino IDE has one, or you can use Pronterface or Octoprint - and use a command like `M48 X110 Y110 P10 V3` to do 10 probes in the middle of the bed.  You will get output telling you the variation between the probes.

benlye, 2017-08-11:
Also, which branch of Marlin did you download and install?

funaro13, 2017-08-11:
1.1.4

funaro13, 2017-08-11:
Look at the result, I do not understand anything.
[image]
[image]

Roxy-3D, 2017-08-11:
a standard deviation of .0039 mm is actually very good.   You should check that number in a few places on the bed.  (Especially near the edges)       But compared to the typical layer height of .2 or .3 mm, you won't have any problems due to non-repeatability of the probe.

funaro13, 2017-08-12:
Ok, the sensor is working normal ???

funaro13, 2017-08-12:
I tried to print a single piece at 0.12mm I could not get the nozzle scraped on the table and the filaments did not come out, the sensor should not check this difference between the Z0 point and the nozzle of the extruder and fix those differences ???

funaro13, 2017-08-16:
?????

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead