MarlinFirmware/Marlin #7482: [FR] Scroll filenames in menu if too long for one line

ReneHezser, 2017-08-11:
### Feature Request
Filenames that are long are cut. That makes it hard to decide which file to print from the SD card.
My suggestion is to scroll filenames, if they are too long to be displayed.

Roxy-3D, 2017-08-11:
Agreed....

marcio-ao, 2017-09-11:
I already put in a pull request for this #7615

ReneHezser, 2017-09-11:
@marcio-ao Great. Thank you.
Will it work with a 20x4 display? In the pull request "graphical LCD" is mentioned...

marcio-ao, 2017-09-11:
@ReneHazser: No, but it could easily be adapted to work with it. The "lcd_implementation_drawmenu_sd" in "ultralcd_impl_HD44780.h" would need to have a few modifications done, similar to the modification I made to "_drawmenu_sd" in "ultralcd_impl_DOGM.h".. I don't really one of those displays to test, so I guess someone else would have to do this.

marcio-ao, 2017-09-11:
@ReneHezser: I updated my pull request with changes to HD44780, but I was not able to test since I don't have compatible hardware. Would you be able to test for me?

ReneHezser, 2017-09-12:
@marcio-ao Thanks. I can test it. But probably not before next week.

ReneHezser, 2017-09-22:
Finally I found some time to test your work. I've uploaded a video [here](https://1drv.ms/v/s!AipoXl75Rt3kgYgEZPN1pzPhv7j7vg).
It kind of works. At least partially. In the video you can see what I mean.

marcio-ao, 2017-09-22:
@ReneHezser: Thanks for posting the video! @Phr3d13 said he would take a look at it once Marlin 2.0 stabilized. This particular fix might end up being a team effort :)

Phr3d13, 2017-10-27:
This can be closed now, right?

ReneHezser, 2017-10-27:
Can it be associated two Marlin 2? Or is it on the Roadmap somewhere? I just don't want it to get lost.

Phr3d13, 2017-10-28:
As of the latest bugfix, this feature is implemented

ReneHezser, 2017-10-28:
I see. The referenced pull request (7637) contains some additional information.
So I guess closing this FR is OK. Thanks for the work.

closed 2017-10-28 by ReneHezser