MarlinFirmware/Marlin #7485: UBL Issue

modem7, 2017-08-11:
Hi guys, 

Attempting to do UBL w/ a BLTouch, using a ramps 1.4 w/ mega2650. 

I'm following this guide from #6961: 

> "Short and Sweet" version:
> 
>     Prepare - Unified Bed Leveling - Tools - Build Mesh - Build Cold Mesh
>     Prepare - Unified Bed Leveling - Tools - Build Mesh - Fill-in Mesh - Smart Fill-in
>     Prepare - Unified Bed Leveling - Tools - Validate Mesh - Validate PLA Mesh or Validate ABS Mesh
>     Prepare - Unified Bed Leveling - Tools - Edit Mesh - Fine Tune All
>     Repeat 3 and 4 until perfect.
>     Prepare - Unified Bed Leveling - Mesh Storage - Save Bed Mesh

Having some issues with it beeping when it's purging, then it leaves UBL and does nothing when trying to validate PLA mesh.

It preheats the bed, preheats the nozzle, starts purging whilst beeping incessantly then leaves g26, sets heaters to 0 then nothing further happens. Only message on the screen is "leaving g26"

Cold Mesh builds fine, Smart Fill-in does it's magic voodoo, but stuck at step 3.

Any ideas on this?

Roxy-3D, 2017-08-11:
G26 is a separate GCode from the bulk of the G29 GGodes.    So in a sense, you are leaving G29 when you do it.

G26 needs to pre-heat the bed and nozzle before it can do its work.   Is that when it is beeping?

modem7, 2017-08-11:
In this particular case, it's finished with it's preheating, and going onto
the purge phase, it's at this point that it's beeping (about 10 secs) then
after doing so, it disables the heaters (hot end and bed)

Roxy-3D, 2017-08-11:
If you give it a G29 P  with no length of filament to prime the nozzle...  It will beep and purge until you give it a click.   Or are you trying to do the Mesh Validation Pattern from LCD Control.

modem7, 2017-08-11:
This is all from the lcd currently

Roxy-3D, 2017-08-11:
It should 
- heat up the bed to a reasonable temp.   
- Then it should heat up the nozzle to the correct temperature.    
- Then it should start priming the filament while beeping.    
- And when you click the encoder wheel, it should start trying to draw the pattern.

modem7, 2017-08-11:
Ahhah! I think I've missed out the part about clicking the wheel.

Running through the entire set of steps now!

Appreciate the time!

modem7, 2017-08-12:
- This is where I am with it currently.

Unsure about the fine tuning aspect?

Roxy-3D, 2017-08-12:
What does the G26 Mesh Validation Pattern look like?   You will probably see some areas of the mesh that printed a little too close or too far from the glass.   You can edit those mesh points to correct that.

modem7, 2017-08-12:
no smart fill in
 Bed Topography Report:
 (0,9)                                                                   (9,9)
 (10,190)                                                                (190,190)
 .          .          .          .          .          .          .          .          .          .
 
 .          .          .          .          .          .          .          .          .          .
 
 [ 0.007]  -0.005    0.082    0.079    0.182    0.032    0.112
 
 .          .      -0.150   -0.035    0.027    0.067    0.042    0.162    0.029    0.019
 
 .          .      -0.138   -0.035    0.014    0.119   -0.008    0.022    0.017   -0.013
 
 .          .      -0.145   -0.073   -0.028    0.122    0.039    0.017   -0.065   -0.033
 
 .          .      -0.155   -0.090    0.002   -0.013    0.002    0.007    0.010    0.082
 
 .          .      -0.153   -0.055   -0.033   -0.015   -0.010    0.067    0.002   -0.010
 
 -0.113    0.007    0.092    0.044    0.082    0.079    0.039
 
 .          .      -0.188   -0.110   -0.050    0.082    0.149    0.102    0.029    0.097
 (10,10)                                                                    (190,10)
 (0,0)                                                                     (9,0)

smart fill in
 Bed Topography Report:
 (0,9)                                                                   (9,9)
 (10,190)                                                                (190,190)
 .          .          .          .          .          .          .          .
 
 .      -0.065   -0.065    0.049   -0.005    0.097    0.117    0.202    0.034    0.204
 
 .      -0.108   -0.108  [ 0.007]  -0.005    0.082    0.079    0.182    0.032    0.112
 
 .      -0.150   -0.150   -0.035    0.027    0.067    0.042    0.162    0.029    0.019
 
 .      -0.138   -0.138   -0.035    0.014    0.119   -0.008    0.022    0.017   -0.013
 
 
 
 .      -0.155   -0.155   -0.090    0.002   -0.013    0.002    0.007    0.010    0.082
 
 .      -0.153   -0.153   -0.055   -0.033   -0.015   -0.010    0.067    0.002   -0.010
 
 .      -0.155   -0.155   -0.113    0.007    0.092    0.044    0.082    0.079    0.039
 
 .      -0.188   -0.188   -0.110   -0.050    0.082    0.149    0.102    0.029    0.097
 (10,10)                                                                    (190,10)
 (0,0)                                                                     (9,0)

Roxy-3D, 2017-08-12:
The 'Smart Fill' is iterative.   That is so you can manually adjust some points if need to.   Just do several G29 P3's until it is all the way filled in.    Any error in the estimated values can be quickly edited out when you have the G26 pattern on the glass.

modem7, 2017-08-12:
FINALLY!

The fine tuning seems to have gone great so far (thank you Roxy), and now it's time to do a test print. 

I've applied changes, activated UBL and M500'd it. 

In my start gcode I've got G29 in there. For UBL, do I keep that in there, or does it need something else?

Roxy-3D, 2017-08-12:
You can have a G29 A in your startup GCode...   But it won't do anything because you have already saved the state in EEPROM as active.    I don't have any G29's in my startup GCode.

You shouldn't need to do anything else with UBL until your machine drifts enough that you feel a need to edit the mesh.    You are probably good for 2 or 3 months.   At least...   That is how long a well tuned mesh works for me.   And then, you can just edit the part of it where the first layer isn't going down perfect.

When you take the glass off the bed, be sure to put back on in the same orientation.   And make sure you don't get any little plastic pieces under it.    Other than that, I always put the glass in the freezer to release a print.   That isn't a problem.

modem7, 2017-08-12:
Awesome, 

Roxy, thank you very much for your time, you've helped an absolute idiot understand and implement something awesome! 

Calibration cube printed, everything working as expected. Appreciate it!

closed 2017-11-03 by thinkyhead