MarlinFirmware/Marlin #7486: Step lock after homing?

Chrankus, 2017-08-12:
I don't think i template is suitable for my question.
This is an ongoing discussion on one of the facebook groups. It has to do with the optimal layer height when printing. Our printers use a 8mm leadscrew and so technically 0.04 increments in layer would be optimal, right? We run 200 step motors with 1/16 microstepping.

My question is: **When the printer homes after the second end stop hit, does the stepper lock to a "full" step or can it lock in an odd number of microsteps?** If so, the optimal 0.04mm increment would just be a waste, right?

Thanks in advance :)

GMagician, 2017-08-12:
Answer to your question is a little bit complicate because it involves different things (mechanical weight, friction etc)

First: with a 200 step round motor, 16 microstep and 8 mm your increment is 8/(200*16) = 0.0025mm x microstep 0,04mm is every full step.

You will always be sure to move about 0,04mm steps (maybe some microsteps less or more) otherwise motor don't move at all (max current is given to phases on full step positions and if its not enough to move then...)

AFAIK discussion is too long to resume but contracted to the end you may have a probability to do what you want (instructing firmware about how many microsteps you have) but it's so small.

If you are lucky it should be enough to disable Z step every move (and always move by 0,04 steps only) and it will tend to move itself to full step position...if you are really lucky ;-)

**Note**: if you are not lucky disabling Z you may lose your position

Maybe other guys here can be more detailed about that

teemuatlut, 2017-08-12:
Marlin doesn't know at what point in the sine curve the driver is and there's no way to retrieve that information from the driver (that has no communication capabilities). You could make an educated guess based on the steps taken and based on the assumption that when you powered up the driver started at position 0 and 256.
If you really want to only lock on to full steps, don't use microstepping and then if the movement is too rough, interpolate the movement with TMC drivers.

And then you enable any form of auto bed tramming and completely throw out the hole discussion.

Chrankus, 2017-08-12:
Thank you so much teemualut. That is what i wanted to know. Thanks you as well GMagician.

oysteinkrog, 2017-08-12:
@teemuatlut Using only full steps will probably lead to a visible degradation in quality, 1/8 or 1/16 microstepping is beneficial (I'm assuming average cartesian mechanics here).

teemuatlut, 2017-08-12:
Normal X/Y steps per mm is 80.
Z is 4000.
Less for lead screws though.

I'm still not recommending it. Auto bed leveling is nicer than trying go for some optimal layer height setting.

fiveangle, 2017-08-12:
I agree: i did testing with my Printrbot Simple Metal that includes a 1/8" ACME rod (instead of 8mm ,etc) and was concerned over Z banding.  In the end, it was moot.  Enable super long timeouts for deactivating the steppers, run bedlevelling with the highest microstepping you can safely use on each axis (Z can often be dropped to 1/4 microstepping without problem) and just print.

closed 2018-02-09 by thinkyhead