MarlinFirmware/Marlin #7488: UBL Mesh edit starts with nozzle far above the bed

FHeilmann, 2017-08-12:
@Roxy-3D I am sorry to disturb you amidst the 32 Bit work, but I have some issues with UBL:

I tried to sync up my copy of bugfix-1.1.x which involved re-doing my UBL mesh, since all the changes concerning `BED_X_SIZE` etc. affected the boundaries of my mesh.

I experience the following issue:

I am using `#define Z_CLEARANCE_BETWEEN_PROBES  2`

```
G28
G29 P0
G29 P1 
```

run fine, afterwards, since my bed flexes slightly under my nozzle-switch probe, I want to adjust all mesh points using

`G29 P4 X150 Y150 R999` (my Bed Size is 305x305)

The nozzle moves above the closest mesh-point, however it does not undo the 2mm clearance, which leaves the nozzle above the bed by ~2mm and switches the LCD to edit mode. At this point I would have to use this menu to lower the nozzle all the way towards the bed (which is a pain since I'm undoing 2mm) and I'm pretty sure the mesh topology is not supposed to hold values close to -2mm for each mesh point.

My configuration and build can be found here:

https://github.com/FHeilmann/Marlin/tree/bugfix-1.1.x

Am I missing something?

Roxy-3D, 2017-08-12:
One thing you can do is turn off     #define UBL_MESH_EDIT_MOVES_Z      And the nozzle will stay the same height above the bed as you edit the points.

FHeilmann, 2017-08-12:
Just so I understand you correctly, will this parameter cause the nozzle to undo the 2mm clearance I have configured or will this simply mean that the nozzle doesn't move at all while adjusting the Mesh?

If this parameter controls the latter, I don't think this "fixes" my issue. Previously (i.e. before I synced today) UBL would move the nozzle down to what it thought was Z0 before it asked me to tweak the mesh point. At that point I could insert my 0.05mm shim and adjust the mesh point. 

As it is now, I can't do this anymore.

Roxy-3D, 2017-08-12:
> If this parameter controls the latter, I don't think this "fixes" my issue. 

It kind of sounded like the Z nozzle moving up and down during an edit got too close to the bed.  That is why I suggest that.

> Previously (i.e. before I synced today) UBL would move the nozzle down to what it thought was Z0 before it asked me to tweak the mesh point. At that point I could insert my 0.05mm shim and adjust the mesh point.

For G29 P4....  The way it is supposed to work is when a new mesh point is being edited, it should position the nozzle  at a height of  Z_CLEARANCE_DEPLOY_PROBE and then move the nozzle to mesh point.   

Is it not doing this?    I just looked and the G29_UBL.cpp file has not changed for over a month.   How old was your previous build of the firmware?

Or are you talking about a G29 P2 for a manual probe of a mesh point?

FHeilmann, 2017-08-13:
@Roxy-3D exactly, my nozzle moves on top of the mesh-point and then stays above the bed. From what I can tell it is not exactly 2mm but 2mm-<some_value> which could either be my z-probe-zoffset or the stored mesh point height at that poisition. I then would have to undo the 2mm via the edit menu before I can use a shim to test the height, which results in a mesh-point-storage value of `-(2mm-some_value)` which, I think, during run-time could cause my print-head to crash hard into the build-plate.

The code I used to generate the mesh may very well have been over a month old. Unfortunately I don't know exactly when I did it since I usually overwrite my branch when I rebase it. I've been re-using that mesh between updates until the MESH boundaries changed due to the introduction of BED_X/Y_SIZE

Roxy-3D, 2017-08-13:
In theory...   You shouldn't have to measure the position of the nozzle.    If the mesh generation process is close enough to print, the system was setup to do a G26 Mesh Validation Pattern.   And that is used to get all the mesh points (even the approximated ones that the probe can not reach) perfect.

Have you tried that approach?   Because when you save that mesh to EEPROM, things just work...

FHeilmann, 2017-08-13:
The issue with my setup is that my bed springs flex when my nozzle probe probes the bed, so I need to make adjustments to the outermost points before I can even think about printing validation patterns.

After G28 a G0 Z0 will move the nozzle to the exact point where the nozzle makes contact with the bed.

I tried moving the printhead to Z0.0 before issuing the G29 P3, but the command moves the nozzle back up over the probe point. If I find the time I'll flash 1.1.4 to verify that the behavior used to be correct.

Roxy-3D, 2017-08-13:
I think what I would do is auto probe the regions of the bed where you can do that.    And then manually probe the corners and midpoints of the edges.    With those numbers, you should be able to fill in a set of 'safe' mesh points.     And even if the G26 pattern doesn't print real nice, you will know how to edit each area.

FHeilmann, 2017-08-13:
I flashed 1.1.4, that code behaves the same as the newest bugfix. So this isn't really an issue since the code seems to work as seems to be intended.

Which leaves the philosophical question of whether the code "should do what I suggest it should do". But we can do that another time.

closed 2017-08-13 by FHeilmann

Roxy-3D, 2017-08-13:
Seriously....  Just get mesh numbers in place that are 'not crazy'.    And you can do a partial G26 and stop it if the nozzle is getting too close to the glass.   You just Press & Hold the encoder wheel until it stops.   And then you can edit those points and start again.

If you don't use the C parameter on G26, it will work its way outwards from the starting location.  So you will have plenty of time to stop it if it gets too close to the glass.

> I think what I would do is auto probe the regions of the bed where you can do that. And then manually probe the corners and midpoints of the edges. With those numbers, you should be able to fill in [...]

If you do this and manually fill in the mesh, you will see how quickly the process converges.