MarlinFirmware/Marlin #7491: Mesh Level stuck in LOOP

ZhuDaHai, 2017-08-13:
### Bug Report
- Description: Marlin 1.1.4 Mesh Level restarts after completing. Cannot exit loop
- Expected behaviour: Once Mesh level is completed, it should return to the menu. It does not, I just restart Mesh leveling again
- Actual behaviour: Mesh leveling begins again after finishing. Cannot break out of loop
- Steps to reproduce: Start Mesh leveling. You can't end it
 Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1220196/Configuration.zip)

oysteinkrog, 2017-08-13:
Mesh leveling or UBL?
Could you try UBL instead?

fiveangle, 2017-08-13:
> Mesh leveling or UBL?

MBL (did you look at config.h ?)

It should not be doing that so we want to fix it (rather than switching to another levelling system).

Can you enable bed levelling debug, rerun the process, then upload a zip of the resulting terminal output ?  This will hopefully show what is happening when you (should) get the end.

ZhuDaHai, 2017-08-14:
Here's some more clues. About 30% of the time, Mesh Leveling completes properly and returns to the main screen. The other 70% of the time it loops back into Mesh Leveling Mode.
However, I've done Mesh Leveling with Debug enabled 3 times and it has not looped. Grrr. Some maybe a timing issue or uninitialized variable?
Anywhere here the the debug output from a (successful) Mesh Leveling process. I'll keep trying to get it to loop with debug on.
[Mesh Level Debug - No Loop.zip](https://github.com/MarlinFirmware/Marlin/files/1221067/Mesh.Level.Debug.-.No.Loop.zip)

EddieFerg, 2017-09-10:
I'm seeing the same behavior as ZhuDaHai posted at the top, but with Marlin 1.1.5
After adjusting Z on the last point and pressing the knob, it Homes XYZ, then displays "Click to Begin".  I'm stuck in the Mesh creation loop and can't exit.
This issue is repeatable on my printer.

fiveangle, 2017-09-11:
You say it happens with debug enabled 50% of the time, so it would be useful to have the debug output for an instance when it exhibits the problem.

EddieFerg, 2017-09-11:
Added Debug text file to my comment above.  This is a Bed Level that did not return to the Main Menu upon completion.

OutsourcedGuru, 2017-09-15:
You should probably ignore me since I've not really looked into this much.

Line 41 in [mesh_bed_leveling.cpp](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/mesh_bed_leveling.cpp) calls a reset after doing the sampling. That reset() sets the status attribute to none. If whatever called the routine is looking at status then this could be why it's looping.

thinkyhead, 2017-11-14:
Moving discussion to #7912

PR #8574 (2017-11-27): [1.1.x] Make MBL work more like PROBE_MANUALLY
Fix #7912 and Fix #7491 by making MBL's `LCD_BED_LEVELING` work in the same manner as `PROBE_MANUALLY`.
PR #8575 (2017-11-27): [2.0.x] Make MBL work more like PROBE_MANUALLY
See also #8574

Fix #7912 and Fix #7491 by making MBL's `LCD_BED_LEVELING` work in the same manner as `PROBE_MANUALLY`.
closed 2017-11-14 by thinkyhead