MarlinFirmware/Marlin #7492: Move Axis via Menu freezes until move completed

ZhuDaHai, 2017-08-13:
### Bug Report
Marlin 1.1.4
- Description: Select Move Axis from menu (Z, Y, or Y) and 1mm move. Turn the knob to move and the knob becomes ineffective until the move has completed.
- Expected behaviour: Previous version would allow continues movement settings via the knob while the movement was being executed on the selected axis. 
- Actual behaviour: When you turn the knob to move the axis, the input halts while the axis is actually moving. Thus locking you out from moving the axis further until the current move is completed. 
Its an annoying time-wasting bug that was not present in previous versions of Marlin.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1220205/Configuration.zip)

Roxy-3D, 2017-08-13:
I'm working in the bugfix-2.0.0 branch with the 32-bit stuff.  I'm seeing some issues with respect to very short moves and I know it has something to do with nested interrupts and the head and tail pointer not being fully 'volatile'.     This kind of feels like it is related.

https://github.com/MarlinFirmware/Marlin/issues/7491

https://github.com/MarlinFirmware/Marlin/issues/7493

ZhuDaHai, 2017-08-13:
Ok. Thanks for the info.
FYI: I'm using Marlin 1.1.4 on an Arduino/Genuino Mega or Mega 260 style 8bit board.

Roxy-3D, 2017-08-13:
These types of problems always are difficult.   But with everybody aware of the issue and the possible cross correlation, we probably will figure it out 'relatively quickly'.

thinkyhead, 2017-11-27:
Please check the latest `bugfix-1.1.x` branch and see if the problem still exists. We're focusing more attention on the LCD menus lately, so this is a good time to address it.

boelle, 2019-02-19:
@ZhuDaHai Please check the latest `bugfix-2.0` branch and see if the problem still exists. We're focusing more attention on the LCD menus lately, so this is a good time to address it.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-06-22:
yep adding a reminder more

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle