MarlinFirmware/Marlin #7493: Screen Blanks during extended Extruder Moves through Menu

ZhuDaHai, 2017-08-13:
### Bug Report
Marlin 1.1.4
- Description: When extruding filament through the menu, the screen will go blank after around 10 seconds while the filament is being extruded and will not come back until extrusion is completed.
- Expected behaviour: Screen should not go blank
- Actual behaviour: Screen goes blank when extruding filament through the menu for more than around 10 seconds. 
Dear lord this release has more bugs than 1.1.0 RC8 !!!
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1220216/Configuration.zip)

oysteinkrog, 2017-08-13:
I'm not an expert on this, but this might be on purpose.
I know that there has been a lot of work done to ensure that updating the screen never affects print quality, this might be a case where the firmware simply has to prioritize mechanical movement over screen updates.

fiveangle, 2017-08-13:
I think there was a "fix" where a redraw had to be enforced to work around another issue, so it could be that possibility of long extrude was overlooked, although I may be thinking of Re-ARM branch.

#7167 was the last big change in this area that I can see.

Curious, we are very close to a 1.1.5 release due to so many changes since 1.1.4.  Can you try BF to see if the behavior is different for this issue (and possibly also for your other one) ?

If no improvement, we're no worse off, but would be a waste if it where already fixed.

ZhuDaHai, 2017-08-14:
Well, crap. I don't know what's going on. I enabled Debug Leveling Feature in configuration.h:
`#define DEBUG_LEVELING_FEATURE`
Recompiled and re-uploaded the code to my printer. And now the screen blanking and freezing issue is no longer occurring. I know I didn't have any errors on the previous upload, and everything compiled properly, but now I can't get any of my previous issues to fail anymore: #7492, #7491, #7493
I hate these kind of bugs - impossible to track down.
So I don't know if it had anything to do with enabling the Debug Leveling Feature, or uploading a fresh copy, or uninitialized variables, or board going bad, or what.
I'll let you know if any of these issues come back, but not sure what I can do now.
Here's a dump from M115, just for grins:
```
Send: M115
Recv: FIRMWARE_NAME:Marlin 1.1.4 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:AuroraThor EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:EEPROM:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: ok
```

Roxy-3D, 2017-08-14:
> Well, crap. I don't know what's going on. I enabled Debug Leveling Feature in configuration.h:
#define DEBUG_LEVELING_FEATURE
> Recompiled and re-uploaded the code to my printer. And now the screen blanking and freezing issue is no longer occurring. 

The serial output takes enough time that there are idle cycles and the LCD Screen can get updated.    I'm fighting a problem with nested interrupts in blocks_queued() where adding SERIAL_ECHO()'s are shifting the timing enough to make the problem not happen.

tcm0116, 2017-08-16:
I'm seeing a similar issue on my Delta machine. During printing, the screen will just go blank for a few seconds and then finally refresh. I had assumed it was due to the processor load required by a Delta machine.

ZhuDaHai, 2017-08-20:
It's back. Then its gone. Then its back again. Seems to be random each time I boot the printer. But if its OK its OK for the entire time the printer is on. If its bad, dido. Not sure how Arduino programs work, but sure sounds like uninitialized variable(s).

thinkyhead, 2017-11-27:
Please check the latest `bugfix-1.1.x` branch and see if the problem still exists. We're focusing more attention on the LCD menus lately, so this would be a good time to address it.

boelle, 2019-02-19:
@ZhuDaHai is the problem still there with the latest bugfix 2.0 ?

closed 2019-02-20 by thinkyhead