MarlinFirmware/Marlin #7494: Z fade print quality

gavatron3000, 2017-08-13:
hey guys,

running the UBL on 1.1.4 release with no problems everything works there but im get what i can only assume is over extrusion during the "fade" section of my prints. i was under the impression that the fading didnt add or reduce the amount of plastic being laid down. every print i do has poor quality for the amount that i set the fade to with M420. 10mm or 5mm makes no difference. as soon as its above the fade part print quality is excellent.
Or am i missing something?

Roxy-3D, 2017-08-13:
The amount extruded is not altered by the Z-Correction.   But with that said, if you have large values in your mesh points, you could get an 'over extrusion'.    For example.  If you have most of your mesh points at 1.0 mm and you fade over 10 mm...    At the start your layers will be 10% closer together and extruding at the same rate will put down too much material.    

Can you post a G29 T please?

gavatron3000, 2017-08-15:
ugh...this might be my own doing here. give me a few days (work) and ill let you know......

gavatron3000, 2017-08-20:
Heres a G29 T report. so i went right over my mesh and z offsets to double check i wasnt too close to the bed, ans spent the last few days dialling the mesh in even more. But, i still get this what seems over extruded during the fade section of the print
Recv: Bed Topography Report for CSV:
Recv: 
Recv: 2.245,1.875,2.011,2.150,2.185,1.937,2.047,2.025,1.770,1.870
Recv: 2.047,1.875,1.969,1.935,1.976,1.937,1.940,1.944,1.875,1.850
Recv: 1.849,1.976,1.927,1.830,1.920,1.972,1.905,1.862,2.008,2.025
Recv: 1.863,2.011,1.877,1.972,2.142,2.125,1.855,1.962,1.979,1.820
Recv: 1.792,1.809,1.855,1.972,1.860,1.894,1.865,1.855,1.823,1.805
Recv: 1.895,1.926,1.972,1.841,1.916,1.880,1.831,2.022,1.958,1.827
Recv: 1.944,2.005,1.955,2.015,1.940,2.025,1.920,1.976,1.997,1.908
Recv: 2.026,1.955,1.940,1.910,2.022,1.950,2.032,2.075,1.835,1.894
Recv: 2.029,2.082,2.103,1.979,2.103,2.146,2.001,2.156,2.011,1.983
Recv: 2.055,2.117,2.231,2.020,2.355,2.302,1.960,2.139,2.196,2.253
Recv: ok
[Config & ConfigAdv.zip](https://github.com/MarlinFirmware/Marlin/files/1236876/Config.ConfigAdv.zip)

Roxy-3D, 2017-08-20:
I'll re-read in the morning.    But having the center of your bed at 1.9 mm.  is a little bit outrageous. 
At:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md it says:

# Synopsis

> 
> UBL currently requires both a host and an LCD display with a rotary encoder. Work is in progress to adapt GCode and controller-based procedures to work with UBL.
> 
> The printer must be already fully functional and tested, with a well-constrained movement system. The more physically level and straight the bed is, the better your results will be. See Configuration.h and Configuration_adv.h for all of UBL's settings.
> 
> The printer should be able to successfully print a small object at the center of the bed. Most problems bringing up the UBL Bed Leveling system occur when this step has been ignored. It is very important to verify the configuration.h settings can make this happen.
> 

If you have a mesh point at 1.8mm at the center of the bed...  There is no way you can print a small object at the center of the bed without bed leveling active.

gavatron3000, 2017-08-20:
Ok no worries. If you look at my configuration.h you'll see my z home position is 1.85 mm as this is the physical point at which my z probe switches  will 1.85 being the gap under my nozzle

gavatron3000, 2017-08-24:
So.....ive taken your advice with regards to the mesh height. i removed the manual z home position of 1.85 and set it to default of zero. ive been printing the cube in the centre with the ubl disabled and prints no worries. i reprobed the mesh after this expecting to be roughly around zero in the centre as you emplied, but, the weird thing is now that its like around -1mm and i cant for the life of me think why. ive definitely missed something here

Roxy-3D, 2017-08-24:
> at this point im just going to manually build a mesh. i must have something wrong in the setup of the probe in regards to the ubl

If you can print a G26 pattern...   It almost doesn't matter what it looks like.   You can use that to edit the mesh points into compliance.

gavatron3000, 2017-08-27:
well i was able to move the old mesh down to a mean of 0mm with the G29 P6 C command, so not forcing a z home position of 1.8 and zeoring the mesh mean seems so far to have resolved the issue, a trap for new players! or me over thinking the setup.
thanks for the wisdom and i shouldve just read the documents to the letter, sorry for being a pain in the arse ;) 
cheers mate

closed 2017-08-27 by gavatron3000

Roxy-3D, 2017-08-27:
Glad to hear you got it going!