MarlinFirmware/Marlin #7497: MKS Gen 1.4 - Killed. Printer Halted

Sineos, 2017-08-13:
Environment: 
Board: MKS Gen 1.4 (Brand Keyes Base v1.4)
Board Power Out: HE1, HE2, FAN
LCD: RepRap Discount 12864
Arduione IDE: 1.8.3 incl. latest board / lib updates
Marlin: 1.1.4 Release

Marlin builds fine but right after the boot splash screen it says "Killed. Printer Halted. Please reset"
Basically `Configuration.h` and `Configuration_adv.h` are default except:
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
For the `MOTHERBOARD `definition I tried 33 to 38 and 43 to 48 with the same result.

Any help would be highly appreciated. Thanks

Tannoo, 2017-08-13:
Try it without the display. Even if you leave it defined... just unplug the display and reset the board.

Sineos, 2017-08-13:
Wow, the board is alive. Many thanks for the hint.

Does it mean the display is trash? Kinda seemed to work, since it showed boot logo and is reacting to its reset button

Tannoo, 2017-08-13:
Could be the display, the cables, or pin configuration.

Pin configuration is unlikely.
I would try different cables... mybe even swap the two cables you have first, to see if the problem moves to something else.

Tannoo, 2017-08-13:
If swapping the cables don't change anything, I would replace the display.

Sineos, 2017-08-14:
When connecting only the display cable to Exp 1 it works, the display shows up. This works with both cables.
Connecting the Exp 2 as well kills it. Gonna order a new display.

Many thanks for your help.

closed 2017-08-14 by Sineos