MarlinFirmware/Marlin #7507: Want help with labeling etc on  issues?

boelle, 2017-08-15:
Just want to offer my help again if its wanted :-D

fiveangle, 2017-08-15:
Is the "**_Issues (8xx)_**" looming at the top of every page for anyone else ?  There's so many that the users that say "That fixed it, thanks !" then never close - can't those be closed outright ?  If tackling the backlog makes sense, what about starting from the bottom up and instead of asking "_can this be closed_", just cruise through them and close dead ones outright ?  For example, if there's a thread that has mounds of discussion and then trailed off without resolution, perhaps leave it (and maybe put a label on it that you devise so you can tell you've reviewed it) but short threads, even if they didn't get resolved, just close or maybe with, "_please test latest release and resubmit if issue is not resolved_" then close.  Probably stopping at least a couple months ago ?  However, we certainly don't want to submit "_Can this be closed ?_" to every old issue because then it pollutes the unread notifications feature for all the active issues actually being worked on.

I also think there have been a lot of good architecture discussions recently.  Scott has labelled a few of them, but there's probably many that could use the label as well along with the topics (U.B.L., 32-bit, etc).  For any 32-bit topic threads that trailed off, perhaps close (after labelling with whatever 32-bit label that exists) and comment with a link to the latest 32-bit thread that Scott created to try and consolidate all of them (no. 7076) ?

Does this sound reasonable ?  This is how most other repos handle it so I don't think it's too autocratic.  For old issues I don't think it will cause public outcry or make anyone feel unwanted :)

closed 2017-08-15 by Roxy-3D

fiveangle, 2017-08-15:
@Roxy-3D - hahahaha ! 🥇