MarlinFirmware/Marlin #7513: Unable to compile, erroring out on expression in macros.h

pyr0ball, 2017-08-15:
### Compile Error
Just updating from 1.0.8, only changed over settings in the configuration.h file to match my previous config. Unable to compile and upload. Debug output:
```
Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\fastio.h:33:0,

                 from sketch\MarlinConfig.h:26,

                 from G:\3DP11\Marlin-3DP11-1.1.4\Marlin-3DP11-1.1.4.ino:31:

macros.h:126: error: missing ')' in expression

 #define WITHIN(V,L,H) ((V) >= (L) && (V) <= (H))

                        ^

exit status 1
missing ')' in expression
```

gavatron3000, 2017-08-18:
silly question but your ide is up to date? i had weird things when updating but my ide was not  the latest...

pyr0ball, 2017-08-18:
Good question, but yes, I did update my boards before trying to update the firmware

closed 2018-02-09 by thinkyhead