MarlinFirmware/Marlin #7521: LIN_ADVANCE does not work with DELTA

tcm0116, 2017-08-16:
I tried `LIN_ADVANCE` on my Delta machine, and it did not go well. With any non-zero value for K, the extruder would shudder violently and ultimately retracted the filament as opposed to feeding anything. I've been looking through the code, and I have a few thoughts as to why this is happening.

First, the `a`, `b`, and `c` parameters passed to `Planner::_buffer_line` for a Delta machine are the distances to move the tower carriages. However, `Planner::_buffer_line` computes the distance of the move as if `a`, `b`, and `c` are in the XYZ coordinate system. As such `mm_D_float` and `block->millimeters` do not represent the actual distance that the nozzle moves.

To correct this issue, `Planner::_buffer_line` needs to know the starting and ending coordinates of the move in the XYZ coordinate system. One solution would be to call `forward_kinematics_DELTA` to determine these coordinates, but that would be rather expensive. Another option is to provide either the XYZ delta in mm to `Planner::_buffer_line` when the machine is a Delta type, so that it can calculate the actual length of the move in the XYZ coordinate system. This should be easily achievable since `prepare_kinematic_move_to` can simply provide `segment_distance` to `Planner::buffer_line` (which calls `Planner::_buffer_line`).

```
  const long target[XYZE] = {
    LROUND(a * axis_steps_per_mm[X_AXIS]),
    LROUND(b * axis_steps_per_mm[Y_AXIS]),
    LROUND(c * axis_steps_per_mm[Z_AXIS]),
    LROUND(e * axis_steps_per_mm[E_AXIS_N])
  };

  #if ENABLED(LIN_ADVANCE)
    const float mm_D_float = SQRT(sq(a - position_float[X_AXIS]) + sq(b - position_float[Y_AXIS]));
  #endif

  const long da = target[X_AXIS] - position[X_AXIS],
             db = target[Y_AXIS] - position[Y_AXIS],
             dc = target[Z_AXIS] - position[Z_AXIS];

  float delta_mm[XYZE];
  delta_mm[X_AXIS] = da * steps_to_mm[X_AXIS];
  delta_mm[Y_AXIS] = db * steps_to_mm[Y_AXIS];
  delta_mm[Z_AXIS] = dc * steps_to_mm[Z_AXIS];
  block->millimeters = SQRT(
      sq(delta_mm[X_AXIS]) + sq(delta_mm[Y_AXIS]) + sq(delta_mm[Z_AXIS])
  );
```

Second, `block->use_advance_lead` is set to `true` if there are steps in the X or Y axes, but these once again correspond to movement of the carriages on the towers and not movement in the XYZ coordinate system. The seemingly simple solution would be to check that `block->steps` are not equal for the ABC axes since moving all three carriages an equal amount results in the nozzle moving up or down in the Z axis. However, I believe the Delta calibration offsets can result in the steps not being equal for a Z only movement. A better solution is to use the same solution as the previous issue and look at the XYZ delta in mm and verify that there is movement in the XY plane.

```
    block->use_advance_lead =  esteps
                            && (block->steps[X_AXIS] || block->steps[Y_AXIS])
                            && extruder_advance_k
                            && (uint32_t)esteps != block->step_event_count
                            && de_float > 0.0;
```

I'll try to implement fixes to these two issues and see what happens. If anyone has any other insights into this problem, please chime in.

In the meantime, we might consider adding a sanity check to verify that `LIN_ADVANCE` is not enabled with `DELTA` since it currently does not work

Sebastianv650, 2018-01-10:
Good and true information. I just wasn't reading them in time. thinkyhead added the sanity check as he also found out that it's incompatible.
I'm working in a version 2 that should fix it (#9048), but I doing tests with the planner system first.

One important thing I learned is to put a note into the LIN_ADVANCE section of configuration_adv to mention "@Sebastianv650" or I will likely never read the related issue reports X-)

filippor, 2018-01-10:
for this i suggest to look at pull request https://github.com/MarlinFirmware/Marlin/pull/8778

mylife4aiurr, 2018-01-10:
I dont experience any issues with lin_advance on my delta since 1.1.6

even 1.1.8 disable sanitycheck.h for Delta and Lin_Advance, no problems

> #if ENABLED(LIN_ADVANCE) && !IS_CARTESIAN
>   #error "DEPENDENCY ERROR: Sorry! LIN_ADVANCE is only compatible with Cartesian."
> #endif

tcm0116, 2018-01-11:
@mylife4aiurr it will probably work in a degraded manner, but I doubt it provides optimal results due to the issues I laid out. 

@filoppor I completely forgot about this issue when I was working on #8878. But to your point, that solution is basically an implementation of what I outlined above.

thinkyhead, 2018-02-11:
@tcm0116 Does #8878 by happenstance do anything positive for `LIN_ADVANCE`?

tcm0116, 2018-02-11:
@thinkyhead I don't see the relevance of #8878. Did you mean some other PR?

iosonopersia, 2018-02-11:
I believe he meant #8778

tcm0116, 2018-02-11:
#8778 does address the first issue I outlined. I haven't looked at #9379 to see if it addresses the second issue.

Sebastianv650, 2018-02-11:
@tcm0116 yes #9379 is now fully DELTA compatible.

closed 2018-02-19 by thinkyhead