MarlinFirmware/Marlin #7524: After G29 UBL prints are still not right (bed dipping to Y-tower)

mbabauer, 2017-08-17:
Marlin version: 1.1.4
Printer: Anycubic Kossel Mini (Delta)
Probe: BLTouch, using [fan mount](https://www.thingiverse.com/thing:2443316)

So I have been fighting with getting the auto-leveling working for months now.  I think I have the G33 auto-height working fine, at least it seems to put the Z-height 0 at 0,0 just about right when I test it with paper.  The problem I have is that the G29 UBL doesn't seem to effect the movement of the print head in that the left-side diggs into the bed and the right-side is slightly too high.

Here are the results of my last bed leveling:

```
>>> G29 P1 V4
SENDING:G29 P1 V4
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
Probing Mesh Points Closest to (0.00,0.00).
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: -9.889 Y: -9.889 Z: -0.126
echo:busy: processing
Bed X: 9.889 Y: -9.889 Z: 0.010
echo:busy: processing
Bed X: 9.889 Y: 9.889 Z: 0.051
echo:busy: processing
echo:busy: processing
Bed X: -9.889 Y: 9.889 Z: -0.037
echo:busy: processing
Bed X: 29.667 Y: -9.889 Z: 0.108
echo:busy: processing
Bed X: 29.667 Y: 9.889 Z: 0.078
echo:busy: processing
echo:busy: processing
Bed X: 9.889 Y: -29.667 Z: -0.095
echo:busy: processing
Bed X: -9.889 Y: -29.667 Z: -0.266
echo:busy: processing
Bed X: -29.667 Y: -9.889 Z: -0.174
echo:busy: processing
Bed X: -29.667 Y: 9.889 Z: -0.075
echo:busy: processing
echo:busy: processing
Bed X: 9.889 Y: 29.667 Z: 0.033
echo:busy: processing
Bed X: -9.889 Y: 29.667 Z: 0.046
echo:busy: processing
Bed X: 29.667 Y: 29.667 Z: 0.008
echo:busy: processing
echo:busy: processing
Bed X: 29.667 Y: -29.667 Z: 0.190
echo:busy: processing
Bed X: -29.667 Y: -29.667 Z: -0.327
echo:busy: processing
echo:busy: processing
Bed X: -29.667 Y: 29.667 Z: 0.061
echo:busy: processing
Bed X: 9.889 Y: 49.444 Z: -0.018
echo:busy: processing
Bed X: 49.444 Y: 9.889 Z: 0.125
echo:busy: processing
echo:busy: processing
Bed X: 49.444 Y: -9.889 Z: 0.023
echo:busy: processing
Bed X: 9.889 Y: -49.444 Z: 0.100
echo:busy: processing
Bed X: -9.889 Y: -49.444 Z: -0.216
echo:busy: processing
echo:busy: processing
Bed X: 29.667 Y: -49.444 Z: 0.695
echo:busy: processing
Bed X: -49.444 Y: -9.889 Z: -0.175
echo:busy: processing
Bed X: -49.444 Y: 9.889 Z: -0.098
echo:busy: processing
echo:busy: processing
Bed X: -9.889 Y: 49.444 Z: 0.083
echo:busy: processing
Bed X: 29.667 Y: 49.444 Z: -0.099
echo:busy: processing
Bed X: 49.444 Y: 29.667 Z: 0.104
echo:busy: processing
echo:busy: processing
Bed X: -29.667 Y: 49.444 Z: 0.178
echo:busy: processing
Bed X: -49.444 Y: 29.667 Z: 0.026
echo:busy: processing
echo:busy: processing
Bed X: -49.444 Y: -29.667 Z: -0.211
echo:busy: processing
Bed X: -29.667 Y: -49.444 Z: -0.235
echo:busy: processing
Bed X: -9.889 Y: -69.222 Z: -0.147
echo:busy: processing
echo:busy: processing
Bed X: -49.444 Y: -49.444 Z: -0.260
echo:busy: processing
Bed X: -29.667 Y: -69.222 Z: -0.285
echo:busy: processing
Bed X: -69.222 Y: -9.889 Z: -0.138
echo:busy: processing
echo:busy: processing
Bed X: -69.222 Y: 9.889 Z: -0.064
echo:busy: processing
Bed X: -49.444 Y: 49.444 Z: 0.185
echo:busy: processing
Bed X: -9.889 Y: 69.222 Z: 0.163
echo:busy: processing
echo:busy: processing
Bed X: 9.889 Y: 69.222 Z: -0.009
echo:busy: processing
Bed X: 29.667 Y: 69.222 Z: -0.155
echo:busy: processing
Bed X: -29.667 Y: 69.222 Z: 0.179
echo:busy: processing
echo:busy: processing
Bed X: -69.222 Y: 29.667 Z: 0.053
echo:busy: processing
Bed X: -69.222 Y: -29.667 Z: -0.173
echo:busy: processing
echo:busy: processing
Bed X: -49.444 Y: -69.222 Z: -0.236
echo:busy: processing
Bed X: -69.222 Y: -49.444 Z: -0.223
echo:busy: processing
echo:busy: processing
Bed X: -69.222 Y: 49.444 Z: 0.133
echo:busy: processing
Bed X: -49.444 Y: 69.222 Z: 0.289
echo:busy: processing
Bed X: -9.889 Y: 89.000 Z: 0.286
echo:busy: processing
echo:busy: processing
Bed X: -89.000 Y: 9.889 Z: -0.078
echo:busy: processing
Bed X: -89.000 Y: -9.889 Z: -0.171
echo:busy: processing
>>> G29 S0
SENDING:G29 S0
Mesh saved in slot 0
Done.
>>> G29 A1
SENDING:G29 A1
Unified Bed Leveling System v1.01 active.
>>> G29 T0
SENDING:G29 T0
Bed Topography Report:
(0,9)                                                                   (9,9)
(-89,89)                                                                (89,89)
     .          .          .          .       0.286       .          .          .          .          .

     .          .       0.289    0.179    0.163   -0.009   -0.155       .          .          .

     .       0.133    0.185    0.178    0.083   -0.018   -0.099       .          .          .

     .       0.053    0.026    0.061    0.046    0.033    0.008    0.104       .          .

 -0.078   -0.064   -0.098   -0.075   -0.037    0.051    0.078    0.125       .          .

 -0.171   -0.138   -0.175   -0.174  [-0.126]   0.010    0.108    0.023       .          .

     .      -0.173   -0.211   -0.327   -0.266   -0.095    0.190       .          .          .

     .      -0.223   -0.260   -0.235   -0.216    0.100    0.695       .          .          .

     .          .      -0.236   -0.285   -0.147       .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .
(-89,-89)                                                                    (89,-89)
(0,0)                                                                     (9,0)
```
And in Cura I use the following start script:
```
G28 ;Home
M501 ;Load EEProm
G29 L0 ;Load the mesh
G29 A ;Activate UBL
G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0
G1 F200 E3
G92 E0
```

But if I print anything that ventures even a couple inches from the center it's either digging into the bed on the left side or way too high on the right.  It seems like the mesh isn't really being applied at all.

I have attached my Configuration.h as well.

I do have a ticket open, #7402, due to G33 not really working as well (my standard dev was not right...seemed to be the version of the FW I was using).  I believe this problem is "fixed" now, with only a slight issue being the back-left prob point seems to digg into the bed before it activates the prob pin.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1229807/Configuration.h.zip)

Roxy-3D, 2017-08-17:
So, it looks like you have a Delta.  @oldmcg is the expert on that.   He might offer different and better advice.   But what I would do is just start editing any area that is wrong.   If one portion of the bed is too high...  Move the nozzle there and do a G29 P4 R and push the center points down .1mm.    (and push down the surrounding points by .05mm)    Then move the next area and if it is too low...  push those mesh points down.

And then do another print....     Repeat the print and edit cycles a few times, and you will be amazed at how quickly the process converges.    Very soon....  You will have a perfect mesh.

oldmcg, 2017-08-18:
Your bed map is incomplete.  Please use G29 P3.15 to fill in all the missing grid points.  I also recommend using an odd-sized grid, like 11x11 so you get one point in exact center at 0,0.

But I don't think that directly addresses your problem.

Looking at your map, it appears that upper left and lower right are negative but upper middle, right, and lower right are positive.  That seems warped, not just tilted, which is more likely to be a delta calibration issue than a bed leveling issue, unless you are really putting enough pressure on the glass mounts to bend the glass.  Yes, UBL will mitigate that to some extent, but I think there are other things going here.

I have a mini kossel with bltouch too.  Few things to check for.  Do a probe repeatability test (M48 V4) to see if you are getting consistent probe results from bltouch.  If fan or heater or bed heater is on during probe then bltouch will return different results than with quiet electromagnetics in the head and bed during probe.  Some people heat the bed first then turn off the bed heater before probing, but that didn't make much difference for me.  If you're getting more than 0.05mm differences on bltouch, try slowing down the touch speed (double touch seems useless since it ignores first value and just uses the slow one anyway).

You mentioned G33, but your configuration.h shows fixed tower angle corrections.  I reset all my tower angle and endstop corrections to zero, squared my bed to towers with a physical square as much as possible, then ran G33 P7 V2.  I applied those results to my config.h.

Then start UBL mapping again from scratch.  Go through G29 P1 and P2 then do G29 P3.15 to fill in grid points outside the bed radius.  Don't forget to save it to eeprom.

Before running a test print, enable the grid G29 A and move head down to Z 1mm, then slowly move the head positive and negative x and y starting with small values and slowly increasing distance from center and see if the head still appears to be roughly same 1mm above the bed.  If that looks good then you can move it down to 0.2mm with paper on the glass, give it some x/y movement and see if it digs into the paper.  In between moves don't let your steppers timeout or the head position will be inaccurate without rehoming.  I think it is DEFAULT_STEPPER_DEACTIVE_TIME.

I know Roxy is a fan of G29 P4 to fine tune grid points, but I haven't found that necessary yet -- I get good results with an 11x11 mesh just using G29 P1 followed by G29 P3.15.  I usually skip the P2 step for the manually reachable but probe-unreachable points, just letting P3.15 extrapolate them.  

If you can re-run G33 and do a fresh G29 P1 map, post the G29 T (before G29 P3.15) and lets see if the probed results look much different than this initial map.

oldmcg, 2017-08-18:
I should also mention that G33 sort of assumes a fairly squared bed.  When it probes at head position 0,0, the probe is actually at X/Y_PROBE_OFFSET_FROM_EXTRUDER, but G33 treats that value as "close enough" to 0,0, so having a fairly level bed before G33 can help calibration.  This is different than UBL/G29 which moves the head x/y so the probe is at 0,0 to test height there, not the head.  Slight difference, but could result in differences if the bed is not roughly squared to the towers.

I personally got much better results when I leveled my glass using a physical square to the towers, then ran G33 after that.

mbabauer, 2017-08-19:
@oldmcg 

I ran the `M48 V4` as instructed:
```
>>> M48 V4
SENDING:M48 V4
M48 Z-Probe Repeatability Test
Positioning the probe...
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Bed X: -35.000 Y: 10.000 Z: -1.098
echo:busy: processing
1 of 10: z: -1.086 mean: -1.0859 sigma: 0.000000 min: -1.086 max: -1.086 range: 0.000
echo:busy: processing
2 of 10: z: -1.086 mean: -1.0859 sigma: 0.000000 min: -1.086 max: -1.086 range: 0.000
echo:busy: processing
3 of 10: z: -1.073 mean: -1.0817 sigma: 0.005891 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
echo:busy: processing
4 of 10: z: -1.073 mean: -1.0797 sigma: 0.006248 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
5 of 10: z: -1.086 mean: -1.0809 sigma: 0.006122 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
6 of 10: z: -1.086 mean: -1.0817 sigma: 0.005891 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
7 of 10: z: -1.086 mean: -1.0823 sigma: 0.005646 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
8 of 10: z: -1.086 mean: -1.0828 sigma: 0.005411 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
echo:busy: processing
9 of 10: z: -1.086 mean: -1.0831 sigma: 0.005195 min: -1.086 max: -1.073 range: 0.012
echo:busy: processing
10 of 10: z: -1.073 mean: -1.0822 sigma: 0.005727 min: -1.086 max: -1.073 range: 0.012
Finished!
Mean: -1.082155 Min: -1.086 Max: -1.073 Range: 0.012
Standard Deviation: 0.005727
```
But when I tried to run `G33 P7 V2` the head seems to crash before the probe triggers on the 3rd test point at the X-tower.  All I have to square the towers is a carpenters triangle, and as far as I can tell it's level but I don't think using a carpenter's triangle is really scientifically accurate.  Is there a better way to do the manual part of this?  I have digital calipers if that helps.

mbabauer, 2017-08-19:
I just tried a few more times, and finally it got past the 3 points at the X-tower, but then did the same thing when it got to the 2nd set to the left of the Z-Tower.

oldmcg, 2017-08-19:
If it's crashing during G33 then I suspect the initial geometry is off too much.  Check that all your tower endstops are at roughly the same height.  This is what head sees as "level" before calibration gives you endstop adjustments.

Try shrinking your DELTA_PRINTABLE_RADIUS to 70 instead of 90 to start.  That might be enough to allow G33 to run to completion without crashing.  Then that should give you a reasonable DELTA_RADIUS and DELTA_ENDSTOP_ADJ, and then using those initial values (in configuration.h) you can increase the DELTA_PRINTABLE_RADIUS and see if G33 will run without crashing.  Maybe just increase printable radius to 80 and try G33, then if that looks good increase to 90.

On my mini kossel with 217mm rods (center pivot to center pivot), I get a DELTA_RADIUS of around 105 and I'm only able to get around 75mm DELTA_PRINTABLE_RADIUS without some X/Y positions hitting ball joint mechanical limits (can't rotate the rods on the balls even a tiny bit).  I haven't tried my shiny new magnetic rods/bearings yet -- those might get me more lateral motion without hitting mechanical limits.

mbabauer, 2017-08-19:
Do I measure the endstops from the top?  They are all pretty close, within 1mm according to my caliper (all ~36.5mm - 37.5mm from the top of the rack to the bottom of the switch).

oldmcg, 2017-08-19:
That's really the only way you can practially measure them, but really you're trying to level those endstop trigger points to the bed, but that's what G33 is trying to do for you.  I assume that you've got the DEFAULT_AXIS_STEPS_PER_UNIT set correctly for all three tower steppers so one tower isn't moving at a different rate than the other two.  I see in your configuration.h that you've got them all set at 80.  Mine are set to 100 for my steppers (200 steps/rotation w/ 16 microsteps, 16 pulley teeth with 2mm pitch).

mbabauer, 2017-08-19:
I took those from the original manufacturer's config.  I don't know if they are correct, but that's what they were before.

When I tried to set the bed radius down to 70 got the following:
```
>>> G33 P7 V2
SENDING:G33 P7 V2
?(M665 B)ed radius is implausible.
```

I tried fiddling with the calibration radius as well, but couldn't find a number that would work so I set it back to it's original values.  I tried running `G33 P7 V2` again, watching the head as it probed, and I can tell that when it's probing on the right side the head is tilted to the left so that the probe hits with the print head well above.  But when it probs to the right side it's tilted the other way so that the head is much closer.

oldmcg, 2017-08-19:
Are you saying that when the head assembly moves laterally it tilts more as it moves further?  The head assembly itself should remain perfectly level across all lateral movements.  If the DELTA_RADIUS is significantly wrong for your mechanics then I could see it tilting as it moved, or if the steps/mm are wrong.  I'm not an expert on delta mechanics -- I just have one machine.  But that would definitely explain your head crashing because the probe logic assumes your head is level (or at least consistently tilted) at all horizontal locations because it moves the probe to where the head would be, then assumes if head at that same location it would be the same height as the probe trigger height adjusted for the probe z offset.

oldmcg, 2017-08-19:
Hmmm, I take it back, even if the DELTA_RADIUS or steps/mm were wrong in software that still wouldn't account for more head tilt as it moves laterally.  If I turn off the power and just move my effector head around by hand, the effector head remains level at all horizontal locations.  So it has to be mechanics.

mbabauer, 2017-08-20:
Is this something that can be fixed or adjusted?

oldmcg, 2017-08-20:
Again, I'm no expert on delta mechanics, but are the rods all the same length?  Pair them as closest matching lengths.  Are the rod pairs parallel to each other (the joints are equally spaced on carriages and the effector)?  Are the rods long enough?  My rod angle when head centered is around 30 degrees from vertical or 60 from horizontal.

Roxy-3D, 2017-08-21:
> Is this something that can be fixed or adjusted?

Yes.   Try bumping the DELTA_EFFECTOR_OFFSET up and down by .1 mm.   Move the nozzle to the extreme positions and see which change makes the nozzles stay more vertical.   And take notice how much of the error it corrected.   

And then keep tweaking it until the effector stays level across the entire bed.

oldmcg, 2017-08-21:
Note that DELTA_EFFECTOR_OFFSET just changes DELTA_RADIUS preprocessor computation, and DELTA_RADIUS can be changed with G33 or M665.

Even with no software running, manually moving the effector laterally should not alter the tilt at different X/Y positions.  A small fixed amount of tilt in one direction is OK, just not varying tilt.  I found a thread that discusses effector tilt with some mechanical suggestions:

http://forums.reprap.org/read.php?178,653565,653735#msg-653735

mbabauer, 2017-08-21:
@oldmcg, @Roxy-3D 

Ok, so rod lengths "seem" to be right at 218mm.  I tried my best to measure them from the center bolt to center bolt.  The widths at effector and carriage also seemed pretty close.  The farthest was the Y-Tower, where the effector measured 69.1mm and the carriage was 68.59mm.

I did notice as I was trying to manually set the endstop screws at the top of the bed (trying to manually level the bed best I could) that when I go to the X-Tower the Y-tower carriage was pulling away some.  I tightened the screw on the side that sets the tension on the pulley wheels and checked all the other towers and noticed the Z-Tower caused the X-Tower carriage to also pull some slightly.  I have the X-Tower tight enough now that it's no longer pulling away at the Z-Tower, but the Y-Tower one is still moving ever-so-slightly, almost all of the movement being when the head gets almost to the X-Tower.  How close should these be?  I mean, of course the closer the better, but what is "normal" tolerance?

Also, how are you measuring the angle of the arm to bed/vertical?  I really wish I could find a good write-up on how to properly measure all these things.

mbabauer, 2017-08-22:
Ok, I think I was able to get it to all run correctly.  Several problems that I wasn't noticing in the movements because I was so focused on the probing part:

1. The carriages were not tight enough
2. The endstops needed to be adjusted
3. The mounting bracket holding the BLTouch was hitting the rods on both the X and Z sides (I trimmed it up so it clears now)

After I got everything corrected best I could I ran though the manual calibrations I had done before:
1. Level center using paper
2. Level X, Y, Z in turn by testing w/ paper, adjusting screws on top of carriage, re-centering (auto-home) and repating
3. Re-level center

Once those steps were done, everything seemed to run through much better:
```
>>> G33 P7 V2
SENDING:G33 P7 V2
G33 Auto Calibrate
echo:busy: processing
echo:busy: processing
Checking... AC
.Height:337.06    Ex:+0.00  Ey:+0.00  Ez:+0.00    Radius:97.89
.Tower angle :    Tx:+0.00  Ty:+0.00  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:+0.01     x:+0.07   y:+0.14   z:-0.31
.                 yz:-0.13  zx:-0.23  xy:+0.27
Iteration : 01                                    std dev:0.195
.Height:336.83    Ex:-0.09  Ey:+0.00  Ez:-0.51    Radius:97.64
.Tower angle :    Tx:+0.10  Ty:+0.02  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:-0.02     x:+0.01   y:+0.03   z:-0.14
.                 yz:-0.07  zx:-0.10  xy:+0.13
Iteration : 02                                    std dev:0.087
.Height:336.77    Ex:-0.12  Ey:+0.00  Ez:-0.72    Radius:97.61
.Tower angle :    Tx:+0.15  Ty:-0.03  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:+0.01     x:-0.01   y:+0.02   z:-0.07
.                 yz:-0.02  zx:-0.07  xy:+0.08
Iteration : 03                                    std dev:0.048
.Height:336.71    Ex:-0.16  Ey:+0.00  Ez:-0.84    Radius:97.52
.Tower angle :    Tx:+0.25  Ty:-0.05  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:+0.00     x:-0.01   y:-0.02   z:-0.03
.                 yz:-0.02  zx:-0.02  xy:+0.04
Iteration : 04                                    std dev:0.023
.Height:336.70    Ex:-0.15  Ey:+0.00  Ez:-0.88    Radius:97.45
.Tower angle :    Tx:+0.32  Ty:-0.08  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:-0.00     x:-0.03   y:-0.05   z:-0.02
.                 yz:-0.02  zx:-0.02  xy:-0.00
Iteration : 05                                    std dev:0.027
.Height:336.70    Ex:-0.15  Ey:+0.00  Ez:-0.87    Radius:97.34
.Tower angle :    Tx:+0.40  Ty:-0.08  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
.      c:+0.02     x:+0.01   y:-0.05   z:+0.02
.                 yz:-0.01  zx:+0.03  xy:+0.00
Calibration OK                                    rolling back.
.Height:336.71    Ex:-0.16  Ey:+0.00  Ez:-0.84    Radius:97.52
.Tower angle :    Tx:+0.25  Ty:-0.05  Tz:+0.00
Save with M500 and/or copy to Configuration.h
[... 182 lines cut ...]
 -0.119   -0.064   -0.058   -0.027   -0.047   -0.130   -0.034    0.173    0.248    0.306    0.359

 -0.134   -0.101   -0.074   -0.052   -0.080    0.252    0.094    0.152    0.220    0.281    0.337

 -0.148   -0.114   -0.082   -0.049   -0.007    0.055    0.098    0.146    0.204    0.262    0.318
(-89,-89)                                                                            (89,-89)
(0,0)                                                                             (10,0)
```

So what I am wondering now is:

- Does this look like decent output for the process now?  I am not certain what "good" numbers would look like
- What do I need to run for the begin script in the slicer?  This is what I have right now:
```
G28 ;Home
M501 ;Load EEProm
G29 L0 ;Load the mesh
G29 A ;Activate UBL
G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0
G1 F200 E3
G92 E0
```

Anyway, thanks for all your help!  I feel I am getting close to having good results!

Roxy-3D, 2017-08-22:
> So what I am wondering now is:
> Does this look like decent output for the process now? I am not certain what "good" numbers would look like

Opinions are going to vary...    But if you do a G26 you can edit the mesh points for any part of the pattern that doesn't look perfect.

LVD-AC, 2017-08-23:
Regarding calibration (G33):
`Iteration : 04                                    std dev:0.023`
and
`.Tower angle :    Tx:+0.25  Ty:-0.05  Tz:+0.00
good enough

`.Height:336.71    Ex:-0.16  Ey:+0.00  **Ez:-0.84**    Radius:97.52

You can leave it like that or correct the z-tower end-stop screw a bit (turn and a half counterclockwise) and rerun G33

closed 2018-02-09 by thinkyhead