MarlinFirmware/Marlin #7525: G29 limted to 3x3 ?

lukeskymuh, 2017-08-17:
------------------------------------------------------------
### Bug Report
for Marlin 1.1.4

- Description: 
When I execute G29 T P5 V4 the it stops before reaching the end of the matrix. It also happens vor P values larger than 5. 
For P values of 4 it scanns the whole heatbed but the corretion values are 0
For P values of 3 there is no probelm.
Leveling type: #define AUTO_BED_LEVELING_LINEAR

- Expected behaviour: It worked for values up to 10 (at least) for older marlin versions.
here the working bahavior G29 T P3 V4.

11:36:32.401 : G29 Auto Bed Leveling
11:36:38.209 : Bed X: 35.000 Y: 20.000 Z: -0.175
11:36:42.648 : Bed X: 117.000 Y: 20.000 Z: -0.008
11:36:47.055 : Bed X: 199.000 Y: 20.000 Z: 0.032
11:36:51.266 : Bed X: 199.000 Y: 75.000 Z: 0.031
11:36:55.677 : Bed X: 117.000 Y: 75.000 Z: 0.030
11:37:00.211 : Bed X: 35.000 Y: 75.000 Z: -0.121
11:37:04.511 : Bed X: 35.000 Y: 130.000 Z: -0.069
11:37:08.931 : Bed X: 117.000 Y: 130.000 Z: 0.020
11:37:13.367 : Bed X: 199.000 Y: 130.000 Z: -0.004
11:37:14.637 : Eqn coefficients: a: 0.00086382 b: 0.00029545 d: -0.15239300
11:37:14.641 : Mean of sampled points: -0.02916686
11:37:14.641 : Bed Height Topography:
11:37:14.641 : +--- BACK --+
11:37:14.641 : |           |
11:37:14.645 : L |    (+)    | R
11:37:14.645 : E |           | I
11:37:14.645 : F | (-) N (+) | G
11:37:14.645 : T |           | H
11:37:14.645 : |    (-)    | T
11:37:14.646 : |           |
11:37:14.649 : O-- FRONT --+
11:37:14.649 : (0,0)
11:37:14.649 : -0.03958 +0.04917 +0.02542
11:37:14.653 : -0.09208 +0.05917 +0.06042
11:37:14.653 : -0.14583 +0.02167 +0.06167
11:37:14.657 : Corrected Bed Height vs. Bed Topology:
11:37:14.657 : +0.07667 +0.09458 +0.00000
11:37:14.657 : +0.04042 +0.12083 +0.05125
11:37:14.661 : +0.00292 +0.09958 +0.06875
11:37:14.661 : Bed Level Correction Matrix:+1.000000 +0.000000 +0.000864
11:37:14.665 : -0.000000 +1.000000 +0.000295
11:37:14.665 : -0.000864 -0.000295 +1.000000
11:37:14.669 : X:166.01 Y:131.00 Z:14.90 E:0.00 Count X:33200 Y:26200 Z:11920

- Actual behaviour: 

Here the output for G29 T P4 V4

11:09:10.338 : G29 Auto Bed Leveling
11:09:15.089 : Bed X: 200.000 Y: 20.000 Z: 0.109
11:09:19.242 : Bed X: 145.000 Y: 20.000 Z: 0.105
11:09:23.460 : Bed X: 90.000 Y: 20.000 Z: 0.041
11:09:27.766 : Bed X: 35.000 Y: 20.000 Z: -0.099
11:09:31.898 : Bed X: 35.000 Y: 56.000 Z: -0.061
11:09:36.088 : Bed X: 90.000 Y: 56.000 Z: 0.055
11:09:40.241 : Bed X: 145.000 Y: 56.000 Z: 0.104
11:09:44.386 : Bed X: 200.000 Y: 56.000 Z: 0.114
11:09:48.404 : Bed X: 200.000 Y: 92.000 Z: 0.094
11:09:52.553 : Bed X: 145.000 Y: 92.000 Z: 0.114
11:09:56.722 : Bed X: 90.000 Y: 92.000 Z: 0.086
11:10:00.970 : Bed X: 35.000 Y: 92.000 Z: -0.021
11:10:05.045 : Bed X: 35.000 Y: 128.000 Z: 0.014
11:10:09.223 : Bed X: 90.000 Y: 128.000 Z: 0.080
11:10:13.384 : Bed X: 145.000 Y: 128.000 Z: 0.091
11:10:17.566 : Bed X: 200.000 Y: 128.000 Z: 0.071
11:10:18.865 : Eqn coefficients: a: 0.00000000 b: -0.00000000 d: 0.00000000
11:10:18.865 : Mean of sampled points: 0.05601549
11:10:18.869 : Bed Height Topography:
11:10:18.869 : +--- BACK --+
11:10:18.869 : |           |
11:10:18.869 : L |    (+)    | R
11:10:18.869 : E |           | I
11:10:18.870 : F | (-) N (+) | G
11:10:18.873 : T |           | H
11:10:18.873 : |    (-)    | T
11:10:18.873 : |           |
11:10:18.873 : O-- FRONT --+
11:10:18.873 : (0,0)
11:10:18.877 : -0.04228 -0.05602 +91.94399 +0.00996
11:10:18.877 : -0.07727 -0.05602 -0.05602 +91.94399
11:10:18.881 : -0.00000 -0.05602 -0.05602 +0.00996
11:10:18.885 : -0.00000 -0.04228 -0.05602 +91.94399
11:10:18.885 : Corrected Bed Height vs. Bed Topology:
11:10:18.885 : 0.00000 0.00000 0.00000 0.00000
11:10:18.889 : 0.00000 0.00000 0.00000 0.00000
11:10:18.889 : 0.00000 0.00000 0.00000 0.00000
11:10:18.894 : 0.00000 0.00000 0.00000 0.00000
11:10:18.894 : Bed Level Correction Matrix:+0.000000 +0.000000 +0.000000
11:10:18.898 : 0.000000 0.000000 0.000000
11:10:18.898 : +0.000000 +0.000000 +0.000000
11:10:18.901 : X:106.00 Y:0.00 Z:14.83 E:0.00 Count X:33400 Y:25800 Z:11920

here the output G29 T P5 V4:
11:21:26.558 : G29 Auto Bed Leveling
11:21:32.369 : Bed X: 35.000 Y: 20.000 Z: -0.178
11:21:36.559 : Bed X: 76.000 Y: 20.000 Z: -0.086
11:21:40.696 : Bed X: 117.000 Y: 20.000 Z: -0.011
11:21:44.817 : Bed X: 158.000 Y: 20.000 Z: 0.021
11:21:48.920 : Bed X: 199.000 Y: 20.000 Z: 0.030
11:21:52.914 : Bed X: 199.000 Y: 47.000 Z: 0.035
11:21:57.030 : Bed X: 158.000 Y: 47.000 Z: 0.039
11:22:01.883 : Bed X: 35.000 Y: 74.000 Z: -0.124
11:22:06.037 : Bed X: 76.000 Y: 74.000 Z: -0.041
              -stops here-

- Steps to reproduce:
  - G29 T P4 V4
  - G29 T P5 V4
  - G29 T P10 V4

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1230805/Marlin.zip)

Roxy-3D, 2017-08-17:
Where are those +91.94399 numbers coming from?   They aren't even in the sampled points.   And how can all those numbers be exactly the same to 5 decimal places?   Something is wrong!
```cpp
11:10:18.877 : -0.04228 -0.05602 +91.94399 +0.00996
11:10:18.877 : -0.07727 -0.05602 -0.05602 +91.94399
11:10:18.881 : -0.00000 -0.05602 -0.05602 +0.00996
11:10:18.885 : -0.00000 -0.04228 -0.05602 +91.94399
11:10:18.885 : Corrected Bed Height vs. Bed Topology
```

lrpirlet, 2017-08-17:
@lukeskymuh 
You seems to have enabled the temperature auto report... just to close a door, could you please Issue M155 S30, followed immediately by your G29 command... 

The effect of M155 S30 is to suspend temperature report for 30 seconds and NOT report anything but G29 output...

If that works, then send M155 Sxxx with xxx the number of seconds between 2 temperature auto reports... 5 note Repetier set xxx to 1 in its initialization sequence...

lrpirlet, 2017-08-17:
Sent my answer too fast...
If that works, then, aftre your G29, send M155 Sxxx with xxx the number of seconds between 2 temperature auto reports... ( note Repetier set xxx to 1 in its initialization sequence...) 

What happens is that the console is NOT locked and any gcode ASSUMES that the console is available till the end of the code. With the introduction of the temperature auto report, output from the current gcode and from the temperature auto report does intermixed... Since the output of the temp report is NOT everywhere the same format, it is hard for the host program to keep the temperature hidden from a human reader.

lukeskymuh, 2017-08-18:
@Roxy-3D 
The numbers came from the G29 T P4 V4 command. Right there is something wrong and I have no idea what.

@Irpirlet. I will test this asap. but currently the printer is offline.

Does it work for you?

lrpirlet, 2017-08-18:
@lukeskymuh 
Well I had a bad output from G29 T in UBL... I was loosing data. Then after some hint from @Roxy-3D I was able to make it work by sending M155 S30 to stop temperature auto report interfering with G29 T output... (see #7021)

So Yes my workaround works... 

I can certify that there will be interferences with other gcode output when temperature auto report is enabled as in your case... Does that explain your way out of range +91.94399 ??? I do not know, but I think it would be worse eliminating that possibility

Roxy-3D, 2017-08-18:
This isn't a UBL setup...  It is Old School Grid based leveling...

fiveangle, 2017-08-18:
Can't dig into it myself right now, but I thought that the max you can specify is the max set with this parameter (which you have set to **_3_** in your _Configuration.h_):

```
#define GRID_MAX_POINTS_X 3
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```
If not true, then it should probably be changed to _GRID_DEFAULT_POINTS_X_ (and this investigation continued).    If true, it needs to be checked enforced somehow if user attempts to specify higher number of points.

thinkyhead, 2017-08-18:
They really do mean "max" — as the arrays to store them are pre-allocated and also need to be pre-specified for the EEPROM to be easier to manage. They do also serve as the defaults when other values are not given, and that can be documented as "default is the max according to config."

thinkyhead, 2017-08-18:
If auto reporting of temperatures is implicated, then since `auto_report_temperatures` is called during `idle()` it would be better for code that includes a pause for serial output to call `safe_delay()` rather than `idle()`. The `safe_delay()` call allows necessary processes to run but won't call the often expensive `idle()` function. I'll poke through the `G29` code and make sure no `idle()` calls are being done when a `safe_delay()` call would be better….

thinkyhead, 2017-08-18:
@lukeskymuh I see that there is no upper bounds check on `P`, `X`, or `Y` for linear leveling, so there seems to be a buffer overflow. Try increasing your `GRID_MAX_POINTS_X` and `GRID_MAX_POINTS_Y` values to 10, and in the meantime I will add the necessary bounds check to make sure you `G29` throws a parameter error.

thinkyhead, 2017-08-18:
Ok, so I have a patch pending to enforce the max grid size as configured. If it seems like auto report temperature is also getting in the way, then it's easy to fix. We can turn it off just temporarily during the critical parts of `G29` to make sure the output doesn't mix into the output of `G29`.

thinkyhead, 2017-08-28:
I believe this is patched now in `bugfix-1.1.x`. We can re-open if needed.

closed 2017-08-28 by thinkyhead