MarlinFirmware/Marlin #7527: Auto bed level Z offset

agustinserranoherce, 2017-08-17:
Marlin 1.1.4 not apply z offset using M851 command
- Description: The printer applies G28, G29 and recognizes the command M851 in EPROM, but prints to 930 mm on the warm bed.
I included configuration file for values.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1231070/Configuration.txt)

Roxy-3D, 2017-08-17:
If you turn on DEBUG_LEVELING and do a M111 S255  first thing...  We will get a very verbose long of what steps the printer is going through.    AND....  please mark the point in the log where the wrong thing is happening and explain why it is wrong.

agustinserranoherce, 2017-08-17:
At this point the extruder has not lowered and is 9.3 mm from the bed.
<<< gcode_G29
  current_position=(169.92, 156.96, 19.91) : sync_plan_position
echo:M105
echo:M105
echo:N10 M851 Z-9.3*109
echo:Desfase Z -9.30
echo:N11 G1 Z15.0 F9000*23
echo:N12 G92 E0*116
echo:N13 G1 F200 E3*24
echo:N14 G92 E0*114
echo:N15 G1 F9000*115
echo:N16 M117 Printing...*44
echo:N17 M107*19
echo:N18 G1 F1500 E-6.5*22
echo:N19 G0 F3600 X73.466 Y83.941 Z.36*37
echo:N20 G1 F1500 E0*45
echo:N21 G1 F900 X74.281 Y83.289 E0.12497*7
echo:M105
echo:N22 G1 X75.163 Y82.682 E0.25317*80
echo:N23 G1 X76.091 Y82.139 E0.38191*89
echo:N24 G1 X77.042 Y81.67 E0.50887*104
echo:N25 G1 X78.028 Y81.273 E0.63614*89
echo:N26 G1 X79.039 Y80.948 E0.7633*110
echo:N27 G1 X80.082 Y80.694 E0.89183*93
echo:N28 G1 X81.146 Y80.518 E1.02096*90
echo:N29 G1 X82.185 Y80.423 E1.14589*82
echo:N30 G1 X83.024 Y80.4 E1.24639*91
echo:N31 G1 X97.295 Y80.4 E2.95515*83
echo:N32 G1 X98.357 Y80.438 E3.08239*87
echo:N33 G1 X99.433 Y80.557 E3.21201*90
echo:N34 G1 X99.942 Y80.653 E3.27403*83
echo:N35 G1 X101.052 Y80.488 E3.4084*84
........
and continue the printing

Roxy-3D, 2017-08-17:
> At this point the extruder has not lowered and is 9.3 mm from the bed.

```cpp
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```
Does your probe really trigger just as the nozzle touches the glass?   I kind of doubt that:
```cpp
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
```

agustinserranoherce, 2017-08-17:
Yes, the probe trigger just as the nozzle touches the glass.

agustinserranoherce, 2017-08-18:
I was not able to deduce the logic to configure, but by testing I found the following values that allowed me to set the values using the command M851 or by the LCD. The offset I use -20

#define Z_CLEARANCE_DEPLOY_PROBE   4 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  2 // Z Clearance between probe points

closed 2017-08-18 by agustinserranoherce