MarlinFirmware/Marlin #7528: Compile error after 'Implement SERVO_DELAY as array' integration

GMagician, 2017-08-17:
A bed integration has been done from my code to definitive.
I had removed
> undef SERVO_DELAY
> #define SERVO_DELAY { 50 }
from Conditionals_LCD.h because when you have more than one servo this will lead to a compile error...
If someone has BLTOUCH and more than one servo it has to specify "50" directly in "configuration.h"

Roxy-3D, 2017-08-17:
50 ms?   That is awful fast.   Do you REALLY need to specify that fast of a time?   That is just asking for a problem.

GMagician, 2017-08-17:
It's not me is firmware default that set 50

Roxy-3D, 2017-08-17:
OK!  It must be fine.   But that seems awful fast to me...

GMagician, 2017-08-17:
Consider that there is a 375ms after detach (custom define for bltouch)

GMagician, 2017-08-17:
[PR] #7529 fix it

closed 2017-08-17 by GMagician