MarlinFirmware/Marlin #7539: Repeated Long Extrusion lengths triggering stop [Bug? Setup issue?]

cozierad, 2017-08-20:
I am trying to print a disk that is approx 154mm in diameter. When my machine starts the infill portion the temperature of the hot end starts jumping around a lot. I have the hot end set to 208C but will start jumping around, when it hits 217C it will trigger a shutdown/restart. I do not get a thermal runaway alarm, but this is the only thing I can think of that triggers the stop. 

I have tried several different .STL files but all have the same issue. Smaller prints/prints that don't have long extrusions for a long time do not seem to cause any issues and print well. 

I am running 2.0.0 bug fix on a delta. 

I have looked in Simplify3D for a "split long moves into short moves" command that I thought I remember seeing years ago in Slic3r but can't seem to find a fix.

I have seen the "prevent long extrusions" setting in configuration.h but won't that just trigger an error? basically causing the same issue? or will that cause Marlin to pay attention to the heater throughout a long move?

Thanks

Roxy-3D, 2017-08-21:
There may be a couple of easy ways to work around the problem.  You could turn off thermal protection.   It probably doesn't matter if the infill is printed 10 degrees too hot.

You could also turn down the max speed of the fan in the slicer.   Having the fan blowing hard and then shutting down can cause the nozzle temperature to jump up.

There are a lot of settings you can play with.   But just turning off thermal protection might get the print done.

cozierad, 2017-08-22:
So I was thinking about turning it off, and after your suggestion I gave it a shot. Unfortunately the whole scenario turns into a positive feedback loop and the temp just keeps going up. After the nozzle reached 260C I shut the system down. 

My fan was already off. I tried with the fan on but it makes the situation worse. 

Is this effect because Marlin is not looking at the temp of the nozzle while it performs moves, just in between moves?

fiveangle, 2017-08-23:
2.0 hasnt had even one release yet and is under very active development for 32-bit support (mostly HAL work, but could definitley affecting 8-bit MCUs at this very early stage) so I would suggest trying bugfix1.1.x first and ensure you still ahve the problem, then focus on getting it fixed in that branch, rather than 2.x.  There was another report similar, but it seemed to be related to bed size ?  @Roxy-3D do you remember that issue ?  I asked them to reduce print speed and the issue occured in half the distance, so the user thought it was due to some timeout ?

cozierad, 2017-08-24:
Fiveangle,

I updated to 2.x in order to try and alleviate the issue. I was using a 1.1.x from May before and had the same issue. I'll try the latest 1.1.x. I was under the impression that 1.1.x wasn't being worked on anymore.

thinkyhead, 2018-02-09:
Presuming solved with `M303 U`.

Note that as extrusion speed increases, more heat is removed from the hotend.

closed 2018-02-09 by thinkyhead