MarlinFirmware/Marlin #7540: Marlin 2 Z offset values wildly different that 1.1.4.

KevMags, 2017-08-20:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

KevMags, 2017-08-20:
This is not a big issue and I'm not complaining, just curious.  After upgrading from 1.1.4 BugFix to Marlin 2.0 BugFix I have to set my Z offset values much higher.  On one machine up from 0.41 to 0.71, another machine up from 0.47 to 0.91. 
Are these arbitrary numbers or are they supposed to reflect a true distance? 
Is the distance displayed in Babystepping proportional to the Z offset distance?  They don't seem to be with 2.0.  
I am also unable to set particular values for the Z offset Via the LCD as mentioned in #6172.

closed 2017-08-20 by KevMags

thinkyhead, 2017-08-28:
Keep checking in as 2.0.x develops. Some issues are sure to come and go up till release. Hopefully we'll have a testable official beta release soon.