MarlinFirmware/Marlin #7541: Printer "shuts off" or hangs with 1.1.4

Erlandsson, 2017-08-20:
### Bug Report
- Description: ---

Printer Velleman K8200 shuts off all motors and heaters and goes into "lockdown". Will not restart even with a reset. Have to rip out both power and usb and wait a while.

- Expected behaviour: ---

Finish print?

- Actual behaviour: ---

Just stops with "firmware unresponsive" in middle of prints and shuts everything off. Sometimes 20 layers, other 50 layers. Not consistent.

I have tried the M502 followed by M500, no difference.

- Steps to reproduce:

Retry printing, but maybe getting more or less print.
 
**I have tried reflashing old firmware 1.x something, theone i have been using the last 5 years and it works, no hangs or issues. So it is not the hardware or software. As soon 1.1.4 is flashed problem is back. Only difference between 2 firmwares is the new is configured with filamentwidthsensor**
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1237349/Configuration.zip)

Sblop, 2017-08-21:
I have the same problem. Believe problem is related to the watchdog. I have changed settings to manual reset, but are yet to try if this solves the problem.

Erlandsson, 2017-08-21:
Where do you change the watchdog? Have you tried any other fw version? I really don't want to try another bacause of all the changes one has to do now. Would be great to have a program for automatic cloning of settings from one fw to another. Old fw i used does need older arduino IDE also, so every time i switch i have to uninstall and reinstall older/newer IDE.

Erlandsson, 2017-08-21:
Fount it configuration.adv.h. I will try and turn it off.

Erlandsson, 2017-08-21:
Uncommenting "#define WATCHDOG_RESET_MANUAL" did not help at all. The printer won't start at all, seems it goes into watchdog reset as soon as it is powered.

Commenting "#define USE_WATCHDOG" with "//#define USE_WATCHDOG"

I still get "firmware unresponsive", this time heaters and motors do not shut off as i commented above line.

So still no solution.

Sblop, 2017-08-21:
Arhh... I finally got to upload the firmware.... Totally unresponsive, and I am unable to upload new firmware... :-(

Erlandsson, 2017-08-21:
That doesn't sound good at all.. Was i lucky? I am thinking of reverting to an older version, i just want to be able to use the filamentwidthsensor. It's just so much to change in the code. Too many places to configure now. The old one i used i only had to change things in configuration.h

fiveangle, 2017-08-21:
Have either of you tried with latest bugfix1.1.x ?  There have been a lot of fixes since the 1.1.4 release. 

@Sblop - New FW cannot "brick" a controller - only if you have used incorrect method to flash (perhaps overwriting boot loader area ?) so perhaps you have a different issue than @Erlandsson

Sblop, 2017-08-22:
I believe I have bricked the GT2560 controller. As per several forums it is possible to flash the arduino bootloader using another arduino, but I have ordered a new one from ebay.
Long story short... I use a very old laptop as my arduino computer, as this was just one I had laying around... Compiling Marlin typically takes around 2-5 minutes. - I forgot I was compiling, when the printer froze (it was uploading fw), so after trying several things I pulled the power... bum... brick... I will revert stronger... If I get time I will experiment burning the bootloader to the controller, but for now I will take a 4 week vacation from 3d printing. :-)

Erlandsson, 2017-08-22:
fiveangle: I am really not that keen on configuring another marlin version, it takes me hours to get everything right. But perhaps i do it all wrong? 
What is the latest real stable and tried version? Perhaps i should check out the first iteration with widthsensor ?

Bergerac56, 2017-08-22:
@Erlandsson . You do not have to redo all the work each time you want to check a new version. There are mainly 2 files which are important for configuration: Configuration.h and Configuration_adv.h (unless you have an exotic printer ;) ). 

What I do: Some times ago, I did like you: spend time to have a correctly configured FW. I added a small directory in the example_configurations main directory with a copy of "my" 2 files.

Now, each time I want to upgrade to a new release (and I do this nearly at each new version of bugfix...), I first copy the most up to date config files (with all the new changes I could have done/tested) in this directory and add a copy in the "example_configurations" of the new firmware ( this way, all versions have their adapted files. Very usefull when you want to downgrade). Then I compare "my" 2 files with the new one's present in the "marlin" main directory. There are normally very few real changes and nearly never in the specific parameters for your printer (size, ...). 

To do this "compare", I use Notepad++ with the "compare" plugin. It is very easy: you load the 2 files to compare / alt D/ and you see the differences. You do not care about what you have deliberately changed and you adapt the rest if needed (don't forget to save what you change ;) ). You do that for both files. It takes  less than 5 min. When  done, you replace the original config files by yours. 

Hope this helps...

fiveangle, 2017-08-22:
If all you want is "latest stable for my printer" that will always be the version supplied by your printer manufacturer, if you bought it commercially.

Otherwise, latest stable is 1.1.4 and bugfix1.1.x for issues found in 1.1.4.  2.0x stream is not released and for development only (quite unstable).

This is software and will always have bugs.  If you want to help fix the issue for your printer in 1.1.4 (if one exists) do so here.  Otherwise, just start working backwards at your leisure trying releases until you're happy, but no one can answer the question "which version of Marlin has no bugs with my specific printer ?" because it is a trick question =D

Erlandsson, 2017-08-22:
@fiveangle : Velleman has not updated firmware since it came out on the market 6-7yrs ago so it is a really old marlin. I think it is still 1.0 something.

@Bergerac56 : I will try that.

Erlandsson, 2017-08-22:
Where in which file can i find the version number of marlin?

Bergerac56, 2017-08-22:
@Erlandsson If I understand correctly, you have a Velleman K8200. If you go in the last bugfix version / marlin/"example_configurations"/ Velleman/ K8200, you will find the correct basic config files for your printer, no? Did you try to copy those 2 files in the marlin main directory and compile it?

Erlandsson, 2017-08-23:
Yes that is what i have done. I downloaded the bugfix version and it seem to work now. I haven't had a lockup in 5 prints now.

But i have found a bug with the filamentwidthsensor.

closed 2017-08-26 by Erlandsson