MarlinFirmware/Marlin #7544: [QUESTION] Relationship bed size vs printer limits

GMagician, 2017-08-22:
Why in latest 'configuration.h' has been made a fixed relationship between bed size and software limits?

Usually softlimits are where a machine can move.
min can be less than 0 and max can be more than bed size.
Only mesh or probing needs to work between 0 and bed_size...or do I miss something?

fiveangle, 2017-08-22:
Because most generic cartesian have the bed also the travel limits, which is a safe default. But the configuration capability is still there to manually set probe limits, etc, greater than bed size.

Short answer: to ease bring up for those confused by old method of not having bed size stipulated.

closed 2017-08-22 by GMagician

fiveangle, 2017-08-23:
There's still some room for improvement in the config that I got half worked through, and then got super busy.  Unfortunately my todo list keeps getting longer, not shorter ! :)