MarlinFirmware/Marlin #7547: [FR] Greater Control Over PRINTER_EVENT_LEDS

AcrimoniousMirth, 2017-08-22:
First off, **thanks to the development team, your work is incredible and I cannot begin to express how grateful I am or how inspiring your work has been**. So much so that I'm starting to teach myself C++ so I may contribute myself and not have to bug you guys!
Basically, I love the new addition, just in time for me finally adding RGBs to my printer. However, I have a particular colour scheme, and while I've been able to roughly go through main.cpp and change a few values it feels messy.
Is there provision to add a separate header file that included all the status LED control? 
It could give several possible event types such as start up, heating up, nozzle change, active nozzle etc etc (and perhaps even an event for connection to computer? Would be handy for OctoPrint). These would be commented and uncommented to add or remove the ones people want.
At the top you could have several simple examples of what the LEDs could do. E.g. fade from colour A to B, blink twice, switch colour or cycle colours. These basic examples can be copy-pasted into the if statements for the events people want and the colours before and after set.

Just a thought, I hope I'm not pestering and thanks again for all your awesome work :)

Tannoo, 2017-08-26:
Yeah... i have that on my printer. But not configurable. I have the colors configurable, but not the events.

Thank you, I will make that happen.

Tannoo, 2017-08-27:
Bte, M150 will do any color you want.

AcrimoniousMirth, 2017-08-27:
Thanks dude, keep being awesome!
It will, but it’s not the most refined solution and I can tell from the developments that have taken place that the team is committed to finesse and quality too :)

closed 2017-11-14 by thinkyhead