MarlinFirmware/Marlin #7548: [FR] Option to prevent head motion prior to homing

marcio-ao, 2017-08-22:
Our printers have X_MAX and Y_MAX switches and we have enabled ENDSTOPS_ALWAYS_ON_DEFAULT to prevent the user from crashing their print heads. However, we have been considering whether we could get rid of the hardware XMax and YMax endstops on our printers. The major reason we added these is that Marlin allows motion in the X or Y direction, even before homing has happened. For this to be safe for our users, we would need to have the ability to prohibit all motion until a G28 had been executed.

There currently is a HOME_AFTER_DEACTIVATE constant in Configuration.h, but this does not seem to block motion via G0/G1 upon power on.

There are possibly other ways to solve this problem, such as auto-home on power on, or automatically homing upon first motion command. But disabling motion until homed might be option that provides the least amount of unexpected motion.

marcio-ao, 2017-08-22:
Some discussion of this has already taken place in #7470

fiveangle, 2017-08-23:
@thinkyhead  - can you envision a reason someone would not want to home after power on, but still want HOME_AFTER_DEACTIVATE enabled ?

I ask becasue it seems the behavior Marcio wants is essentially HOME_AFTER_ACTIVATE of the steppers, not after deactivate.  Here is what I was thinking in https://github.com/MarlinFirmware/Marlin/issues/7470#issuecomment-322327939 : 

> Sounds like we need to extend/rename the feature to HOME_AFTER_ACTIVATE then make it cover the case after startup as well.
> 
> @thinkyhead - it looks like HOME_AFTER_DEACTIVATE is getting checked during runtime here: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L1781-L1815
> 
> But where would be the best place to inject this same check at startup before movement ? Do you anticipate any problem with this ?

What do you think ?  Is this viable ?  It seems like it probably should have been that way from the start.

marcio-ao, 2017-08-23:
Although I personally am in favor of home-on-power on, I don't make the decisions and others at my company have expressed a concern that users may not be expecting the motion. They also feared users plugging/powering their printers before fully removing the shipping foam that restricts the various axis during shipping (not entirely sure that is a valid complaint, as one could forget to remove those and manually home anyway).

Simply blocking motion until a G28 is issued seems like the most conservative option that solves the problem. Maybe one of two behaviors could be selected in Configuration.h. NO_MOTION_UNTIL_HOMED being a conservative option that causes no motion, and HOME_AFTER_ACTIVATE being the more aggressive option that causes motion at power on. Having neither would be the present behavior of allowing motion while unhomed.

marcio-ao, 2017-08-23:
Also, I suppose HOME_AFTER_ACTIVATE could cause the print head to crash against a partial print if there was a power outage or brownout that caused the controller board to reset. This is one reason that NO_MOTION_UNTIL_HOMED  may be preferable to some users.

fiveangle, 2017-08-26:
Marcio- Yes, HOME_AFTER_DEACTIVATE triggers homing only upon request for movement (otherwise it would home immediaterly after steppers deactivated : )

AFAIK, turning a printer on does not trigger any movement request.  My proposal to change HOME_AFTER_DEACTIVATE to HOME_AFTER_ACTIVATE was in the UX sense (even though steppers drivers may be active at power on, the intent would be only to have the bot home upon request for movement, even after power on).  Changing the name to NO_MOTION_UNTIL_HOMED may be more clear, but I think our ideas for the actual implimentation are the same.

GMagician, 2017-09-02:
I'll try to address this adding NO_MOTION_UNTIL_HOMED in configuration.h and adding logic where needed

fiveangle, 2017-10-02:
@marcio-ao - @GMagician 's PR got merged for the new feature:

`//#define NO_MOTION_BEFORE_HOMING  // Inhibit movement until all axes have been homed.`

Does that do the trick for you ?

GMagician, 2017-11-18:
I think this can be closed

fiveangle, 2017-12-08:
@marcio-ao can you please close since this has been implemented already ?

marcio-ao, 2017-12-08:
Yup. Thank you!

closed 2017-12-08 by marcio-ao