MarlinFirmware/Marlin #7549: Linear Advance: source code should suggest Simplify 3D compatibility

totalitarian, 2017-08-22:
The source for LIN_ADVANCE mentions this

>    * Slic3r (including Průša Slic3r) produces Gcode compatible with the automatic mode.
>    * Cura (as of this writing) may produce Gcode incompatible with the automatic mode.

As probably one of the top 3 slicers, can anyone suggest if simplify 3d is also compatible with auto mode? It uses variable line width too.

klcjr89, 2017-09-06:
@totalitarian I really like the dynamic single extrusion features in version 4 of S3D, so I do not think it's going to play nicely with LIN_ADVANCE unfortunately. Maybe a post process Gcode script could be made to generate automatic mode like slic3r.

fiveangle, 2017-09-18:
There is no evidence observed that I am aware showing any problem with S3D.  If you're having an issue, please provide details, or close this issue out.

Thanks !

-=dave

closed 2017-11-14 by thinkyhead