MarlinFirmware/Marlin #7550: part cooling fan pin callout

hardin162, 2017-08-22:
##configuration problem

I recently created a dual x carriage Prusa clone mod using a ramps board and a stepper expander and fan expander (http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender). I cannot find the pin call out for the part cooling fan. Marlin still thinks that the part cooling  fan is attached to D9 and when I send a fan speed it still heats up the second hotend. The hotends work with M104 S200 T0 and M104 S200 T1 command so they seem to be called out properly. If anyone knows how to define the part cooling fan pin I would greatly appreciate it.

fiveangle, 2017-08-23:
http://marlinfw.org/docs/gcode/M043.html

hardin162, 2017-08-24:
Ok thank you so much I did not know that existed.

closed 2017-08-24 by hardin162