MarlinFirmware/Marlin #7553: Marlin 1.1.4 RGB strip not working as expected

ctyler2621, 2017-08-23:
### Bug Report
- Description: --- On D5 pin M42 command works but M150 does not
- Expected behaviour: --- M150 G255 should turn on green LED
- Actual behaviour: --- Nothing happens on the green pin. D5 and D6 work as expected.
- Steps to reproduce:
  - Setup and using pins for servo pins D4, D5 D6 since I'm using the LCD and the servo pins are available and readily accessible.
  - Issue M42 P5 S0 through 255 to control the green LED and it works just fine.
  - M150 G255 does nothing perceptible.
  - Getting "non-standard PWM mode" when issuing the M43 I1 (output below):
----
Recv: PIN:   5   Port: E3        RGB_LED_G_PIN               Output = 0    TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt enabled
Recv: .                          SERVO2_PIN                  Output = 0

ctyler2621, 2017-08-24:
Okay, I changed the pin for the Green LED to pin 44 and now M42 works throughout the scall, full PWM. But the M150 command still doesn't work. Any suggestions?

closed 2017-08-24 by ctyler2621

wardnica, 2018-02-04:
Might be the case for someone else too:
I found "parser.seen('U') ? ..." instead of 'G' in the inline void gcode_M150() of the Marlin_main.cpp. This means that M150 U255 turns on the green LED!
Changing the 'U' to 'G' solved this problem.
Before: [https://i.imgur.com/q6I2bgt.png](url)
After: [https://i.imgur.com/yeFOk8n.png](url)

italocjs, 2018-02-24:
Is there any way to fix this without changing the pin?

thinkyhead, 2018-02-25:
The correct letter is `U`, not `G`. To turn on green use `M150 U255`.

italocjs, 2018-02-25:
Thinkyhead, the issue is not about using U or G, The issue is on the output D5, which is a pwm pin, but is not able to work the pwm,  its working fine on D4, D6 and D11 (but i'm using D11 on a BLTouch sensor)

I can change the pin, its not a problem, but it would be nice if someone could take a look on why it isnt working

AnHardt, 2018-02-25:
https://github.com/MarlinFirmware/Marlin/issues/5011#issuecomment-253397753

thinkyhead, 2018-02-26:
If you enable `PINS_DEBUGGING` (and assuming it will compile right now) what does `M43` say about which pins are in use by which timers?

thinkyhead, 2018-02-26:
BLTouch makes use of servo stuff. And servo stuff uses Timer 3. And timer 3 seems to be tied to pin 5. So my guess is you will see it works if BLTOUCH is disabled. If that is the case, we should look for that conflict and work around it in the code. Note that if we move the timer it might generate a whole new pin conflict.