MarlinFirmware/Marlin #7559: Sanity check errors when update from 1.1.0 RC8 to 1.1.4

EnricoSx, 2017-08-25:
Sanity check error when update from 1.1.0 RC8 edit by anycubic for Delta Linear **PLUS** to 1.1.4

` #error "DELTA requires GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y to be odd numbers."`
`#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."`

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1251213/config.zip)

Anycubic configuration.h contain
```
  #define GRID_MAX_POINTS_X 5
  #define GRID_MAX_POINTS_Y ABL_GRID_POINTS_X
```
```
// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// To use a probe you must enable one of the two options above!
```

Other info:
Anycubic Delta Kossel Linear **_PLUS_**
Firmare stock Marlin 1.1.0-RC8 edit from Anycubic
Arduino IDE 1.8.1

fiveangle, 2017-08-25:
> Other info:
> No AutoLeveling

`#define AUTO_BED_LEVELING_BILINEAR`

?

EnricoSx, 2017-08-25:
what?

fiveangle, 2017-08-26:
You say "No Autoleveling" and yet have autolevelling defined (sorry, I thought the conext was obvious).

fiveangle, 2017-08-26:
Your configuration.h is highly modified from the default: please hand-port your changes into configuration*.h in bugfix1.1.x and see if you still ahve an issue.  The obvious reason for your error above is you have an undefined ABL_GRID_POINTS_X that you're trying to set MAX_GRID_POINTS_X to.

You need to go back and fix your modifications to configuration*.h as they are extensive and if not the source of the problem, the source of at least SOME problems :)

If you can simplify and still show an issue then we might be able to help you.

fiveangle, 2017-08-26:
also, when you got back and edit your original post to remove incorrect information that was pointed out, as if it was never there, you make things even more confusing :)

EnricoSx, 2017-08-26:
mmm... ok

EnricoSx, 2017-08-26:
there is an easy way to do this?

EnricoSx, 2017-08-26:
i try marlin firmware migration tool
but with error
DELTA_SMOOTH_ROD_OFFSET is obsolete :\ and now?

EnricoSx, 2017-08-26:
SanityCheck.h:564: error: #error "Enable only one option: Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN."

       #error "Enable only one option: Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN."

EnricoSx, 2017-08-26:
i try correct and update

Tannoo, 2017-08-26:
I use WinMerge to compare differences of files.

closed 2018-02-09 by thinkyhead