MarlinFirmware/Marlin #7561: Change SD card not re-mounting the card

zmeglic, 2017-08-25:
Hi.

I installed Marlin 1.1.4 on Anet A8 printer.
I have default Anet A8 configuration

When clikcing Change SD card and Print from SD card repeatidely, eventualy the new card is read... So I am not sure what the right procedure for mounting/unmounting the SD card is.

------------------------------------------------------------
### Bug Report
- Description:  When changing SD card, the "Change SD card option" does nothing...
- Expected behaviour: The card gets unmounted/mounted
- Actual behaviour: nothing happened

zmeglic, 2017-08-26:
To add to the problem above. I found out some more strange behaviour.  Selecting the "change SD card" from the menu starts the cooling fan for a few moments. More clicks after that the fan motor just barely moves, but it can be heard. 
This is obviously not a normal behaviour.

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead