MarlinFirmware/Marlin #7562: Understanding AUTOBEDLEVELING

studiodyne, 2017-08-25:
I write this for all that have some bad knowledge about AUTO BED LEVELING
On google or anywhere else , just a few knows really how it works , i will  say here all  i know , if i make a mistake , give me the right version , i will update my comment ,  and in a few day , this topic , will be on google and everybody will have a perfect knowlege of autobedleveling

What ABL do or not 
- ABL is not a rectifier , but only a repeter of the bad surface on all your object
- ABL is just made to make adhesion but the price of adhesion is your object deformation
- ABL  copy your surface and will make the safe surface for the roof of your object
- ABL  don't modify extrusion for some first layers , to make a plane vertical , and parallel to the ' Y ' rods
- ABL , construct a tower perfectly perpendicular to your surface
- ABL is not serious , you must have an absolute PERFECT PLAN SURFACE , it's not serious to build and object on a mountain buildplate , the perfect surface is the first tool needed to build and object
- For information , the only material that can help you to have a perfect bed immediatly , is the vitroceram glass or borosilicate , 3mm glass can hold 2kg of plastic without bending , if you can buy 4mm or 5mm glass , you will have a buildplate impossible to bend with the heat or the height of the object
My buildplate is 400x400 and 3mm and with this , i don't need autolevel , the bed is so perfect , i just use the probe , just for swapping the nozzle and adjust offset by firmware .

For the moment , no ABL have the ' FIRST LAYERS CORRECTION' , by modifying the Z height and the extrusion to make a magic perfect layer absolutely linear to your Z axis

So , the first job , is to make 'mechanicaly' a amazing buildplate , and if you have this , you can use a probe ' bltouch' or other to probe the zero , because when you change the nozzle , you don't have to touch your bed screw , but just to find the offset by software , no mechanical modifications
The probing with sensor offer to never touch the bed screws , but , sometimes , you can verify it  lol 

If i make a mistake , please give me all information , on google , nobody knows the difference between all the automated bed leveling , and the documentation is so little , and so many questions , we have after reading it

I don't want to ask THINKYHEAD to work all days and make PR to comment the ABL , it's why i have the idea to make this topic , to see it on google search in a few days 

Thks for all that want to complete this topic 
'Please just add information , no discussion please '

Roxy-3D, 2017-08-25:
> 'Please just add information , no discussion please '

BULL FU#$K!#g $H!+     You put stuff like this in your post and you don't expect some discussion?

> UBL ???? Horrible and so many settings , we are all totally AFRAID , to play with UBL , first we need to know how the minimal ABL works . I never touch this , and i know we are so many to be afraid ! 

At: 
https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

There are simple directions that say to do this AFTER you can

> The printer should be able to successfully print a small object at the center of the bed. Most problems bringing up the UBL Bed Leveling system occur when this step has been ignored. It is very important to verify the configuration.h settings can make this happen.

```cpp
M502          ; Reset settings to configuration defaults...
M500          ; ...and Save to EEPROM. Use this on a new install.
M501          ; Read back in the saved EEPROM.  

M190 S65      ; Not required, but having the printer at temperature helps accuracy
M104 S210     ; Not required, but having the printer at temperature helps accuracy

G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 P2 B T    ; Do manual probing of unprobed points. Requires LCD.
G29 P3 T      ; Repeat until all mesh points are filled in.

G29 T         ; View the Z compensation values.
G29 S1        ; Save UBL mesh points to EEPROM.
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
G29 A         ; Activate the UBL System.
M500          ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
;---------------------------------------------
;--- Fine Tuning of the mesh happens below ---
;---------------------------------------------
G26 C P T3.0  ; Produce mesh validation pattern with primed nozzle  PLA temperatures
              ; are assumed unless you specify B 105 H 225   for ABS Plastic
G29 P4 T      ; Move nozzle to 'bad' areas and fine tune the values if needed
              ; Repeat G26 and G29 P4 T  commands as needed.

G29 S1        ; Save UBL mesh values to EEPROM.
M500          ; Resave UBL's state information.
```
How about you ACTUALLY TRY THOSE DIRECTIONS before you post crap like that.    I bet those steps pretty much get you perfect adhesion across 100% of your bed 100% of them.

studiodyne, 2017-08-25:
I'm sorry ... all is right now , discussion allowed loll

studiodyne, 2017-08-25:
**ok** M502          ; Reset settings to configuration defaults...  
**ok** M500          ; ...and Save to EEPROM. Use this on a new install.
**ok** M501          ; Read back in the saved EEPROM.  

**ok** M190 S65      ; Not required, but having the printer at temperature helps accuracy
**ok** M104 S210     ; Not required, but having the printer at temperature helps accuracy

**ok** G28           ; Home XYZ.
** Not ok** G29 P1        ; Do automated probing of the bed.
**Question** Where are the point location of the definition of the points for probing  ? In firmware ?

** Not ok** G29 P2 B T    ; Do manual probing of unprobed points. Requires LCD.
**Question** The user must stay in front the machine because the lcd need human action ?
** Not ok** G29 P3 T      ; Repeat until all mesh points are filled in.
**Question** I don't know what is it , and how it works , and what it mades ? Explain please

G29 T         ; View the Z compensation values.
**Question** In USB mode and on the screen of repetier , or on the Lcd ?
**ok**G29 S1        ; Save UBL mesh points to EEPROM.
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
**Question** 10mm is too big , what is the minimum possible , what is the value that inform ubl to fade the more quickly as possible ? ' 0 ' ? or do we need to fix a ' raisonnable ' value by ourself  ?
**ok**G29 A         ; Activate the UBL System.
M500          ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
;---------------------------------------------
;--- Fine Tuning of the mesh happens below ---
;---------------------------------------------
G26 C P T3.0  ; Produce mesh validation pattern with primed nozzle  PLA temperatures
**Question** Please explain more what means ' primed ' nozzle  , and when you say ' pla temp' , it's the temp you have used for probing just before  ?
              ; are assumed unless you specify B 105 H 225   for ABS Plastic
G29 P4 T      ; Move nozzle to 'bad' areas and fine tune the values if needed
**Question** ok we move , but , by lcd , gcode , or by adding coordonates to this instruction  ?
              ; Repeat G26 and G29 P4 T  commands as needed.

G29 S1        ; Save UBL mesh values to EEPROM.
M500          ; Resave UBL's state information.

Thks for help , these are serious questions that everybody ask theirselves when we read you post

Roxy-3D, 2017-08-25:
> ok G28 ; Home XYZ.
> ** Not ok** G29 P1 ; Do automated probing of the bed.
> Question Where are the point location of the definition of the points for probing ? In firmware ?

Show some initiative.   JUST DO IT and see what happens.

studiodyne, 2017-08-25:
You can make gain of time to me , if i have a max information by these answers , i will avoid some hours on the machine  , the time is precious , one minute is one day  in this world , all time gain , is our job all the time loll

Roxy-3D, 2017-08-25:
The automated G29 P1 probing does NOT need any configuration of probe points.   It is building a mesh.   Just do it and watch it probe the entire bed to get the first iteration of the automatically generated mesh.

Everything after this step is aimed at filling in unprobed mesh points and further refinement of the accuracy of those mesh points.

studiodyne, 2017-08-25:
@Roxy-3D 
Look i have remade the first post and i will make a very good tuto 

Thks to you , and please be always here to correct all my  mistakes

Tannoo, 2017-08-26:
Isn't there already a guide in the documentation?

studiodyne, 2017-08-26:
@Roxy-3D . I'm googlized now ! 
Thks for help

studiodyne, 2017-08-28:
@Roxy-3D @thinkyhead 
Please can you answer me about this 

1-) Why G29 Jxxxxx make a probing grid but impossible to save it  ?
2-) Why Fade height not on LCD MENU ?
3-) How to know the actual Fade Height ?
3.1-) what is the defaut fade height value in UBL
4-) How to purge EEPROM and delete all grid
5-) why impossible to change grid number points in menu ?

I have tried to choose by lcd my grid number , but impossible to save ,

Roxy-3D, 2017-08-30:
> 1-) Why G29 Jxxxxx make a probing grid but impossible to save it ?

G29 J is used to tilt the mesh.  You can save the tilted mesh if it really does make sense.   But I doubt that really makes sense most of the time..

> 2-) How to know the actual Fade Height ?

I use G29 W    (What? command)

> 3-) what is the defaut fade height value in UBL

10mm.   UBL was developed with that number and it is well tested.    It is possible some people need or want a different number.   But pretty much, 10mm works for most printers.

4-) How to purge EEPROM and delete all grid

You could do a G29 P0    which will zero the current mesh.    And then save it on top of any mesh you want destroyed.

5-) Why the ubl utility in lcd menu tell me to choose my number point and my custom grid , and when i make a grid outside of the ubl menu , the grid is made with the number points defined on the firmware

I'm not sure what you are looking at.  You should probably give us the full path through the LCD Menus so we know what you are seeing.   @Tannoo might know the answer off of the top of his head.

> 1-) Why no LCD menu for FADE HEIGHT ? very usefull to SEE and to SET

I guess nobody has gotten around to adding that if it isn't there.   For that kind of stuff, I would just use PronterFace to send a G29 F 12.34 if I wanted to change it.   For me, command line control is easier.

> 2-) why no initialising memory steps , in ' step by step ubl ' VERY USEFULL FOR BEGINNER

I'm not sure that is true.  This text can be found at: https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

The following command sequence can then be used to home, level, and then fine-tune the results:

M502          ; Reset settings to configuration defaults...
M500          ; ...and Save to EEPROM. Use this on a new install.
M501          ; Read back in the saved EEPROM.  

> 3- Why no possibilites to change the grid resolution by LCD ' may be i don't know how '

The mesh is very expensive from a  RAM memory perspective.   And the access to the mesh is optimized in a number of ways.   Pretty much, if you want to change the mesh dimensions, you need to re-build the firmware.    Most people don't change mesh sizes.  In fact, the most common mesh dimensions are 10 x 10 and that seems to work for most machines.   I'm currently using 10 x 10 on my 400 mm x 400 mm printer.

> 4-) why no ' storing eeprom ' for mesh , and for setup in ' step by step ' Very usefull to have all in one

I'm not sure this is true.   I see at least two G29 S1's in the step by step directions at:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

And the M500 at the end will save the mesh if the firmware is configured that way.

Tannoo, 2017-08-30:
> Why the ubl utility in lcd menu tell me to choose my number point and my custom grid , and when i make a grid outside of the ubl menu , the grid is made with the number points defined on the firmware

I am lost.
There is no UBL Utility in the LCD menu.
At no point is there a selection to change the "grid" from the LCD. 
The "grid" is hard-coded in the firmware. 
Change that in `Configuration.h` and re-compile then flash.

>why no ' storing eeprom ' for mesh , and for setup in ' step by step ' Very usefull to have all in one

You mean this one?
```
      MENU_ITEM(function, "7 " MSG_UBL_SAVE_MESH, _lcd_ubl_save_mesh_cmd);
```

I show this on the LCD: `7 Saved Bed Mesh`

studiodyne, 2017-08-30:
@Tannoo 
I don't know who you are .... But I LOVE YOU !
All i want to read , you write it ... 
Thk you very much 
Now it's too late , i go to sleep , but tomorow i will play with ubl to test my new knowledge and i will be back

_' I'm sure everybody is ok to say  UBL is not an Easy access for '3d maker ' " , too much possibilities and user is lost between them. Not all maker are 'informaticians ' or ' devellopers ' , some are just ' artist or designer ' , and UBL technology not help them
The old leveling was one touch ready , and now , it's a big adventure and it's really impossible to understand it without a long learning.
The future of UBL is to be understood quickly and by anyone , not for big brains, printing must be just printing and not the  C++ and Gcode and REVERSE ENGINEERING '_

Tannoo, 2017-08-30:
UBL is always evolving. The thought is, that one day it will be just a "point-and-level" solution.

studiodyne, 2017-08-31:
Some bad questions have been made , just for the only reason of my bad 'english' , but now , i have enough knowledge to understand them
I'm very surprise , of the firmware integrated grid resolution ,  i believed it was just a database used databasely , and not a compiled built map inside the engine.
It's ok 

For fun , i will give here my own idea of the new ubl 

ONE CLICK UBL 
 Made , stored , activated , and an icon in LCD to see if activated
ACTIVATE  ON/OFF , just one line menu   
FADE HEIGHT     

For gcode  just activation on/off , fade height , and makemesh    it's enough

Roxy-3D, 2017-08-31:
Why stop there?   Let's have one single GCode that says "Print the entire object perfectly."

studiodyne, 2017-08-31:
And with 300 000 km/s  feedrate  of course  !

studiodyne, 2017-08-31:
@Tannoo 
-Why the probing is not made by line like the ' tilted meshing'  ,  my nozzle go everywhere and probe points randomly ?

Tannoo, 2017-08-31:
When building the mesh, it probes the closest points to the current postion of the nozzle.
Because it homes first, you could say that it probes the closest points to home.
All in a close to a circle or arc as it can until all points are probed.

studiodyne, 2017-08-31:
thks 👍

Duinishuodedui, 2017-10-12:
I have a question regarding this function:
Could it take some height parameters on the fly from an external sensor?
I'm asking this question because I'm using Marlin to control my plasma cutter CNC. Due to the heat during the cut, the material warps quite a bit, so you need to continuously adjust the height of the torch to keep it about 2mm close from the thing you're cutting. 
Problem is, probing before the cut is useless in this case, since the warping occurs continuously during the cut. I can relatively easily build a sensor to detect the distance, but then I would need to feed this information to Marlin somehow. Since the Bed leveling function allows to adjust the height on the fly, I was wondering if there were any way to feed this parameter to the firmware, so that it can adjust the torch height continuously based on actual distance measurements?

I 'm pretty sure this is not possible yet, but I think it would be a really nice addition to Marlin, especially considering the crazy high prices of torch height control devices, which makes them difficult to afford for DIYers. I have no idea if this is complicated to do or not to program though.

Thanks a lot, sorry for the rather long message.
Hope you can work on this feature, even if it is not really 3D printing, it would be awesome.

Roxy-3D, 2017-10-12:
> I 'm pretty sure this is not possible yet, but I think it would be a really nice addition to Marlin, especially considering the crazy high prices of torch height control devices

I don't think this would be that hard to do.   The problem you are fighting is most of the people doing the firmware have 3D-Printers and do not have CNC machines.   So, they only see the problems that 3D-Printers have.   They don't see the problems that CNC machines experience.

There are a couple of options...   We could probe a single location close to the cut and see how much it curled up...  And raise the whole mesh that much.   Or we could just raise the mesh points close to where the probe happened.     And there are other questions:   How often can we stop a cut and probe?   If we can stop the cut often enough and probe to see what is happening...    we could probably come up with an algorithm that does the 'right thing'.   

One problem to think about is the curl won't be uniform.   Even if you get one cut done right, if you move some where else on the sheet and start cutting....   You will need a different offset because that location won't be curled.   Maybe the firmware could always do a probe before it starts a cut so it knows how to correct?

Duinishuodedui, 2017-10-16:
Thanks for your answer Roxy :)
I'll try to better describe the way this should work:
The height control works by sensing the voltage of the arc between the torch and the metal plate to be cut. It needs to be done continuously during the cut, while the torch is cutting, because warping happens continuously during the cut, due to the very high temperatures of the plasma. 
So, basically, it should work something like that:
-Lower the torch, to probe a point
-Store this point into the memory as the zero height
-Move the torch up to about 4mm to start the cut (the torch needs to fire a bit higher to avoid metal splashing during the very first seconds)
-Fire the torch
-Move the torch down to the set distance (1 or 2 mm usually)
-Start the cutting path while continuously measuring the voltage and correcting the height in accordance to the measurement (higher voltage = need to low down a bit, lower voltage = need to move up a bit)
-Once the cutting path is finished, move up the torch and prepare for travel
-Repeat this process for the next path

Probing from time to time or only at the beginning of a path wouldn't work, because warp up is very significant on these machines. For some small sheets, warping during the cut can be more than 4-5 millimeters! That's why voltage measurement and height adjustments must be continuous.
Also, you cannot stop the cut and probe too much, because every start and stop of the cutter will leave a big and ugly spot.

Roxy-3D, 2017-10-16:
OK...   We can use (or allow) the various bed leveling systems to work and help...  But it sounds like we need a whole new GCode for doing laser cuts.    And this new GCode would have various parameters.
- initial starting arc height.
- initial starting cut height.
- voltage (or percentage of something)  to track during the cut.
- end point of cut.
- other things that can make things work better.

Do you write C code?   If somebody else doesn't help, I'll help...   We can get this merged with very little work.

Duinishuodedui, 2017-10-18:
Yes, indeed, It may need those parameters. I don't see any other ones aside from the ones you mentioned so I thing you pretty much covered it :)

Unfortunately, I completely suck at coding, I can only do very, very basic things, and would be totally unable to code this in Marlin... If you could help with that to make it an "enable/disable" option, it would really be awesome, I think it could help many people who struggle with plasma height systems and have to pay big bucks for these ! Could finally make plasma cutting an option for the average DIYer!
 I'm not sure if I'm allowed to post external links here, but just so you can get a better understanding of what I'm trying to achieve and what is the machine I'm using, here is my building thread (it is not an ad, I'm not selling anything, it's just me building my DIY 3 in one plasma cutter/giant 3D printer/milling machine. I currently use Marlin to control it with a Ramps):  https://www.vicious1.com/forum/topic/my-mpcnc-made-in-china/

Again, a million thanks for your help !

Roxy-3D, 2017-10-18:
Wow!  Very cool!

fiveangle, 2017-10-18:
I want one ! Haha

studiodyne, 2017-10-19:
Use a loud glass on your parts . To press the objects on the table and to have constant height
Or learn c++ and math and marlin planner timing algorithm 😂😂😂😂

And after you will have to motorise the sensor to place it always in the next direction . If you measure the rear and go to the front .. measurement is wrong . Or place 4 sensors .
After this you will have to be placed in the output of the planner for immediate correction... 
And may be the 16mhz pic avr processor and all routine will never have enough speed to make a true teal time correction .

studiodyne, 2017-10-19:
The only solution is to unconnect z motor and add a new arduino board just programmed for the z teal time correction with a sensor capable of a peripheral area around the laser to make a mean correction for all direction

studiodyne, 2017-10-19:
Or glue the material to fight against warping

Duinishuodedui, 2017-10-23:
These wouldn't work. I cannot glue the material to anything, because the way plasma cutters work. They blast high pressure air to evacuate the melted metal through the plate, so I cannot have anything on top of the plate or even under. The plate will be supported by some kind of mesh.
The only appropriate way to sense the height is by taking voltage measurement of the plasma arc, which shouldn't be very difficult. Any other way to sense the height will fail, due to many reasons (sensor going over already cut spots, very high magnetic interfecences, very bright light, localized warping and so on) this is really the only way.

But these are no concerns to me, the only problem is how can I feed this voltage information in real time to the control board, so that it can compensate the height automatically during the cut. I'm sure this is technically possible, because height compensation is something that Marlin already does in the autobed mesh leveling. The only difference here would be that the mesh leveling would not be done before the cut, but while cutting, on the fly.
I think that Roxy understands the problem perfectly, lets just hope that it can be solved in a future release :)

Roxy-3D, 2017-10-23:
> But these are no concerns to me, the only problem is how can I feed this voltage information in real time to the control board, so that it can compensate the height automatically during the cut. I'm [...]

Yes.  To simplify things we have removed this extra term.  But there was a z_offset term that got added to every move.    In the case of a CNC doing plasma cutting, we could put back in a term that factored in the current voltage and tracked that too.    And we could make it break up mesh cells into 100 (10 x 10) smaller mesh cells where the voltage is factored in at each sub-mesh cell crossing.

But to do this...   Hardware is required.   If you can put together a sensor that accurately measures the instantaneous cutting voltage...   We can make a few additions to the code to factor that into compensation.    I don't know that we will be successful.   But from what I'm hearing I think it is likely to work.

The only difference here would be that the mesh leveling would not be done before the cut, but while cutting, on the fly.
I think that Roxy understands the problem perfectly, lets just hope that it can be solved in a future release :)

Duinishuodedui, 2017-10-23:
Yes, I will make the sensor soon, it should not be too complicated. Just a little voltage divider and some filters to get a clean 0-5V signal from the original 0-96V of the cutter. I may use another small Arduino in between to clean the signal and add a few functions, then connect this Arduino to the Main Arduino controlling the CNC machine, through an unused pin.

Tannoo, 2017-10-23:
Ooooo... a trinket pro would work nicely with that.

Roxy-3D, 2017-10-23:
>  Just a little voltage divider and some filters to get a clean 0-5V signal from the original 0-96V of the cutter. 

I know I'm paranoid.  But a couple of zener diodes to clamp the voltages going to the A to D converter might make sense too....

AnHardt, 2017-10-23:
My first idea would be to regularly (100ms?) test if to high or ot low and insert some babysteps into the right direction. This will resonate a bit, but be better than nothing.

studiodyne, 2017-10-23:
Marlin use a buffer of 16 lines , but the z axis can be adjusted for each moves , 
Converting a 0/5v a in an analog pin and add a mapped or may be ' linear 'z correction ok possible
But , if possible a little math computation , and a little table map of correction to have a correct feedrate
On this repository , a user have made a firmware , with all memory allocation by features
but in your situatin you deactivate the heaters , the ubl , the bltouch , zero extruders/hotends , you will have enough progmem to make all possible
And if not possible , now with an arduino board , you can add a 32 bit teensy , with infinite progmem
For me , the most difficult ,is to take the information at the source and build a map , but i can see it's a pleasure for you and you never talk about complications
👍

studiodyne, 2017-10-23:
I have a question : Before to make this , a waste so many nights and day to debugg all you never believe it appears in this adventure , i'm sure you're not the one that have this , and may be , a big effort on google and all communauties , and may be some professionals , to know ,if something exist , and may be something ' chinese ' that cost nothing ' may be ' lol

Roxy-3D, 2017-10-23:
> My first idea would be to regularly (100ms?) test if to high or ot low and insert some babysteps into the right direction. This will resonate a bit, but be better than nothing.

Yeah...   That was part of the idea of sub-dividing the normal mesh cells into something with a much finer resolution.  Sort of like the Delta's do with the number of steps per second.   But...  The problem with that is those moves are calculated and buffered in advance.

Your idea of doing it in real time is better.   There could be some code down in the stepper motor ISR that is monitoring the voltage and tracking it with baby steps up or down.   That would do a much better job tracking the cut.

Duinishuodedui, 2017-10-24:
Oh, yes, I didn't mention it but I did quite a lot of research and made my homework before asking this question here, of course.  There are in fact two solutions, but none of them really satisfy me:
-First one is to create a secondary Z axis, controlled with a DC motor in addition to the normal Z axis controlled by the stepper. This guy did something similar: http://www.tamarisktechnicals.com/pages/CheapTHC.html
That is currently what I'm building to make my first tests, but I do not consider this an elegant solution. It may work ok, but it needs way more hardware and complications, and makes things complicated for the beginner. Also, the guy seems to use Mach3 for controlling the machine, while my plan is to use Marlin with the SD card reader, so no computer at all. So I'm not sure I can actually make this work using Marlin (I guess I can, but I'll need to try)
There are some Torch height controllers available commercially, but they all seem very expensive to me (the cheapest I saw was about 300 bucks, the allegedly "good ones" are usually over 1500... that's maybe 3 times the cost of my entire machine!). I might worth it for commercial applications, but it is way too expensive for DIYers.
Also, I love Marlin, I use it for all my CNC/3D printing applications, and I bet having a torch height control function available could bring plasma cutting enthusiasts to using this firmware, since an Arduino/Ramps is extremely cheap and easy to use setup.

"Your idea of doing it in real time is better. There could be some code down in the stepper motor ISR that is monitoring the voltage and tracking it with baby steps up or down. That would do a much better job tracking the cut."

Yes, I think this would be the perfect solution

Roxy-3D, 2017-10-24:
> Also, I love Marlin, I use it for all my CNC/3D printing applications, and I bet having a torch height control function available could bring plasma cutting enthusiasts to using this firmware, since an Arduino/Ramps is extremely cheap and easy to use setup.

> "Your idea of doing it in real time is better. There could be some code down in the stepper motor ISR that is monitoring the voltage and tracking it with baby steps up or down. That would do a much better job tracking the cut."

If you want to do 'cut leveling' with Marlin's auto bed leveling systems....   I'll help....    I suspect there are are number of people that want to do Additive & Subtractive printing with Marlin.   If somebody can do plasma cutting height sensors for a few dollars using a voltage divider and a couple of zener diodes...   It opens a whole bunch of new doors....

mjc619, 2017-11-19:
I have a dual lead screw system where my bed just needs tilt correction before printing, but with UBL over the past two days, I have followed the guide and made a good mesh for PLA, but when I execute G29 L1, G29 J in my gcode to compensate for tilt on that specific print, the leveling stops and does not tram at all.

I think this is an issue with fade hight, loading of the mesh via gcode send via octoprint and needs to be sent with a pause in between, or a factor In UBL that makes it not work well with a moving z bed. but I dont know where to start troubleshooting other than just going though the UBL guide a fourth time

fiveangle, 2017-11-20:
@mjc619 - I applaud you for searching before posting, but if you want support for your problem, highjacking someone elses thread is not the way to do it.

You really need to open a new issue so your problem (and please pay attention to the template contents because if you include all the relevant info asked for when you open it, you'll have the best chance of getting a response from someone who can help).

Good luck !

-=dave

closed 2018-02-09 by thinkyhead