MarlinFirmware/Marlin #7565: UBL - Printing outside mesh margin ignores Z height

Todesengelvr6, 2017-08-26:
You're going to have to forgive me, as I only have about 100 hours under my belt with 3d printing. I'm having an issue I have yet to be able to solve, and that's why I'm here.

I'm utilizing this radial fan mount with a bltouch
https://www.thingiverse.com/thing:2175956
`#define SENSOR_LEFT        39
#define SENSOR_RIGHT       0
#define SENSOR_FRONT       -21
#define SENSOR_BEHIND      0`

Due to this, I can't have a BED_MARGIN less than 39mm which is quite a lot on a 200mm bed. So anything that prints wider than 122mm, the UBL mesh is ignored and on my bed, the Y axis goes way up and just extrudes a lot of filament and then continues along like nothing was wrong. I'm using 15 grid points on my UBL and my mesh is entirely filled in.

`#define BED_MARGIN         39`

If I go less than 39, I receive the error:
`static assertion failed: UBL_PROBE_PT_1_X can't be reached by the Z probe.`

Which makes sense, but how can I resolve this? I can't do any large prints.

Roxy-3D, 2017-08-26:
First...  The UBL_PROBE_PT_x numbers are just used if you decide to tilt your mesh.   Most likely...  You don't need them.   If you print on glass, and take the glass off of the bed to remove a print...   And then are a little bit sloppy and get a small piece of plastic between the glass and the bed when you put the glass back on the printer...  You will have a small tilt.   Those numbers are so you can do a 3 point probe and tilt the glass.

> READ: yi out of bounds in z_correction_for_x_on_horizontal_mesh_line(lx0=160.00,x1_i=14,yi=4)

This error is because you are trying to send the nozzle past 160mm.   That is the X value of the last mesh point.   No correction can be calculated.   If you do a G29 T you probably will see on the right hand side 160 mm for  the last column of mesh points.    

Unless you need an inset...  Don't use it.   With the big probe offset you will have some number of mesh points that can't be auto probed.   Just fill them in with G29 P3 's  and then do a G26 so you can edit them with a G29 P4.    That way, you will have the full 200mm x 200mm bed size to print on.

Todesengelvr6, 2017-08-26:
> This error is because you are trying to send the nozzle past 160mm. That is the X value of the last mesh point. No correction can be calculated. If you do a G29 T you probably will see on the right hand side 160 mm for the last column of mesh points.

Unless you need an inset... Don't use it. With the big probe offset you will have some number of mesh points that can't be auto probed. Just fill them in with G29 P3 's and then do a G26 so you can edit them with a G29 P4. That way, you will have the full 200mm x 200mm bed size to print on.

Yes, you are exactly right. I'm show a 160mm when doing a G29 T. Can you elaborate on that last paragraph? Are you saying don't use the UBL and switch to manual leveling? Or don't use the offsets? As you can see I'm a little confused on that.

Todesengelvr6, 2017-08-26:
Ok I think I got it. My G29 T now displays 200x240. I did a G29 P1 and it measured everything but 3columns and 3 rows. I then ran a G29 P3, but I can't figure out how to adjust the measurement? None of the jog controls work inside Simplify3d, the rotary encoder does nothing except save and move to next point. How do I actually change the value on the G29 P3?

Todesengelvr6, 2017-08-26:
So I got the bed leveling working and I can utilize my full bed length. However, my prints are severely stretched the first half of the print, then the 2nd half of the print is normal. You can see an [example ](http://i.imgur.com/SNxCbVM.jpg)here. Any ideas? If I turn off UBL, my prints are fine.

Roxy-3D, 2017-08-27:
> Edit: Or are you saying comment this line out?  #define UBL_MESH_INSET BED_MARGIN // Mesh inset margin on print area

Set the UBL_MESH_INSET number to 0 or 1.

> Ok I think I got it. My G29 T now displays 200x240. I did a G29 P1 and it measured everything but 3 columns and 3 rows. I then ran a G29 P3, but I can't figure out how to adjust the measurement? [...]

I use PronterFace.   But you can also use the LCD Panel to edit mesh points.    

> So I got the bed leveling working and I can utilize my full bed length. However, my prints are severely stretched the first half of the print, then the 2nd half of the print is normal. You can see an example here. 

Can you print without any bed leveling system active?   Can you post a G29 T map for us to look at?   My guess is you have very large negative mesh points.    In a perfect world, your mesh points should be very close to 0.00 mm   If you have large values for your mesh points, the bed leveling system will expand (or squish) the layers within the first 10 mm (or Fade Height) to fill the gap.   If you have very large negative mesh points, you could see that kind of 'stretch'.

> Any ideas? If I turn off UBL, my prints are fine.

I've never seen that before.   Obviously, something is wrong.   It might take some work to figure out what is configured wrong.    One thing to try is setting the Fade Height to 2mm and do the print.   Then set the Fade Height to 100 mm and do the print.  Do the two prints look different?

Todesengelvr6, 2017-08-29:
So I thought for sure disabling the Z height or lowering it was for sure going to fix it. My bed is pretty level, but it does have a -9 to -10mm offset due to where my bltouch is mounted.
[image]

When I had Z height set to 1mm, it raised after approx the 2nd layer and extruded on top of the print and it was a mess. When I disabled the Z height, part of the nozzle rubbed on some infill housing and pulled it up and created another mess.

I can print perfectly without it, however sometimes my larger prints fail and its very annoying. 

With Bed Leveling disabled, my Z offset is -12.68 and I get perfect prints. The bltouch is still my Z end stop.
With Bed Leveling enabled, the original Z offset of -12.68 causes it to crash into the bed hard! My Z offset with bed leveling enabled is -2.38, again bltouch is also Z end stop.

Let me know what other information I can get for you, I would really like to figure out what is going on here. I'm also open to skype calls, I'll buy you a roll of filament!

Roxy-3D, 2017-09-02:
-10mm for mesh points is wrong.   It is going to cause all kinds of problems.

Turn of the Auto Bed Leveling....    Get your Z_OFFSET_FROM_EXTRUDER set correctly so you can print a small calibration cube at the center of your bed with no bed leveling turned on.    Then we can work through the issues of bringing up UBL.

Todesengelvr6, 2017-09-03:
So I can print just fine with how the bed is now. I can do small prints and medium prints. Only problem I have is when I try to use most of the bed. Here are some images.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Roxy-3D, 2017-09-03:
Those prints look nice.    What does a G26 Mesh Validation Pattern look like?   That will help you know if an area of the mesh needs to be adjusted?

Todesengelvr6, 2017-09-03:
> Those prints look nice. What does a G26 Mesh Validation Pattern look like? That will help you know if an area of the mesh needs to be adjusted?

I will print that as soon as I'm done with my current print. My Z offset is still -12.75 with UBL disabled. When UBL is enabled, that crashes into the bed. My Z offset with UBL enabled is -2.35. As far as the setting "Z_OFFSET_FROM_EXTRUDER". I did a Ctrl+F in the firmware and did not find that line at all. I searched all files, forward and back, case and uncase. Is that the same as M851?

thinkyhead, 2018-02-09:
`Z_PROBE_OFFSET_FROM_EXTRUDER`

closed 2018-02-09 by thinkyhead