MarlinFirmware/Marlin #7567: Bugfix-2.0.x 32 bit - questions on implementing U8g for 32 bit

Bob-the-Kuhn, 2017-08-26:
Rather than muddy up an existing thread I'm opening this one.

I'm trying to make the LCD system in the ARM section of Bugfix-2.0.x generic so that we can stop writing LCD specific code and use the current Marlin LCD code as is. .

I need some advice/help/suggestions.

The goal is to implement the changes in a way that does **NOT** require modification(s) of the U8g library.

In the **U8glib-ARM_ID1691** library there's code wrapped with a **U8G_WITH_PINLIST** define that is required to maintain compatibility with the current MARLIN LCD code.   The options I've explored are:
- Put all the needed code under HAL and then add a "#include "HAL_U8g.h" right after every "#include <U8glib.h>".  That got messy fast.
- Enabling U8G_WITH_PINLIST outside of the library.  Since the library doesn't use any Marlin controlled headers then the library isn't aware of any defines I create within Marlin.  That's a shame because the HAL specific code becomes much simpler.

Bob-the-Kuhn, 2017-08-26:
Never mind - too many late night sessions ...

The external define works.  Now to try to remember what the problem was.

closed 2017-08-26 by Bob-the-Kuhn

p3p, 2017-08-26:
> The goal is to implement the changes in a way that does NOT require modification(s) of the U8g library.

if its not possible U8glib-ARM is my library, it can be changed.