MarlinFirmware/Marlin #7568: ramps 1.4 can't connect to pronterface

DJ0202, 2017-08-26:
Hello,

I have a Ramps 1.4 board witch is running on Marlin 1.1.4 and i use the g-code sender pronterface.

now when I connect my ramps board to my computer and try to connect pronterface with the board, my ramps board resets and pronterface keep saying "connecting...".  When i type a command pronterface says 'error# printer is offline' and i cant get it working not on a baud rate of 115200 neither on a baudrate of 250000.

plz help.

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1253493/configurations.zip)

DirtyDerv, 2017-08-26:
is your comport correct mine shifted for some reason about 2 days ago

DJ0202, 2017-08-26:
In the configuration.h I have set COM 3 and Pronterface is also set to COM 3. 
The weird thing is, when I click on the button connect in Pronterface the ramps bord does restart itself, but Pronterface keep saying "connecting...".

yashunandansureka, 2017-08-26:
Is your arduino having an active connection with some other program?

For example, i used to upload my firmware from the arduino IDE and then try to connect the printer in pronterface without it being disconnected in the arduino IDE.

Tannoo, 2017-08-26:
The serial port in `configuration.h` does not refer to your PC's port. That is the Arduino port. Leave it at 0.

DJ0202, 2017-08-26:
@Tannoo thank you for your comment, that was the problem. tanks

closed 2017-08-26 by DJ0202