MarlinFirmware/Marlin #7569: Im trying to compile the marlin 2.0 for my delta printer and it comes up with the response In file included from sketch\src\HAL\HAL_LPC1768\HardwareSerial.cpp:25:0:  sketch\src\HAL\HAL_LPC1768\HardwareSerial.h:31:28: fatal error: debug_frmwrk.h: No such file or directory anyone know what im missing?

DirtyDerv, 2017-08-26:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

kfazz, 2017-08-26:
Just to confirm, you are on latest commit on bugfix-2.0.x, and you have selected Re-Arm target on bottom toolbar in platform-io?
debug_frmwrk.h is part of the CMSIS/SoC code for LPC1768, located in frameworks/CMSIS/LPC1768/...
scratch that. just remove the "#include <debug_frmwrk.h>" line from src/HAL/HAL_LPC1768/HardwareSerial.h

bobc, 2017-08-26:
PR https://github.com/MarlinFirmware/Marlin/pull/7546 added some files in HAL which need "#ifdef TARGET_LPC1768" added. 

There may be other files like that.

p3p, 2017-08-26:
We should probably make platformio.ini behave more like the Arduino build so these things are picked up in Travis. Currently only the correct HAL folder is added to the source files for each platform, (a remnant from when the framework was included in the HAL folder)

DirtyDerv, 2017-08-26:
sorry whats this re-arm im supposed to click as i cant see it on the tool bar at all

bobc, 2017-08-26:
> We should probably make platformio.ini behave more like the Arduino build so these things are picked up in Travis.

That's a good idea.

@DirtyDerv  The 2.0 branches are not stable, perhaps you should be using 1.1 branch?

DirtyDerv, 2017-08-26:
Sweet i can do that ty :) just need to get the printer working again lol

fiveangle, 2017-08-26:
2.0 has yet to be released.  Use bugfix1.1.x branch.

fiveangle, 2017-08-26:
2.0 travis build will need lots of work to bring up all the 1.1 test cases, and that is probably the last thing to do before the merge, as it's changing so rapidly.

p3p, 2017-08-26:
> 2.0 travis build will need lots of work to bring up all the 1.1 test cases

Not sure what you mean by that, at least for AVR the test cases are in place, obviously depending on the platform features may not be available so we cant run the same set of tests for each ( because that would have made it too easy)

fiveangle, 2017-08-26:
What I mean is that when was trying to get the AVR Travis tests for 1.1 working on the new Trusty platform, I considered just ditching the AVR built chain entirely and switching to PIO, since your work on 2.0 to bring PIO up seemed like it had all been done, but when I reviewed there were a lot of tests that just couldn't be run on the PIO due to various reasons, so they each need to be tracked down.  Not difficult I'm sure, but would probably take the better part of an entire evening to work through all of them.  If someone else wants to tackle, go for it, but me, I'm busy packing for Burningman ;)

fiveangle, 2017-08-26:
Oh, now I remember... I was going to wait until 2.0 was ready to merge with 1.1 and at that time tackle segmenting travis out to so the builds are parallelized for faster TI feedback (you may recall our discussion on that).  Oh, and on your PIO "speed" comments, it looks like the excrutiating slowness I observed was due to something happeing on Sublime Text 3 with DevIoT plugin and release PIO.  Switching to dev branch of PIO and bf version of DevIoT and it's compiling every bit as quick (if not quicker) than the AVR buildchain 👍

essgcee, 2017-08-26:
Devs please comment if I'm wrong about this:
My impression of the 2.0 codebase is that it won't compile at all for AVR processors at the moment; it is still under development.  As far as I can see from an hour of so of trawling the errors (compiling on PlatformIO) the SPI implementation is being refactored and the code and some definitions are simply not there. (SPI implementation is only present for LPC17xx).

p3p, 2017-08-26:
2.0 is in pre-alpha (not released yet) and under active development, you would need the bugfix-2.0.x branch to have any chance of it working. It should work for AVR though the last PR introduce a compile issue in the Arduino IDE (still works in PlatformIO), the AVR platform in bugfix-2.0.x should be at feature parity with 1.1.4 though it is experimental only use it if you can cope with things going wrong (and tell us when they do ;) )

closed 2018-02-09 by thinkyhead