MarlinFirmware/Marlin #7571: heating bug

yashunandansureka, 2017-08-26:
Hi All,

What i am trying to do is use serial2 of the Arduino Mega to send G codes.
I have marlin 1.1.4 running on RAMPS 1.4. I found the appropriate pins and set the appropriate serial port in the code.

`#define SERIAL_PORT 2`

When i connect the printer using pronterface and send G28, i.e. homing, the printer works just fine. I can also move the head using the move buttons.

Now the issue comes when i try to set the temperature. The printer keeps resetting.

When i send M104 S200, the printer resets. I am not able to figure out why is it resetting.
All G commands and M commands work fine and the  printer axis can be moved by pressing the buttons in pronterface. But as soon as i try to set the temperature, the Printer Resets.

Any idea as to what may be the issue?

Tannoo, 2017-09-05:
#7464

marcio-ao, 2017-09-05:
Make sure your power supply is able to handle the load of powering on the heater. What may be happening is that the voltage is dropping enough to reset the board.

yashunandansureka, 2017-09-06:
I checked it.

The thing is the problem occurs only when i am using **serial 2.** 

If i use **serial 0**, i.e. the default setting, then everything works just fine.

thinkyhead, 2017-11-27:
Resolved in the current `bugfix-1.1.x` and/or `bugfix-2.0.x` branches?

closed 2018-02-09 by thinkyhead