MarlinFirmware/Marlin #7572: Delta UBL Start Probing at the Center

kevintong116, 2017-08-26:
After I sent the command `G29 P1`, the printer start probing at 0,0 position

Here is the mesh that I got:
```00:14:07.800 : Mesh invalidated. Probing mesh.
00:17:51.747 : X:0.00 Y:0.00 Z:304.65 E:0.00 Count X:39374 Y:39374 Z:39374
00:17:52.833 : Bed Topography Report:
00:17:52.855 : (0,9)                                                                   (9,9)
00:17:52.871 : (1,179)                                                                (179,179)
00:17:53.071 : .          .          .          .          .          .          .          .          .          .
00:17:53.071 : 
00:17:53.202 : .          .          .          .          .          .          .          .          .          .
00:17:53.202 : 
00:17:53.372 : .          .          .          .          .          .          .          .          .          .
00:17:53.372 : 
00:17:53.534 : 
00:17:53.534 : 
00:17:53.671 : .          .          .          .          .          .          .          .          .          .
00:17:53.671 : 
00:17:53.834 : 0.403    0.352    0.317       .          .          .          .          .          .          .
00:17:53.834 : 
00:17:54.007 : 0.370    0.344    0.310    0.322       .          .          .          .          .          .
00:17:54.007 : 
00:17:54.172 : 0.355    0.313    0.315    0.290    0.363       .          .          .          .          .
00:17:54.172 : 
00:17:54.335 : 0.328    0.343    0.336    0.363    0.451       .          .          .          .          .
00:17:54.335 : 
00:17:54.504 : [ 0.313]   0.333    0.354    0.386    0.523       .          .          .          .          .
00:17:54.504 : (1,1)                                                                    (179,1)
00:17:54.520 : (0,0)                                                                     (9,0)
```
Here is my configuration
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1253807/Marlin.zip)

Roxy-3D, 2017-08-26:
Doesn't your Delta home to the center of the bed?   It looks like you started the G29 P1 with the nozzle at a corner of the bed.    

Also...   Can you print a small object at the center of your bed with no bed leveling active?   Because your probed mesh points are .35 mm off the bed.     You need to be able to print something small in the center of the bed before you try to bring up a bed leveling system.

kevintong116, 2017-08-27:
My Delta home to the center of the bed. The XY coordinate are 0,0 after homing

The printer is able to print small object at the center with no bed leveling. The 0.35 should be the thickness of my probe. I just forgot to set that.

kevintong116, 2017-08-27:
`G29 P1 V4` gives
```12:07:50.937 : Mesh invalidated. Probing mesh.
12:07:50.937 : Probing Mesh Points Closest to (0.00,0.00).
12:07:57.136 : Bed X: 1.000 Y: 1.000 Z: 0.363
12:07:57.737 : Bed X: 1.000 Y: 20.778 Z: 0.378
12:07:58.439 : Bed X: 20.778 Y: 1.000 Z: 0.395
12:07:59.056 : Bed X: 20.778 Y: 20.778 Z: 0.381
12:07:59.774 : Bed X: 1.000 Y: 40.556 Z: 0.405
12:08:00.906 : Bed X: 40.556 Y: 1.000 Z: 0.379
12:08:01.524 : Bed X: 40.556 Y: 20.778 Z: 0.361
12:08:02.230 : Bed X: 20.778 Y: 40.556 Z: 0.363
12:08:02.847 : Bed X: 40.556 Y: 40.556 Z: 0.340
12:08:03.828 : Bed X: 1.000 Y: 60.333 Z: 0.407
12:08:04.435 : Bed X: 20.778 Y: 60.333 Z: 0.382
12:08:05.800 : Bed X: 60.333 Y: 1.000 Z: 0.398
12:08:06.401 : Bed X: 60.333 Y: 20.778 Z: 0.363
12:08:07.011 : Bed X: 60.333 Y: 40.556 Z: 0.328
12:08:07.719 : Bed X: 40.556 Y: 60.333 Z: 0.360
12:08:08.705 : Bed X: 1.000 Y: 80.111 Z: 0.465
12:08:09.322 : Bed X: 20.778 Y: 80.111 Z: 0.389
12:08:10.278 : Bed X: 60.333 Y: 60.333 Z: 0.335
12:08:11.512 : Bed X: 80.111 Y: 1.000 Z: 0.548
12:08:12.104 : Bed X: 80.111 Y: 20.778 Z: 0.451
12:08:12.730 : Bed X: 80.111 Y: 40.556 Z: 0.401
12:08:13.886 : Bed X: 40.556 Y: 80.111 Z: 0.417
```
Is the program treating my printer as cartesian but not deltabot?

Roxy-3D, 2017-08-27:
Can you attach your configuration.h (and configuration_adv.h) files?    How big are your X_OFFSET_FROM_EXTRUDER and Y_OFFSET_FROM_EXTRUDER numbers?   Because those will cause some unprobed areas of the bed.   But certainly not that much area.

DirtyDerv, 2017-08-27:
Mine is doing  the exact same thing

Roxy-3D, 2017-08-28:
@oldmcg    Are you running the most recent code base?   I'm wondering if some regression of the Delta code happened?    I don't think the Cartesian machines have been impacted, but I'm thinking I'll load up the latest code and double check that as soon as I can.

oldmcg, 2017-08-28:
Sorry, have not updated in a while, could be regression. I'm unable to look into it for a few more days, weather and family issues.

From: Roxy-3D
Sent: Sunday, August 27, 2017 20:54
To: MarlinFirmware/Marlin
Cc: oldmcg; Mention
Subject: Re: [MarlinFirmware/Marlin] Delta UBL Only Probe 1/4 of the Bed(#7572)

@oldmcg Are you running the most recent code base? I'm wondering if some regression of the Delta code happened? I don't think the Cartesian machines have been impacted, but I'm thinking I'll load up the latest code and double check that as soon as I can.

Roxy-3D, 2017-08-28:
OK!   Thanks!   If the weather issue is this Harvey non-sense....   I'm definitely being hurt by that!

Roxy-3D, 2017-08-28:
> Is the program treating my printer as cartesian but not deltabot?

I don't know from the probe points.   But the Delta version of UBL does a lot of stuff similar to how a Cartesian printer does stuff.   The big difference is it restricts movement and probing to the DELTA_PRINTABLE_RADIUS and DELTA_PROBABLE_RADIUS.    The other thing it does is it 'wastes' some of the mesh points that are far away from the DELTA_PRINTABLE_RADIUS just to keep everything simple.

My guess is some 'regression' happened.   The code that does the probing is not the big.   I don't have a functioning Delta or I would fire it up and duplicate the problem so I could debug it.    But even so...  By looking the commits to that part of the code, we should be able to find the cause of the problem.   My internet access is very limited because of the Harvey problems right now.   And then on top of that, my Git skills are not super good.   This might be something a 'motivated user' can jump in and help resolve.   If you have good Git skills, please locate what has changed to the Delta UBL G29 P1 code in the last 6 weeks.    I bet with those changes identified we can figure this out very quickly.

DirtyDerv, 2017-08-28:
I wish i knew more about programmign i can see all the changes but dont knwo what im looking for

DirtyDerv, 2017-08-28:
Just a s a quick look through would this piece of code have anythign to do with it

 +    #define X_CENTER ((X_BED_SIZE) / 2)
 +    #define Y_CENTER ((Y_BED_SIZE) / 2)
thats would be fine for a cartesian bed as they work from corner.  But a delta works in the negative too

Sorry if im miles off

It was committed on 22 july

Roxy-3D, 2017-08-28:
Well...  If we don't find the problem ourselves...  In 10 days oldmcg will probably bless us with a few lines of code that fix things.

arthurkok2, 2017-08-31:
I am having similar issues as well, see the topography report below:
```

Bed Topography Report:
(0,9)                                                                   (9,9)
(1,179)                                                                (179,179)
     .          .          .          .          .          .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .

  0.084   -0.092    0.306       .          .          .          .          .          .          .

  0.127    0.012   -0.034   -0.250       .          .          .          .          .          .

  0.125    0.126    0.073   -0.031   -0.231       .          .          .          .          .

  0.108    0.133    0.096   -0.006   -0.105       .          .          .          .          .

[-0.001]   0.034   -0.050   -0.068   -0.049       .          .          .          .          .
(1,1)                                                                    (179,1)
(0,0)                                                                     (9,0)

```

emailcraigs, 2017-09-02:
Yikes....I am having this same issue except that I am attempting UBL for the first time and didn't know what the heck was going on.

Roxy-3D, 2017-09-02:
Can a few of you post your Configuration.h, Configuration_adv.h, and a G29 W output?

I bet this has something to do with the switch to using BED_SIZE as the preferred setup mechanism.

emailcraigs, 2017-09-02:
I'm In.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1271908/Configuration_adv.txt)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1271907/Configuration.txt)

G29 W

00:49:44.305 : Unified Bed Leveling System v1.01 inactive.
00:49:44.356 : Mesh 0 Loaded.
00:49:44.407 : UBL object count: 1
00:49:44.408 : planner.z_fade_height : 10.0000
00:49:44.409 : zprobe_zoffset: 0.8000000
00:49:44.411 : UBL_MESH_MIN_X  1=1
00:49:44.411 : UBL_MESH_MIN_Y  1=1
00:49:44.437 : UBL_MESH_MAX_X  (((85.0) * 2) - (1))=169.00
00:49:44.439 : UBL_MESH_MAX_Y  (((85.0) * 2) - (1))=169.00
00:49:44.465 : GRID_MAX_POINTS_X  10
00:49:44.466 : GRID_MAX_POINTS_Y  10
00:49:44.492 : MESH_X_DIST  18.67
00:49:44.493 : MESH_Y_DIST  18.67
00:49:44.776 : X-Axis Mesh Points at: 1.000  19.667  38.333  57.000  75.667  94.333  113.000  131.667  150.333  169.000
00:49:45.032 : Y-Axis Mesh Points at: 1.000  19.667  38.333  57.000  75.667  94.333  113.000  131.667  150.333  169.000
00:49:45.032 : Kill pin on :41  state:1
00:49:45.082 : ubl_state_at_invocation :0
00:49:45.084 : ubl_state_recursion_chk :0
00:49:45.134 : Meshes go from 0x0270 to 0x0F7F
00:49:45.185 : sizeof(ubl) :  1
00:49:45.187 : z_value[][] size: 400
00:49:45.213 : EEPROM free for UBL: 0x0D0F
00:49:45.264 : EEPROM can hold 8 meshes.
00:49:45.290 : Unified Bed Leveling sanity checks passed.

kevintong116, 2017-09-02:
@Roxy-3D  I'm sorry I can only help until next weekend as I left my laptop in my dormitory. 
Hope others can help.

DirtyDerv, 2017-09-02:
Ok hopefully someone can maybe say if im right with this.

this bit of code 

// Default mesh area is an area with an inset margin on the print area.
// Below are the macros that are used to define the borders for the mesh area,
// made available here for specialized needs, ie dual extruder setup.
#if ENABLED(MESH_BED_LEVELING)
  #define MESH_MIN_X MESH_INSET
  #define MESH_MAX_X (X_BED_SIZE - (MESH_INSET))
  #define MESH_MIN_Y MESH_INSET
  #define MESH_MAX_Y (Y_BED_SIZE - (MESH_INSET))
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X UBL_MESH_INSET
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y UBL_MESH_INSET
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))

in configuration_adv.h says the x an y min are basically 0 or 0 +ubl inset.
On a delta that would mean it cant go into the negative maybe?

so i will try this once i get to my printer

// Default mesh area is an area with an inset margin on the print area.
// Below are the macros that are used to define the borders for the mesh area,
// made available here for specialized needs, ie dual extruder setup.
#if ENABLED(MESH_BED_LEVELING)
  #define MESH_MIN_X MESH_INSET
  #define MESH_MAX_X (X_BED_SIZE - (MESH_INSET))
  #define MESH_MIN_Y MESH_INSET
  #define MESH_MAX_Y (Y_BED_SIZE - (MESH_INSET))
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X -(X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y -(Y_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))

Can anyoen tell me if tis is in the right direction?

Cheers

DirtyDerv, 2017-09-02:
OK so i do have a workaround it seams,  Now before you all go flaming me etc im not a coder i have a tiny bit of programming background but by no means a coder.

So as well as the changes in the post above i also changed 
// @section machine

// The size of the print bed
#define X_BED_SIZE (DELTA_PRINTABLE_RADIUS)
#define Y_BED_SIZE (DELTA_PRINTABLE_RADIUS)

taking out the *2. This is in configuration.h file.

I know this is probably not the way to do it but it seamed to work as here is my bed  probe

14:00:04.104 : Bed Topography Report for CSV:
14:00:04.333 : NAN,NAN,NAN,NAN,NAN,-0.118,NAN,NAN,NAN,NAN,NAN
14:00:04.501 : NAN,NAN,0.031,-0.017,-0.033,-0.012,-0.021,-0.018,-0.106,NAN,NAN
14:00:04.669 : NAN,0.059,-0.009,-0.054,-0.041,-0.030,-0.014,-0.004,-0.032,-0.056,NAN
14:00:04.841 : ,-0.050,-0.006,0.008,0.044,0.048,NAN
14:00:05.009 : NAN,0.027,0.008,0.004,-0.049,-0.029,0.016,0.080,0.149,0.170,NAN
14:00:05.181 : -0.103,0.091,0.053,0.029,0.023,0.004,0.086,0.203,0.257,0.329,0.136
14:00:05.348 : NAN,0.104,0.060,0.089,0.113,0.123,0.188,0.302,0.373,0.319,NAN
14:00:05.565 : NAN,0.084,0.082,0.149,0.164,0.206,0.242,0.347,0.369,-0.003,NAN
14:00:05.922 : NAN,NAN,-0.099,0.100,0.201,0.229,0.190,0.198,0.023,NAN,NAN
14:00:06.086 : NAN,NAN,NAN,NAN,NAN,0.026,NAN,NAN,NAN,NAN,NAN

So thats where im at at the minute any more ideas?

DirtyDerv, 2017-09-02:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1272178/Configurations.zip)

oldmcg, 2017-09-02:
I can look at this issue in depth on Tuesday if it is not yet resolved -- sorry I am unable to get to it before that.

Looks like @DirtyDerv is on the right track -- thanks!

Hey, @thinkyhead, since you broke UBL delta with X_BED_SIZE addition, maybe you could jump in and fix it?

DirtyDerv, 2017-09-02:
well it works as a temporary patch managed to get g29 to run through etc so just need to try a print next

DirtyDerv, 2017-09-02:
Now just got to figure out the ubl to level it lol

Roxy-3D, 2017-09-02:
In Configuration.h we have:
```cpp
// The size of the print bed
#define X_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)
#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS
```
That all looks good...   But inn Configuration_adv.h we have:
```cpp
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X UBL_MESH_INSET
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y UBL_MESH_INSET
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
...
#endif
```
That is wrong for Delta's.    I think @oldmcg should be the one that decides how we do the Delta's.  But if you can change that block of code in Configuration_adv.h to be:
```cpp
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #if ENABLED(DELTA) 
    #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
    #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
    #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
    #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
  #else
    #define UBL_MESH_MIN_X UBL_MESH_INSET
    #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
    #define UBL_MESH_MIN_Y UBL_MESH_INSET
    #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
  #endif
#endif
```
It will probably work for you until the 'correct' fix is put in place.

DirtyDerv, 2017-09-02:
you need to change the bed size bit from 
`// The size of the print bed
#define X_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)
#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS`

to 
`// The size of the print bed
#define X_BED_SIZE (DELTA_PRINTABLE_RADIUS)
#define Y_BED_SIZE (DELTA_PRINTABLE_RADIUS)

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS`

Or it thinks the bed is twice as big as it actually is i just tried it

Roxy-3D, 2017-09-02:
OK...   maybe that is what we need right now so people can use the Delta printers.  But
```cpp
#define X_BED_SIZE (DELTA_PRINTABLE_RADIUS)
#define Y_BED_SIZE (DELTA_PRINTABLE_RADIUS)
```
doesn't feel right.    The Printable Radius is only 1/2 the bed going out from the center.

DirtyDerv, 2017-09-02:
Yeah it is a bit clumsy, but for now it works the gurus like yourself and oldmcg will be able to sort the code properly, just trying to give back to a community thats helped me :)

emailcraigs, 2017-09-03:
I tried G29 P1 with the changes proposed by DirtyDerv and the process seemed to be going well, at least it was going, until the head clipped off the bed, came back and started just barely scrapping across the bed. I pressed emergency stop at that point, because I'm not very brave. ;-)  I have an FLSUN Delta. The DELTA_PRINTABLE_RADIUS is 85.0 so shouldn't it only be probing a maximum of 85mm from the center or less? What do I need to change to define my probe area correctly?

Roxy-3D, 2017-09-03:
Well...   There are a couple of things you can do.   First....  In just a few days oldmcg will be back in town and it won't take him long to figure out what to tweak in to code to make the problems go away.

But independent of that...   You can do a brute force solution too.  You can do a G29 P0 and set the whole mesh to 0.000 mm.    Assumming you really can print a small object at the center of the  bed with no bed leveling...   you put the nozzle at (0,0) and start a G26.   As long as the pattern looks reasonable, you let it go.   If the nozzle gets too close to the bed you press and hold the encoder wheel and stop it.

And you go to G29 P4 and edit as much of the mesh that you can see to be closer to the truth.   And then you restart the G26 (with a clean bed) and go further out from the center.   In 20 or 30 minutes you will have the mesh perfectly defined.

emailcraigs, 2017-09-03:
Cool, I will try that. I downgraded to 1.1.4 and still had the same issue with the probe wanting to go off the bed.....wouldn't 1.1.4 be before the issues with 1.1.5?

Roxy-3D, 2017-09-03:
Find the date when BED_SIZE entered in.  Roughly 2 or 3 weeks ago.   Grab
the code just before that happened.  I think you will find the Delta stuff
is correct and works well.

emailcraigs, 2017-09-03:
yeah, I looked around and the only reference I could find was in the 1.1.5 release so I went back to the 1.1.4. I did the G26 and it ran completely through. I'm attaching a pic. The white is where I tested a print with no leveling to be sure my center was good; it comes out perfect. I could probably have upped the retraction on the G26? So it seems to know my bed dimensions pretty well. The left and right fringes are a bit high...the rest low in places and probably fixable by decreasing Z-offset?
[image]

emailcraigs, 2017-09-04:
Well, I've reverted back as far as 1.1.3 and still have issue of probe, which is my hotend, trying to crash off the bed. When I first do the G29 P1 it doesn't go straight down to where X0 Y0 would be, it shifts left first and then goes down to start probing; it is also the left side that it attempts to crash off. I wonder if something is wrong with the way my middle of the bed is being calculated? The G26 seems to map the entire bed just fine.....so I guess I will take Roxy's suggestion and continue with brute forcing the mapping.

Roxy-3D, 2017-09-04:
Don't mess with the Z-Offset now.   Just edit the mesh points.   Any place the filament is too squished down...  Raise those mesh points a little bit.   Maybe .1mm for the bad ones.    Any place you are not getting good adhesion, lower those mesh points.  Maybe .1mm for the first pass.     

Once you can get the whole pattern to print, you can continue editing.  But only move the mesh points by .05mm at a time.   The process will converge very quickly.   In 3 or 4 G26 patterns, you will have the mesh very well tuned.

emailcraigs, 2017-09-04:
How do I tell it to cycle through each mesh point and let me adjust them? I'm using G29 P4 T but it doesn't let me go to all of them.

Roxy-3D, 2017-09-04:
It should do all the reachable ones...   But for a Delta, I don't know the logic used.   If you move the nozzle to the center of an area that needs editing....   Probably a G29 P4 R will let you edit as many as you want.   If that doesn't do that...  Try a G29 P4 R99      

oldmcg did the conversion for Delta's and I'm just guessing about things.     Also...   One thing that needs to be done is the ability to edit mesh points outside the Delta Printable Radius.  That isn't possible with G29 P4 yet.  If you need to edit those mesh points, you need to use the more crude M421 command.

DirtyDerv, 2017-09-04:
If it doesnt go to the centre of the bed do you have a probe offset set like x -10 or something ?  My probe is my nozzle so i dont have any offsets accepts z .4

oldmcg, 2017-09-04:
Admittedly we need a way to tune the grid points just outside the printable radius that can't be reached by probe or nozzle.  I have a solution in mind for this but it's going to be a while.

In the meantime, while tuning grid with G29 P4, initially ignore the grid points outside the printable radius, intentionally leaving them uninitialized (not zero or other value).  After tuning the grid with G26 and G29 P4, then use G29 P3.15 to set all the unreachable (uinitialized) points using nearest points and average bed tilt.  This will produce the best results for the area between the printable radius and the nearest mesh points inside the printable radius.

I recommend using an odd number of mesh points, like 11 or 13 or 15 so that you get one point in exact center at 0,0.  You can leave UBL_MESH_INSET at 0 as long as your nozzle can safely reach all of DELTA_PRINTABLE_RADIUS.

With respect to head/nozzle moving outside of DELTA_PRINTABLE_RADIUS, it should never do that.  Most likely something related to the BED_SIZE changes added recently might have short circuited the code that prevents any nozzle movements outside DELTA_PRINTABLE_RADIUS.

Starting tomorrow I'll have more time to look at the code and see about proper fix for BED_SIZE definitions and DELTA_PRINTABLE_RADIUS movement enforcement.

emailcraigs, 2017-09-04:
I'd like to try a version from before the code change but I've tried back to 1.1.3 and I've tried the only version of bugfix-1.1.x that seems downloadable from the main link. I can't seem to find an earlier version from before the code change. Could someone point me in the right direction?

Roxy-3D, 2017-09-04:
Its a little bit buried...  But you can download a .ZIP file of any branch at any point in time.

emailcraigs, 2017-09-05:
I know this might be a dumb question, but I'm going to ask it anyway. If I change my DEFAULT_AXIS_STEPS_PER_UNIT does it invalidate the mesh I have previously created? If not, what would invalidate the mesh.

Roxy-3D, 2017-09-05:
It the mesh locations are different from the previous setup...   The mesh is no longer valid.   If you change the axis steps per unit but also compensate by using a different stepper motor or what ever, and the mesh points are in the same location...  The mesh would still be valid.

But really...  It only takes 15 minutes to get a well tuned mesh.   Or 30 minutes if you do several passes with G26 and editing of the mesh points.   There really is no reason to try to use an old and obsolete mesh.

emailcraigs, 2017-09-05:
Thanks Roxy!

jam3sward, 2017-09-09:
I just ran into the same issue with my Kossel on 1.1.5, and tried fixing it by modifying the UBL_MESH_... defines in Configuration_adv.h as follows:
`
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - UBL_MESH_INSET)
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS - UBL_MESH_INSET)
`
I also increased UBL_MESH_INSET to avoid collisions with the towers.

j3qq4hch, 2017-09-28:
Hi! I ran into same issue and here is how I fixed it.
I made changes in `Configuration_adv.h`

I changed 
```
  #define UBL_MESH_MIN_X UBL_MESH_INSET
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y UBL_MESH_INSET
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
```
To 
```
  #define UBL_MESH_MIN_X (-DELTA_PROBEABLE_RADIUS)
  #define UBL_MESH_MAX_X DELTA_PROBEABLE_RADIUS
  #define UBL_MESH_MIN_Y (-DELTA_PROBEABLE_RADIUS)
  #define UBL_MESH_MAX_Y DELTA_PROBEABLE_RADIUS
```

I also use this manually mountable probe
https://www.thingiverse.com/thing:1976680

I use version2, It mount just almost right under the hod-end, so XY offsets are not an issue.

Roxy-3D, 2017-09-28:
This seems like a good way to fix the issue!

oldmcg, 2017-09-28:
DELTA_PROBEABLE_RADIUS wasn't previously used by UBL on delta, intentionally.

DELTA_PRINTABLE_RADIUS is the definitive "don't move nozzle outside this circle", and the probeable area (with probe offset from nozzle) looks like a crescent moon, not a circle, so it cannot be described by a "radius".  There are routines in marlin.h called position_is_reachable_raw_xy and position_is_reachable_by_probe_raw_xy that define where the limits of horizontal motion are.

UBL_MESH_MIN_X etc should probably be defined relative to DELTA_PRINTABLE_RADIUS and UBL_MESH_INSET, but I haven't studied the other implications and usage of UBL_MESH_MIN, etc since someone else added X_BED_SIZE, etc, and broke UBL on delta.  

I would think UBL_MESH_MIN_X would be (-DELTA_PRINTABLE_RADIUS+UBL_MESH_INSET), and UBL_MESH_MAX_X would be (+DELTA_PRINTABLE_RADIUS-UBL_MESH_INSET), etc.

I wish I had time to look into this further but I just don't right now.  Sorry.

j3qq4hch, 2017-09-28:
What is the point of having a definition DELTA_PROBEABLE_RADIUS  if you don't use it to define PROBEABLE RADIUS? 
What is UBL_MESH_INSET? Why it even exists? 
Anyway, I shared my fix and I must say, after I repaired UBL this way, my anycubic linear plus suddenly became a very decent machine. 
UBL is a must.

Roxy-3D, 2017-09-28:
> I would think UBL_MESH_MIN_X would be (-DELTA_PRINTABLE_RADIUS+UBL_MESH_INSET), and UBL_MESH_MAX_X would be (+DELTA_PRINTABLE_RADIUS-UBL_MESH_INSET), etc.

Yes...   That looks even more 'correct'.   Certainly, it is better than what we have right now.  I'll make the change to this when I get a chance.

> I wish I had time to look into this further but I just don't right now. Sorry.

Not a problem!   

> *What is UBL_MESH_INSET? Why it even exists?

UBL_MESH_INSET is a carry over from the original Mesh Leveling System.   There were places on the bed where clamps were used to hold the glass, and the inset was used to limit movement of the nozzle to keep it from going to the edge of the glass.   Independent of that...  I have a very large bed printer but I could not get a silicon heat-pad and a piece of borosilcate glass the right size.  So I have a 45mm inset to keep the nozzle in the center of the bed.   The Inset number is of limited value.  But when a few weird circumstances come up, it is very helpful to resolve them.

> Anyway, I shared my fix and I must say, after I repaired UBL this way, my anycubic linear plus suddenly became a very decent machine.   **UBL is a must.**

Yes.  Agreed.

ashthespy, 2017-10-26:
Just to check, with 84470f6b2a623988824fa436ca687f8b34110604 should fix this correct?
Will test tomorrow. :-)

Roxy-3D, 2017-10-26:
Yes.  I believe so.    Grab the latest bugfix and give it a try.    But a few more changes have been made to get all the bed leveling systems to use the same option names.   There is a small chance another problem has crept into the mix.

jam3sward, 2017-10-27:
I fixed this in a different way, with:
`
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - UBL_MESH_INSET)
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS - UBL_MESH_INSET)
`
Where those limits are defined as follows for Delta:
`
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
`
Here's my working branch:
https://github.com/jam3sward/Marlin/commit/6504089fcbf47778f28dda547a68d400268759ec#diff-18dab9d337cee2f9d2209940ed08dbacR682

I've been running this for a while now without problems. The UBL_MESH_INSET is useful on a Delta to give a bit of clearance to keep the effector and rods away from the towers.

UBL is outstanding, an absolute must - it has transformed my printer!

ashthespy, 2017-10-27:
I just tried the latest bugfix branch. May be I don't complete grasp the idea of UBL, but here is what I have now:
[image]

I would expect the probe points to be somewhat centred?

jam3sward, 2017-10-27:
If it helps, I've got a tested and working build with UBL here on
thingiverse:
https://www.thingiverse.com/thing:1976680

ashthespy, 2017-10-28:
@jam3sward Would you be able to share what your probe mesh looks like? 
I'm trying to understand the logic of the mesh generation. 
It should be a grid based on `MESH_MAX_X` , `MESH_MAX_Y` , `MESH_MIN_X` ,`MESH_MIN_Y` 
i.e, using the new naming convention of:
```c++
  #define MIN_PROBE_X (X_CENTER - (DELTA_PROBEABLE_RADIUS))
  #define MIN_PROBE_Y (Y_CENTER - (DELTA_PROBEABLE_RADIUS))
  #define MAX_PROBE_X (X_CENTER +  DELTA_PROBEABLE_RADIUS)
  #define MAX_PROBE_Y (Y_CENTER +  DELTA_PROBEABLE_RADIUS)

 #define _MESH_MIN_X (MIN_PROBE_X + MESH_INSET)
 #define _MESH_MIN_Y (MIN_PROBE_Y + MESH_INSET)
 #define _MESH_MAX_X (MAX_PROBE_X - (MESH_INSET))
 #define _MESH_MAX_Y (MAX_PROBE_Y - (MESH_INSET))
```
Or am I being silly here?

Roxy-3D, 2017-10-28:
I don't have a Delta that I can test that on.   But that looks 'reasonable'.     However...   The mesh points have to extend one more outward from the DELTA_PRINTABLE_RADIUS just because we need any correction calculation to have mesh points all the way around it.   @oldmcg made the G29 P3 handle that correctly for Delta's.

ashthespy, 2017-10-28:
@Roxy-3D to clear up any ambiguities - my [probe map](https://github.com/MarlinFirmware/Marlin/issues/7572#issuecomment-340049233) is with the settings from my comment above (i.e the default ones now (f06f4b4) 
It still seems to want to probe only a quarter of the points?

Roxy-3D, 2017-10-28:
I think for Delta's, these values used to be:
```cpp
    #define MESH_MIN_X (X_CENTER - (DELTA_PROBEABLE_RADIUS) + UBL_MESH_INSET)
    #define MESH_MIN_Y (Y_CENTER - (DELTA_PROBEABLE_RADIUS) + UBL_MESH_INSET)
    #define MESH_MAX_X (X_CENTER +  DELTA_PROBEABLE_RADIUS - (UBL_MESH_INSET))
    #define MESH_MAX_Y (Y_CENTER +  DELTA_PROBEABLE_RADIUS - (UBL_MESH_INSET))
```
That is what used to be there (except with a UBL_ in front of UBL_MESH_MIN_?).   But now there seems to be a concept of 'usable mesh area' which is inset into the printable bed:
```cpp
  #if IS_KINEMATIC
     // Probing points may be verified at compile time within the radius
     // using static_assert(HYPOT2(X2-X1,Y2-Y1)<=sq(DELTA_PRINTABLE_RADIUS),"bad probe point!")
     // so that may be added to SanityCheck.h in the future.
     #define _MESH_MIN_X (MIN_PROBE_X + MESH_INSET)
     #define _MESH_MIN_Y (MIN_PROBE_Y + MESH_INSET)
     #define _MESH_MAX_X (MAX_PROBE_X - (MESH_INSET))
     #define _MESH_MAX_Y (MAX_PROBE_Y - (MESH_INSET))
   #else
```

ashthespy, 2017-10-30:
> 
>  > I just saw f06f4b4
> (I'm presuming that is what broke the correct behavior. If that is not true... Please correct me!!! )

To clarify, everything is working fine without touching anything in `Configuration_adv.h` as well, leaving the values that f06f4b4 have set - as in your post.
It just had to manually set regions that were outside the `DELTA_PRINTABLE_RADIUS` due to the probe offset. So with a `G29 P2` everything is good! :-)

I think (at least for me) this issue can be closed.

mylife4aiurr, 2017-10-31:
I can report this still is a bug, even though it was just closed. 
G33 probes whole bed.
G29 with ubl only probes half my bed.

I tried j3qq4hch's code:

  #define UBL_MESH_MIN_X UBL_MESH_INSET
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y UBL_MESH_INSET
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
To
  #define UBL_MESH_MIN_X (-DELTA_PROBEABLE_RADIUS)
  #define UBL_MESH_MAX_X DELTA_PROBEABLE_RADIUS
  #define UBL_MESH_MIN_Y (-DELTA_PROBEABLE_RADIUS)
  #define UBL_MESH_MAX_Y DELTA_PROBEABLE_RADIUS

That works beautifully. Has this been fixed another official way?

[10 more replies]

closed 2018-02-02 by thinkyhead