MarlinFirmware/Marlin #7574: Temperature drops at the begining

DaniDiaz71, 2017-08-27:
Hello,

I would like to know if I'm doing something wrong, I use Marlin 1.1.4 at my Prusa i3 (200x300 heated bed, plus inductive probe). Everytime printer starts, heats OK nozzle and bed then moves to (X=10,Y=10) feeds about 20 mm of filament and finally moves to next point to print first layer. But when extrusor moves from (10,10) to the first point to print, temperature drops about 10-20 ºC, it seems like temperature control process does not respond when extruder moves. I think this behaviour only happens with G0 command, but I don't know almost nothing about how Marlin works.

Here you can find GCODE sample:

```
;FLAVOR:RepRap
;TIME:3256
;Filament used: 0.974652m
;Layer height: 0.1
;Generated with Cura_SteamEngine master
M190 S100
M104 S240
M109 S240
; -- START GCODE --
G21                     ;set units to millimetres
G90                     ;set to absolute positioning
M107                    ;set fan speed to zero (turned off)
G28
G1 X10 Y10 Z15.0 F1200  ; X=10; Y=10; z=15
G92 E0                  ;zero the extruded length
G1 E20 F200             ;extrude 20mm of feed stock
G92 E0                  ;zero the extruded length again
;LAYER_COUNT:98
;LAYER:0
M107
G0 F600 X33.101 Y29.271 Z0.05 // **Maybe at this point temp drops?**
;TYPE:SKIRT
G1 X33.218 Y29.154 E0.00906
G1 X33.444 Y28.985 E0.02453
G1 X33.708 Y28.885 E0.03999
.
.
.
```

Also a screenshot:

[image]

And finally config files:

[Configuration.h](https://pastebin.com/m2c4HZZp)
[Configuration_adv.h](https://pastebin.com/Uj8XdE4v)

Thank you to everybody is working improving Marlin.

emartinez167, 2017-08-27:
There could be a loose cable in the hot end that momentarily loses continuity due to the move. Check your cable under different positions and see if this repeats.

DaniDiaz71, 2017-08-27:
I don't think so, this problem only happens at the beginning, after that printer works OK even large objects (12 hours or more). But if helps, here you can see what happens from OctoPrint point of view:

[OctoPrint capture](https://youtu.be/zm19xQIY3H8)

This issue also happens using SD card.

Tannoo, 2017-08-27:
Is it probing for the start point? Because it seems as though both heaters are shutting down. (The bed drops also)

Tannoo, 2017-08-27:
Try turning off `PROBING_HEATERS_OFF` iand `PROBING_FANS_OFF` in `configuration.h` and see if that helps.

DaniDiaz71, 2017-08-30:
HI,

both PROBING_HEATERS_OFF and PROBING_FANS_OFF are defined in configuration.h, but i found a way to simulate this behaviour. This are the steps:

```
G28  // HOMING
M104 S240V// HERE I WAIT AFTER HOTEND REACH 240ºC
G0 Z150 // MOVES EXTRUDER TO Z=150
G28 // HOMING AGAIN
```

In the last G28 I can see how temperature controller does not respond, so when extruder is coming down hotend temprature drops from 240ºC to 191ºC!. Here you can see an snapshot of two test (look only hotend):

[image]

So my question is, is there any place in configuration or configuration_adv to modify this behaviour? I didn't found it.

Tannoo, 2017-08-30:
Again..  disable `PROBING_HEATERS_OFF`.

DaniDiaz71, 2017-08-30:
OK, sorry I didn' read slowly :)

closed 2017-08-31 by DaniDiaz71